realsenseai/librealsense #10252: Requested extrinsics are not available! - D415

mnauf, 2022-02-21:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |  N/A                                      | 
| Platform                                 | PC (Lenovo Legion 5i  |
| SDK Version                            |  { legacy / 2.50.0 }                          | 
| Language                            |  Python  | 
| Segment			|  not sure what it means | 

### Issue Description

I want to find out extrinsic parameters between two different views. However, I only have a single camera. I recorded videos of a certain static scene from two different viewpoints. And tried to find out extrinsic params, but it says "Requested extrinsics are not available!". I thought maybe it couldn't find any matching points between two different angles, so I now I am trying to compare the same example with itself to make sure this error isn't related to matching points. But it still gives me the same error.

`import pyrealsense2 as rs
import numpy as np
import cv2

pipeline_1 = rs.pipeline()
config_1 = rs.config()
anchor_example = r"D:\librealsense2\data\anchor.bag"
config_1.enable_device_from_file(anchor_example, False)
config_1.enable_stream(rs.stream.depth, rs.format.z16, 30)

pipeline_2 = rs.pipeline()
config_2 = rs.config()
second_view_example = r"D:\librealsense2\data\anchor.bag"
config_2.enable_device_from_file(second_view_example, False)
config_2.enable_stream(rs.stream.depth, rs.format.z16, 30)

# Start streaming from both cameras
profile1 = pipeline_1.start(config_1)
profile2 = pipeline_2.start(config_2)
profile_1 = profile1.get_stream(rs.stream.depth)
profile_2 = profile2.get_stream(rs.stream.depth)

def get_extrinsics(src, dst):
    extrinsics = src.get_extrinsics_to(dst)
    R = np.reshape(extrinsics.rotation, [3,3]).T
    T = np.array(extrinsics.translation)
    return (R, T)

def write_to_file(stream):
    with open("stream_left", "w+") as f:
        f.write(str(stream))

counter = 0
try:
    while True:
        counter += 1
        # Get frameset of color and depth
        aligned_frames_1 = pipeline_1.wait_for_frames()
        aligned_frames_2 = pipeline_2.wait_for_frames()

        # Get aligned frames
        aligned_depth_frame_1 = aligned_frames_1.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        aligned_depth_frame_2 = aligned_frames_2.get_depth_frame()

        # Validate that both frames are valid
        if not aligned_depth_frame_1 or not aligned_depth_frame_2:
            print(":(")
            continue

        depth_image_1 = np.asanyarray(aligned_depth_frame_1.get_data())
        depth_image_2 = np.asanyarray(aligned_depth_frame_2.get_data())

        # Stack all images horizontally
        images = np.hstack((depth_image_1, depth_image_2))

        # Show images from both cameras
        cv2.namedWindow('RealSense', cv2.WINDOW_NORMAL)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)

        stream_left = profile_1.as_video_stream_profile()
        stream_right = profile_2.as_video_stream_profile()
        streams = {"left": stream_left,
                   "right": stream_right}
        (R, T) = get_extrinsics(streams["left"], streams["right"])
        print("R: ", R)
        print("T: ", T)
        break
finally:
    pipeline_1.stop()
    pipeline_2.stop()
`

The strange thing is, if I change `(R, T) = get_extrinsics(streams["left"], streams["right"])` to `(R, T) = get_extrinsics(streams["left"], streams["left"])`(Note that I am comparing the same stream with itself, it works fine and returns me the T = [0,0,0] and R = 
[[1,0,0],
[0,1,0],
[0,0,1]].

Long story short, when streams from different config objects(`rs.config()`) are compared, it doesn't show me extrinsics, however, If I compare the same stream with itself(`get_extrinsics(streams["left"], streams["left"])`) using a single config object, it gives me the answer but of course isn't useful to me since I can't have two different streams with a single config object.

MartyG-RealSense, 2022-02-21:
Hi @mnauf  In your script, you define profile1 and profile2 as being pipe start instructions, and then in the following two lines change what the profile1 and profile2 variables are storing.

[image]

Does the program perform better if you change the second set to 'depth_profile_1' and 'depth_profile_2'

```
depth_profile_1 = profile1.get_stream(rs.stream.depth)
depth_profile_2 = profile2.get_stream(rs.stream.depth)
```

mnauf, 2022-02-21:
Hello @MartyG-RealSense . I haven't overwritten `profile1` and `profile2` in the two lines you have highlighted. Because those variables are `profile_1` and `profile_2` and not `profile1` and `profile2`. Please note the `_`

MartyG-RealSense, 2022-02-21:
Thanks for the clarification.

As you noted, the problem seems to be occurring with the line **(R, T) = get_extrinsics(streams["left"], streams["right"])**

This line and other in the script are used in the T265 Tracking Camera's **t265_stereo.py** example for simulating depth on the T265 (as T265 does not have depth sensing built in).  

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py#L157

This suggests that the lines are structurally correct, though in the case of the T265 script it is analyzing the extrinsics between left and right **fisheye** sensors in the same camera instead of depth sensors.

Extrinsics transforms like the ones in the code are typically used to describe the position and rotation transform between two different sensor types that are mounted in different physical positions in the camera (depth and color, infrared and depth, gyro and depth, etc).  The RealSense SDK does not generate extrinsics for comparing depth and depth, because that would not make sense in terms of extrinsics - it would be like checking the positions of two sensors that are in the same place within the camera.  

Because the extrinsics compare the transform of two different sensors **inside the same camera**, the extrinsics are not going to change if the entire camera is moved to a different position in the scene because the camera is moving but the sensors inside it are fixed in the same position **relative to each other**.

***************

It sounds like what you are doing is trying to align data recorded into two different files at two different perspectives.  If this is your aim then the SDK's **rs2_transform_point_to_point** instruction may be more suitable, as it performs an **affine transform** to align two different sets of data in the same 3D coordinates.  An example of this would be capturing two point clouds from different angles and then performing an affine transform to set them to the same position and rotation in 3D space to form a single combined ('stiched') view.

MartyG-RealSense, 2022-02-28:
Hi @mnauf  Do you require further assistance with this case, please?  Thanks!

mnauf, 2022-02-28:
> Hi @mnauf  Do you require further assistance with this case, please?  Thanks!

Let me try rs2_transform_point_to_point in a few hours. Meanwhile please keep this issue opened.

MartyG-RealSense, 2022-02-28:
Ok, no problem at all, @mnauf - it s fine to keep this case open for a further time period.  Thanks for the update!

mnauf, 2022-03-19:
@MartyG-RealSense thanks. You may close this issue now :D

MartyG-RealSense, 2022-03-19:
Thanks very much @mnauf for the update!  As you suggested, I will close the case.   Thanks again!

closed 2022-03-19 by MartyG-RealSense