realsenseai/librealsense #10253: Unstable installation for d435i

JaeYoungLee79, 2022-02-21:
Ubuntu 18.04 / ROS1 / D435i
-------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 3, in <module>
    from realsense_depth import *
ModuleNotFoundError: No module named 'realsense_depth'
---------------------------------------------------------------------------------------
I feel that d435i camera installation is more difficult than Full Dynamic of Mobile Manipulator. 
Because of that there is so many installation guide. But, it not operating.

Best Regards

MartyG-RealSense, 2022-02-21:
Hi @JaeYoungLee79  As you mentioned ROS, are you attempting to run a Python script from ROS, please?

Does librealsense and the RealSense ROS wrapper function normally for you except for the problems with your test.py script?

JaeYoungLee79, 2022-02-22:
$ realsense-viewer and Other python examples are operating well.
But, there is not installed with "from realsense_depth import *" as I guess.
Now, I'm not use that function module.
I'm searching internet about "ModuleNotFoundError" and the result is
1) wrapper
2) python path

Conclusion, I don't understand 1) and 2).
I just want to use D435i example by using "from realsense_depth import *".

Thanks
Best Regards

MartyG-RealSense, 2022-02-22:
The only reference to **from realsense_depth import** that I can find is https://github.com/IntelRealSense/librealsense/issues/9901 - is this your case under a different GitHub account name, please?

I understand that the script is trying to import a function called **realsense_depth_import**.  Pyrealsense2 scripts are never programmed like this though and always include **import pyrealsense2 as rs** in the import list.  So I do not know what to suggest, unfortunately.

roelofvandijk, 2022-02-22:
Hello JaeYoungLee, the file `realsense_depth.py` is not a file from the realsense python package, but contains helper functions and is created by an external person.

If you want to use that file, you will need to download it from the tutorial page here:
https://pysource.com/2021/03/11/distance-detection-with-depth-camera-intel-realsense-d435i/

If you want to get started with the official `pyrealsense2` package, you can find an example here:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

closed 2022-03-06 by MartyG-RealSense