realsenseai/librealsense #10256: Affect of realsense parameters on Depth & IR frames

rupak-d, 2022-02-21:
| Camera Model                        | D400 | 
| Firmware Version                    | 5.12.10.00 | 
| Operating System & Version   | Linux | 
| Kernel Version (Linux Only)    |  5.4.0 | 
| SDK Version                            |  2.43.0 | 
| Language                                |  C++  | 
| Segment			          | Robot | 

### Issue Description
Wanted some more info on the following realsense camera options I saw here [https://github.com/IntelRealSense/librealsense/blob/v2.50.0/include/librealsense2/h/rs_option.h](url)
1. Backlight compensation
2. Auto exposure mode
3. Enable motion correction
4. Auto exposure priority
5. Freefall detection enabled
6. Invalidation bypass
7. Digital gain
8. Enable max usable range
9. Receiver sensitivity

Only using the depth and IR frames from D430 module, so mainly concerned with how it affects those two outputs.

MartyG-RealSense, 2022-02-22:
Hi @rupak-d  Descriptions for each option are provided in the **rs_option.h** file that you linked to, and the official RealSense SDK documentation for these options has the same description text. 

https://github.com/IntelRealSense/librealsense/blob/v2.50.0/include/librealsense2/h/rs_option.h#L22

Of those 9 options, the two main ones that should be focused on in regard to your D430 are **Auto-Exposure** and **Auto-Exposure Priority**.

Auto-Exposure automatically adjusts the camera's exposure values depending on the current lighting conditions in the scene that is being observed.

Auto-Exposure Priority allows the camera's frames per second (FPS) rate to be dynamically altered depending on the current lihting conditions.  This is often considered to be undesirable, as if depth and RGB streams are both enabled then one stream may have a lower FPS rate than the other.  Having Auto-Exposure enabled and **disabling** Auto-Exposure Priority should enforce a constant FPS rate for both streams instead of permitting the rate to fall significanty below the speed that was set for the stream.

rupak-d, 2022-02-23:
Thanks @MartyG-RealSense!
Does `RS2_OPTION_ENABLE_MAX_USABLE_RANGE` help with depth data quality if there is a lot of light or glare affecting the camera?

MartyG-RealSense, 2022-02-23:
The API Changes notes list that **RS2_OPTION_ENABLE_MAX_USABLE_RANGE** was a feature for the RealSense L515 lidar depth camera model and so would not apply to 409 Series stereo depth cameras like the D430.

https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2400

MartyG-RealSense, 2022-03-01:
Hi @rupak-d  Do you require further assistance with this case, please?  Thanks!

rupak-d, 2022-03-01:
No that'll be all. Thanks!

closed 2022-03-01 by rupak-d