realsenseai/librealsense #10257: RGB stream seems slower than Depth stream in the viewer

WellYoungIOE, 2022-02-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    |     | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0                          | 
| Language                            |                     | 
| Segment			|  others                         | 

### Issue Description
The RGB stream seems slower than the Depth stream in the realsense-viewer. Is it normal?[image]

The above issue does not happen when I use the realsense-viewer on Jetson Xavier.

MartyG-RealSense, 2022-02-22:
Hi @WellYoungIOE  Do the frames per second (FPS) rates of the depth and RGB streams equalize if you go to the **RGB** options in the side panel, expand open the **Controls** section and enable the option called **Auto-Exposure Priority** by setting the box beside it to blue (On)

WellYoungIOE, 2022-02-22:
1. The issue remains when I turn Auto-Exposure Priority on.
2. The FPS of depth and RGB stream are both around 30.

I found that when I move the realsense-D455, the RGB stream did not synchronize with the movement. The RGB stream is slower than the movement.

MartyG-RealSense, 2022-02-22:
Auto-Exposure should be On (blue box) and the separate option Auto-Exposure Priority should be **Off** (black colored box) in order for a constant FPS rate to be enforced.  It sounds as though your problem is not with the FPS rate though if both are 30.

The RGB sensor on the D455 has a **global shutter** that is suited for capturing fast motion.  The depth sensor also has a fast global shutter.  So the D455's RGB should be less vulnerable to negative effects from fast motion than earlier RealSense models such as D415 and D435 that have a slower 'rolling' shutter on the RGB sensor.

It looks as though the location that the camera is being used in is full of fluorescent ceiling strip lights, which can have a negative effect on performance because they flicker at rates that are hard to see with the human eye.  If fluorescent lights are responsible then these negative effects may be reduced by going to the **RGB** options in the Viewer, finding an option called **Power Line Frequency** and selecting from its drop-down menu a frequency (50 or 60) that closely matches the operating frequency of the lights.  60 would be typical for North American countries, whilst 50 is more common in European regions.

[image]

MartyG-RealSense, 2022-03-01:
Hi @WellYoungIOE  Do you require further assistance with this case, please?  Thanks!

WellYoungIOE, 2022-03-01:
> Hi @WellYoungIOE Do you require further assistance with this case, please? Thanks!

Hi, I‘m so sorry that I have been busy with other things these days. 

The solution you mentioned above has not solved my problem. Is it possible that this situation is caused by the low computing power of Mac? Because it does not happen when I use realsense-viewer on Jetson Xavier.

Thanks!

MartyG-RealSense, 2022-03-01:
Which version of MacOS are you using please (Catalina, Big Sur, Monterey, etc).  If it is Monterey then there have been problems reported with using RealSense cameras with it at https://github.com/IntelRealSense/librealsense/issues/9916

WellYoungIOE, 2022-03-02:
> Which version of MacOS are you using please (Catalina, Big Sur, Monterey, etc). If it is Monterey then there have been problems reported with using RealSense cameras with it at #9916

The macOS is Big Sur.

MartyG-RealSense, 2022-03-02:
On your Jetson installation of librealsense, do you have **CUDA** support enabled?  This would be enabled by default if you built librealsense from Jetson packages, or if you built from source code with CMake with the **-DBUILD_WITH_CUDA=true** build flag included in the build instruction.

When CUDA support is enabled on a device with an Nvidia graphics GPU (which includes Jetson boards) then acceleration for color conversion would be automatically provided by offloading processing work from the CPU onto the GPU.  The link below provides examples of CPU percentage utilization when performing color conversion with and without CUDA support.

https://github.com/IntelRealSense/librealsense/pull/1866

WellYoungIOE, 2022-03-03:
> On your Jetson installation of librealsense, do you have **CUDA** support enabled? This would be enabled if you built librealsense from Jetson packages, or if you built from source code with CMake with the **-DBUILD_WITH_CUDA=true** build flag included in the build instruction.
> 
> When CUDA support is enabled on a device with an Nvidia graphics GPU (which includes Jetson boards) then acceleration for color conversion would be automatically provided by offloading processing work from the CPU onto the GPU. #1866 provides examples of CPU demand when performing color conversion with and without CUDA support.

Yes, CUDA support is enabled on my jetson installation of librealsense. Do you mean it is normal that the RGB stream is slower than the depth stream because I can not install librealsense with CUDA on Mac?

WellYoungIOE, 2022-03-04:
> You mentioned in [#10257 (comment)](https://github.com/IntelRealSense/librealsense/issues/10257#issuecomment-1047701931) that both your depth and RGB streams have an FPS of 30, so it does not seem as though there is an FPS lag on the RGB stream on your Mac that could account for the RGB not staying aligned with depth during motion. I could not rule out that there might be a **graphics** rendering issue on your Mac though.
> 
> CUDA support is only available for computers / computing devices with Nvidia graphics GPUs and so **-DBUILD_WITH_CUDA=true** should not be included in a CMake build instruction if building librealsense with CMake on a Mac.
> 
> Does the problem still occur if you set depth to 30 FPS and RGB to 60 in the Viewer? You would only be able to use a maximum RGB resolution of **848x480** instead of 1280x720 at 60 FPS though.

The problem is disappeared after I set RGB to 60 FPS, but the highest resolution I can set is 640x480.

MartyG-RealSense, 2022-03-04:
That's unusual.  848x480 RGB at 60 FPS should be available on a D455.

[image]

An alternate method of reducing RGB blurring than using 60 FPS and auto-exposure is to disable RGB auto-exposure, set the FPS to minimum (5 FPS on D455) and set RGB exposure to a lower manual value of 78.

WellYoungIOE, 2022-03-13:
> That's unusual. 848x480 RGB at 60 FPS should be available on a D455.
> 
> [image]
> 
> An alternate method of reducing RGB blurring than using 60 FPS and auto-exposure is to disable RGB auto-exposure, set the FPS to minimum (5 FPS on D455) and set RGB exposure to a lower manual value of 78.

Yes, 848x480 RGB is available on D455, but it will make the RGB stream slower than the depth stream.

Since realsense running well on Xavier, I choose not to use realsense with Mac.

Thanks for your reply and all your patience. 

The problem is not solved but I can avoid it by using another platform.

MartyG-RealSense, 2022-03-13:
Thanks very much @WellYoungIOE for the update.

closed 2022-03-24 by MartyG-RealSense