realsenseai/librealsense #10258: Multi-device sync via rs2::syncer broken

roelofvandijk, 2022-02-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.13.00.50 
| Operating System & Version |   Win 11 | 
| Kernel Version (Linux Only)    |                                 | 
| Platform                                 | PC |
| SDK Version                            | 2.50                     | 
| Language                            |  C++                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

After upgrading from 2.38 to 2.50, using a rs2::syncer to sync frames from 2 D415 cameras does not work anymore.

This comment contains a minimal example that seems to be broken now: https://github.com/IntelRealSense/librealsense/issues/4158#issuecomment-505559462

The syncer now returns a frameset only containing half of the expected frames.

This issue points towards a potential change that causes this: https://github.com/IntelRealSense/librealsense/issues/9030

Is there any way to make this work using the current release, or should I revert to an older one?

roelofvandijk, 2022-02-23:
It seems possible that the reason for this is missing HW timestamps on Windows 11, even after running the script.
Per this comment: https://github.com/IntelRealSense/librealsense/issues/2662#issuecomment-442589025
`timestamp_domain` shows GLOBAL_TIME instead of HARDWARE_CLOCK

MartyG-RealSense, 2022-02-23:
Hi @roelofvandijk   If you suspect that the problem that you are experiencing began with SDK 2.43.0 then a logical starting point in investigating this case would be to install the **2.42.0** version of the full RealSense SDK for Windows to see whether your program works currently under that SDK version.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0

[image]

The syncer changes that were introduced in 2,43.0 are quoted below in an extract from the 2.43.0 release notes.

[image]

MartyG-RealSense, 2022-03-01:
Hi @roelofvandijk  Do you require further assistance with this case, please?  Thanks!

roelofvandijk, 2022-03-01:
Hi @MartyG-RealSense,

The current syncer behaviour seems like a regression that makes a common use-case (software-sync for frames from separate cameras) impossible.

The syncer currently alternates between framesets of a single D415 unit, while the previous (and desired) behaviour was receiving single synchronized framesets that contained all frames from both D415 units.

It would be great if Intel could reproduce this regression and - if confirmed - solve this in a next release.

If I am doing something wrong, it would be great to know how to achieve software-based sync for two different D415 devices using the current version of the SDK (2.50).

MartyG-RealSense, 2022-03-01:
If you only need to capture a single frame from all attached D415 cameras simultaneously and export the frames as a PNG image then there is a C++ script shared by a RealSense user at https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468 that can do this and does not make use of syncer.

Conceivably if you needed to capture multiple frames then you could just loop the script round continuously until it is stopped.

roelofvandijk, 2022-03-07:
Hello @MartyG-RealSense, it would be great if this is either confirmed as a regression with a forthcoming fix, or if you can confirm that this is the intended behavior for the syncer now (no support for multiple devices anymore).

MartyG-RealSense, 2022-03-07:
A number of changes have been made to syncer inbetween SDKs 2.48.0 and 2.50.0.

**2.48.0**

[image]

[image]

**2.49.0**

[image]

**2.50**

[image]

[image]

***********

It is unlikely that multiple device support in syncer would deliberately be removed.  If other RealSense users can identify a potential bug when using syncer with multiple cameras and resolve it, I would encourage them to submit the fix as a Pull Request so that it can be considered for merging into the SDK.

https://github.com/IntelRealSense/librealsense/pulls

closed 2022-03-19 by MartyG-RealSense