realsenseai/librealsense #10259: rs2::align buggy behavior

ilyak93, 2022-02-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D465                                                       | 
| Operating System & Version | Win 10                                                    | 
| Platform                                 | PC                                                           |
| SDK Version                           | 2.0                                                          | 
| Language                               |  C++                                                      | 

### Issue Description
Hi, Im dealing with a very strange and bad behavior making alignment of depth and color. 

I looked at the example of "rs-align.cpp" in the batch of examples provided by Real Sense, and it is very simple to use as it seems. Without it everything works great each time I run it, but with a few code lines of creating and align object and process with it the frameset, I got the bad behavior.

My code is performed during each callback which align the frames and then saves them to a mapped_file:

```
public:
    RSCallback(char* cmfd_ptr, char* dmfd_ptr, vector<long long int>& ccdts,
               std::atomic<volatile int>& idxc, int& idxd, std::mutex& mx,
               boost::shared_mutex& smx) : idx_color(idxc), idx_depth(idxd),
               color_depth_ts(ccdts), mux(mx), shared_mux(smx){
        *color_mfd_ptr = cmfd_ptr;
        *depth_mfd_ptr = dmfd_ptr;

    }
    // This operator overloading enables calling
    // operator function () on objects of increment
    void operator () (const rs2::frame &frame) {
        boost::shared_lock<boost::shared_mutex> shared_lock(this->shared_mux);
        std::lock_guard<std::mutex> lock(this->mux);
        if (rs2::frameset fs = frame.as<rs2::frameset>()) {
-------------------------------------------------------------------------------------------------
            rs2::align align(RS2_STREAM_COLOR); // Without this line
            auto fs_aligned = align.process(fs); // and this line everything works great
-------------------------------------------------------------------------------------------------            
            // With callbacks, all synchronized stream will arrive in a single frameset
            const std::chrono::time_point<std::chrono::steady_clock> now =
                    high_resolution_clock::now();
            long long int loc_ts =
                    std::chrono::duration_cast<std::chrono::nanoseconds>(
                                            now.time_since_epoch()).count();
            color_depth_ts.push_back(loc_ts);
            for (const rs2::frame f: fs_aligned) {
                auto vf = f.as<rs2::video_frame>();
                if (vf.get_bytes_per_pixel() == 2) {
                    size_t sz = vf.get_data_size();
                    memcpy((void *) ((uint8_t *) (*depth_mfd_ptr) +
                                    idx_depth * sz), vf.get_data(), sz);
                    idx_depth++;
                } else {
                    size_t sz = vf.get_data_size();
+                    memcpy((void *) ((uint8_t *) (*color_mfd_ptr) +
                                    idx_color * sz), vf.get_data(), sz);
                    idx_color.fetch_add(1, std::memory_order_relaxed);
                }
            }
        }
    }
};
```

Nothing special a starting the pipe:
```
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH);
cfg.enable_stream(RS2_STREAM_COLOR);
rs2::pipeline_profile profiles = pipe.start(cfg, rs_callback);
```

The bad behavior is sometimes the frames of depth are 'stuck', I get the same frame of depth for different color frames. I did once get same color frames across all the recording and different depth (the correct stream), but that is rare, and it is mostly the depth stuck, sometimes all of it in one frame (same frame saved across the recording and streaming), sometimes it just half of the stream is stuck, rarely but also happen that everything looks fine, but I wonder what is the reason for such unstable behavior.

I've debugged it and I found that sometimes one of the rs2:frames of the frameset doesn't change, the frame number  is the same, while the other frame numbers are changed each time the callback is called. In the cases I described I see mostly the depth frame is stuck and I underline that it doesn't happen at all without the alignment, which is 2 lines of code, just the align operation. 

I've also tried changing the `RS2_STREAM_COLOR` to `RS2_STREAM_DEPTH` and back, it seemed that it happens with both, but I think with `RS2_STREAM_COLOR` it happened less, but that could be random.

Is there any buggy issues with rs2::align using it in such way in callbacks ?

Thank you in advance.

MartyG-RealSense, 2022-02-23:
Hi @ilyak93  The code of rs-align.cpp has a note that the results of align are **synthetic streams**.

https://github.com/IntelRealSense/librealsense/blob/master/examples/align/rs-align.cpp#L13

A RealSense team member advises at https://github.com/IntelRealSense/librealsense/issues/543#issuecomment-312414926 that callbacks and synthetic streams cannot be combined.

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/3495 who tried using callbacks and align and experienced segfaults shared the workaround script that they created instead.

ilyak93, 2022-02-23:
@MartyG-RealSense,  I wonder if using `software-device` will work after saving all the images as bytes, then loading it into streams through `software-device` and then trying to align through callbacks or even using regular rs stream:

Like talked about here: [#2534](https://github.com/IntelRealSense/librealsense/issues/2534)

MartyG-RealSense, 2022-02-23:
Have you seen the RealSense SDK's official software_device C++ example program that demonstrates aligning depth and color frames, synchronizing them with the **syncer** class to generate a textured pointcloud?

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

ilyak93, 2022-02-23:
@MartyG-RealSense, is there any alignment actually ? I don't see explicit use of rs2::align.

MartyG-RealSense, 2022-02-23:
That particular software_device example is using **map_to** rather than align_to, in order to map depth to color to create a depth-color aligned textured pointcloud.

https://github.com/IntelRealSense/librealsense/blob/master/examples/software-device/rs-software-device.cpp#L198

When generating textured point clouds, the two most common approaches are to (1) use deproject_pixel_to_point and then perform alignment with align_to, or (2) use **pc.map_to** to map color onto depth.

closed 2022-03-03 by ilyak93

MartyG-RealSense, 2022-03-03:
It's great to hear that you found a solution, @ilyak93 - thanks very much for the update  :)