realsenseai/librealsense #10264: D415: 1920x1080 at 6Hz possible?

roelofvandijk, 2022-02-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D415 | 
| Firmware Version                   | 5.13.0.50  | 
| Operating System & Version |  Win11  | 
| Platform                                 | PC |
| SDK Version                            | 2.50  }                          | 
| Language                            |  C++                          | 

I would like to stream the NIR images from a D415 at 1920x1080 and 6Hz, but cannot find a streaming profile that supports this combination among the profiles reported by the device.

The minimum reported FPS for 1920x1080 seems 15, this is also the lowest choice available in the Realsense Viewer.

Is 1920x1080 at 6Hz possible?

MartyG-RealSense, 2022-02-24:
Hi @roelofvandijk  15 FPS is the lowest rate that an infrared stream on a D415 with 1920x1080 resolution is able to use.

A way to achieve custom frame rates that are not officially supported is to program an application that sets the FPS at a supported FPS speed for a particular configuration but only uses every 'nth' frame.  An example of this would be setting the FPS speed to 30 but only using every 5th frame to achieve an FPS of 6 (since 6 x 5 = 30).  This method is discussed at https://github.com/IntelRealSense/librealsense/issues/3169

MartyG-RealSense, 2022-03-03:
Hi @roelofvandijk  Do you require further assistance with this case, please?  Thanks!

roelofvandijk, 2022-03-03:
@MartyG-RealSense No thanks, I'll close this.

closed 2022-03-03 by roelofvandijk