realsenseai/librealsense #10266: Will the kernel changes made by the DKMS package every be upstreamed?

mtandy1, 2022-02-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |   Linux (Ubuntu) | 
| Platform                                 | PC  |

### Issue Description

Hello,

I'm using the D415 on an Ubuntu PC. Some of my colleagues are pursuing the implementation of "Secure Boot" and they say DKMS makes their life very hard.

Do you have any intention of upstreaming the UVC changes made by the dkms package? If so, do you have any sort of schedule for such work?

Thanks very much!

MartyG-RealSense, 2022-02-24:
Hi @mtandy1   The most recent information on this subject that I am aware of is from advice provided by a RealSense team member in November 2019 at https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604 when comparing libuvc-backend / RSUSB-backend versus kernel patching.  They said:

**********

All kernel patches should eventually get upstreamed into Linux or abandoned,  From the user's perspective the main obstacle in accepting the kernel patch model is the deployment scheme that requires certain engineering skills level.  We address this by by wrapping and redistributing the patches with DKMS Debian package, at least for Ubuntu LTS kernels.

*********

My interpretation of the above advice would be that changes should get upstreamed at a future date, but there is no specific time schedule that can be provided.  

If your colleagues have difficulties with DKMS and require a patch-based librealsense build then there is also the option to build the SDK from source code with the **V4L2 backend**.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

mtandy1, 2022-03-01:
Disappointing that there isn't a schedule, but thanks anyway!

closed 2022-03-01 by mtandy1