realsenseai/librealsense #10268: SLAM in low-light environment with D435i

enqinn, 2022-02-25:
Hello, is it an advantage to use a led dot pattern projector for lighting and mapping in the dark room?

With Realsense, the objects in the environment are visible with the IR projector, but there is no result in terms of mapping. I would be glad if you share your views on this subject.

MartyG-RealSense, 2022-02-25:
Hi @enqinn  Whether an external LED pattern projector would benefit your particular project may depend on how much noise you have in the image and how far you need to project the dot pattern.  

An external projector has higher power and so can project the pattern across a greater distance range.  Using an LED projector instead of the RealSense camera's built-in laser projector can reduce a type of image noise known as 'laser speckle'.

RealSense cameras can make use of any visible light source when performing depth analysis of objects and surfaces.  An alternative to a pattern projector in a dark environment could therefore be an external IR illuminator, as discussed at https://github.com/IntelRealSense/librealsense/issues/2000

MartyG-RealSense, 2022-02-25:
If your Rugo kit projects light or near-infrared light then its light should be usable by a RealSense 400 Series camera to aid depth analysis, as the RealSense 400 Series stereo depth camera can make use of any visible light.

A 400 Series camera can interfere with nearby non-RealSense products that use IR, though the RealSense camera does not experience interference in return from the other infrared device.  Multiple 400 Series cameras do not interfere with each other.

The RealSense **L515** lidar depth camera uses ToF principles.  It has reached end of life (retired) status and will not receive further updates, though is still in stock in the official RealSense Store at the time of writing this.  The L515 would experience interference however if used with a 400 Series camera.  L515 also has the limitations of being best suited to indoors-only use in a controlled lighting environment without sunlight due to being sensitive to interference from infrared light sources.

https://store.intelrealsense.com/buy-intel-realsense-lidar-camera-l515.html

A user guide for the L515 to help you decide whether the L515 would be suitable for your project is available at the link below.

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

I am not familiar with the quoted Flir and Swissranger products and so cannot offer advice regarding their suitability.  The Flir projectors mentioned appear to only provide illumination and not project a dot pattern though.

enqinn, 2022-02-25:
First of all, thank you for your answers.
Actually, my first goal is to improve the results with the D435i camera and an external IR projector.
Before using the L515 lidar camera, I want to experiment and test with the D435i and an external IR projector and see the results.
Compared to all other products, the L515 will/should give better results in the dark. However, in sunlight, it will give worse results. maybe it won't give any result :)
You said "The L515 would experience interference however if used with a 400 Series camera". I think we will have the same problem using a different stereo camera, right?

MartyG-RealSense, 2022-02-25:
An L515 would be liable to experience interference from a non-RealSense camera that was projecting a visible IR light source. 
 Multiple L515 can also interfere with each other unless hardware sync between the L515 cameras is implemented to negate that interference.

The L515 is able to handle a small amount of sunlight in a scene by using a camera configuration preset called **Low Ambient Light**, as discussed on page 10 of the L515 user guide.

enqinn, 2022-02-25:
I'll use the D435i (that's all I have) and use an IR projector to illuminate the environment and compare the results.
I aim to draw a conclusion in order to be able to SLAM in unlit and low-light environments.
Thanks for your support

MartyG-RealSense, 2022-02-25:
Thanks very much, @enqinn  - good luck!

MartyG-RealSense, 2022-03-03:
Hi @enqinn  Do you require further assistance with this case, please?  Thanks!

enqinn, 2022-03-04:
Hi MartyG,
What would be a downside to providing illumination with IR projectors and not projecting a dot pattern?
Finally, can we use a more powerful external IR projector instead of the IR projector used in the D435i?
So is it possible to disable the IR projector in realsense and use an external IR projector?
Thanks,

MartyG-RealSense, 2022-03-04:
Yes, the projector can be disabled in the RealSense Viewer or in scripting by setting the instruction **RS2_OPTION_EMITTER_ENABLED** to '0'.  The projector will also be disabled if the **Laser Power** option is set to '0'.

Yes, you can use a more powerful external projector with a 400 Series camera.  Intel has a white-paper guide for doing so at the link below.

https://dev.intelrealsense.com/docs/projectors

Using an IR illuminator instead of a pattern projector may be a disadvantage if the objects / surfaces that are being illuminated have low texture detail or no texture, such as doors, walls and desks.  A dot pattern projection will add analyzable texture to such surfaces that would otherwise be difficult for the camera to analyze for depth detail.

enqinn, 2022-03-04:
Thank you, I'll start trying these.
I would like to share the results with @MartyG-RealSense too.
Have a nice day

closed 2022-03-16 by MartyG-RealSense