realsenseai/librealsense #10271: pyralsense2.config.enable_device_from_file cropped .bag file

mhestraathof, 2022-02-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| Language                            |  python                         | 

We recorded videos with color frames arriving at 30 fps and depth and infrared frames at 3 mps to [avoid interference](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup?_ga=2.91332188.75580210.1645782479-182057765.1637068965). Because the size of the .bag files increases rapidly, we tried to remove the color frames that did not arrive immediately after a depth or infrared frame. The script is based on an example of [creating a cropped bag file](http://wiki.ros.org/rosbag/Cookbook#Create_a_cropped_bagfile) in the rosbag cookbook:

```
import rosbag
import os, shutil
... 
orig = file.replace('.bag','.old')
shutil.move(file, orig)
with rosbag.Bag(file,'w') as outbag:
    with rosbag.Bag(orig,'r') as inbag:
        for topic, msg, t in inbag.read_messages():
            # Save messages with image data depth frames
            if topic == '/device_0/sensor_0/Depth_0/image/data':
                outbag.write(topic, msg, t)
                ...
os.remove(orig)
```
 
After running this script, the new .bag file cannot be opened with the rs-convert.exe tool or with pyralsense2.config.enable_device_from_file('file.bag') anymore. However, it can be opened using the 'Load Recorded Sequence' function of the Intel RealSense Viewer.
 
When the script writes each message with outbag.write (not removing color frames or color metadata), the new .bag file will be the same size, but it won't open either. We also tried using [threading Lock acquire-release](https://answers.ros.org/question/281279/high-rates-of-corrupt-bags-with-rosbag-python-api/) around all bag writes, but it gave the same result.
 
Does anyone know what the problem is with the cropped .bag files and why the Intel RealSense Viewer can open them but the rs-convert.exe tool and pyralsense2.config.enable_device_from_file('file.bag') cannot? How can we fix the recorded files or otherwise prevent this from happening in the future?

MartyG-RealSense, 2022-02-28:
Hi @mhestraathof   Using ROS website information as the basis to create a cropped bag file with the RealSense SDK may not work.  This is because although the SDK's bag format is based on the ROS rosbag standard it has some differences that may result in incompatibility, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

As the L515 supports color at a minimum of 6 FPS (unlike depth and infrared that only has 30 FPS available), an easier solution for reducing bag size may be to set color to 6 FPS instead of using 30 and selecting certain frames.

[image]

May I ask if you are using one L515 camera or multiple L515.  A single L515 will not experience IR interference unless there is another IR light source nearby.  This could be an artificial IR emitting device or natural sunlight in the scene.  If the interference is from a small amount of sunlight then setting the **Low Ambient Light** L515 camera configuration preset may help to deal with it.

Another way to reduce bag size would be to record a bag using the **Z16H** compressed depth format - as described at https://github.com/IntelRealSense/librealsense/issues/9295#issuecomment-874045003 - or if you are recording with the RealSense Viewer then you could go to the Viewer's **Settings** interface and set bag files to always be compressed in the **Playback & Record** section of the Settings.

[image]

You can also create a custom frame rate by setting a script to only use every 'nth' frame.  For example, setting the stream at 30 FPS and then only using every 5th frame to achieve a frame rate of 6 FPS (or every 10th frame to achieve 3 FPS).  A Python reference for doing this is at https://github.com/IntelRealSense/librealsense/issues/3169

mhestraathof, 2022-03-04:
Hi @MartyG-RealSense, thank you for your reply. We are using two L515 cameras, so there is IR interference. We will try to open the cropped files another way and try recording with the viewer from now on.

MartyG-RealSense, 2022-03-04:
Okay, thanks very much @mhestraathof for the update!

MartyG-RealSense, 2022-03-10:
Hi @mhestraathof  Do you have an update about this case that you can provide, please?  Thanks!

mhestraathof, 2022-03-14:
Hi @MartyG-RealSense, I don't have an update yet. If we find a workaround to open the bag files, I'll post it here.

MartyG-RealSense, 2022-03-14:
Thanks very much @mhestraathof for the update.  Good luck!

MartyG-RealSense, 2022-03-27:
Hi @mhestraathof  Do you have any progress with this case to report regarding a workaround to open the bag files, please?  Thanks!

mhestraathof, 2022-03-28:
Hi @MartyG-RealSense. Unfortunately we don't have an update. We decided to make new recordings and move on, so we stopped trying to open the old files. If we find a solution in the future, I'll post it.

MartyG-RealSense, 2022-03-28:
Thanks very much @mhestraathof for the update.  As you have moved on to new recordings, I will close this case.  Thanks again!

closed 2022-03-28 by MartyG-RealSense