realsenseai/librealsense #10272: Very low accuracy Depth result.

Thien223, 2022-02-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version | Win 10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  2.49.0.0 (runtime version: v2.0.50727)                          | 
| Language                            |  C#                          | 
| Segment			|  others | 

### Issue Description
I'm trying to generate high accuracy Depth camera frame to use in Deep Learning object detection (I want to measure the length, width of object, so Depth camera is required)

But my Depth frame is not accurate. It's display a straight line (same depth, i guess) even on not-a-line object border.

[image]

Here is how I filter the frame: 
```
block = new CustomProcessingBlock((f, src) =>
                {
                    // We create a FrameReleaser object that would track
                    // all newly allocated .NET frames, and ensure deterministic finalization
                    // at the end of scope.
                    using (var releaser = new FramesReleaser())
                    {
                        foreach (ProcessingBlock p in blocks)
                            f = p.Process(f).DisposeWith(releaser);

                        // First make the frames spatially aligned
                        f = f.ApplyFilter(alignToColor).DisposeWith(releaser);

                        //// To make sure far-away objects are filtered proportionally
                        //// we try to switch to disparity domain
                        f = f.ApplyFilter(depth2Disparity).DisposeWith(releaser);

                        //// Decimation will reduce the resultion of the depth image,
                        //// closing small holes and speeding-up the algorithm
                        f = f.ApplyFilter(decimationFilter).DisposeWith(releaser);

                        f = f.ApplyFilter(thresholdFilter).DisposeWith(releaser);

                        // Apply spatial filtering
                        f = f.ApplyFilter(spatialFilter).DisposeWith(releaser);
                        f = f.ApplyFilter(holeFillingFilter).DisposeWith(releaser);

                        // Apply temporal filtering
                        f = f.ApplyFilter(temporalFilter).DisposeWith(releaser);

                        //// If we are in disparity domain, switch back to depth
                        f = f.ApplyFilter(disparity2Depth).DisposeWith(releaser);

                        f = f.ApplyFilter(colorizer).DisposeWith(releaser);

                        var frames = f.As<FrameSet>().DisposeWith(releaser);

                        var colorFrame = frames[Stream.Color, Format.Rgb8].DisposeWith(releaser);
                        var colorizedDepth = frames[Stream.Depth, Format.Rgb8].DisposeWith(releaser);

                        CurrentColorFrame = colorFrame;
                        float distance = DepthFrameCallback(frames.DepthFrame.DisposeWith(releaser));
                        GetCurrentDistanceCallback?.Invoke(CenterDistCallback(frames.DepthFrame.DisposeWith(releaser)));

                        // Combine the frames into a single result
                        var res = src.AllocateCompositeFrame(colorizedDepth, colorFrame).DisposeWith(releaser);
                        // Send it to the next processing stage
                        src.FrameReady(res);
                    }
                });
```

I'm newbie in this area, so I don't no where to check and debug this. Please help. 
Thank you so much.

MartyG-RealSense, 2022-02-28:
Hi @Thien223  Performing depth to color alignment correctly in C# has traditionally been more difficult than in C++ and Python.  https://github.com/IntelRealSense/librealsense/issues/9765#issuecomment-919249014 has a C# aligment method that worked for the RealSense user that shared it.

Intel also recommend performing alignment **after** post-processing, whilst your script performs it before the list of filters is applied.

There is also a recommended order in which filters are applied, described in the Intel post-processing documentation link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#using-filters-in-application-code

Your order is: Depth2Disparity, Decimation, Threshold, Spatial, Hole-Filling, Temporal, Disparity2Depth

Intel's recommendation is:

**Decimation, Depth2Disparity, Spatial, Temporal, Disparity2Depth, Hole-Filling, Filtered Depth**

I would assume that the last one in the list, Filtered Depth, refers to the Threshold filter.

Thien223, 2022-03-01:
@MartyG-RealSense  thank you so much.

I have tried change the filtering order, the image (depth image) is better a bit, but still very bad.

It looks like the depth color is shifted to the right and bottom direction.
[image]

My problem is not about distance threshold, I have not tried the https://github.com/IntelRealSense/librealsense/issues/9765#issuecomment-919249014 Comment jet. If there is no luck, I will try it latter.

Thien223, 2022-03-02:
@MartyG-RealSense  Thank for your helps.

closed 2022-03-02 by Thien223