realsenseai/librealsense #10273: Intrinsic parameters of D415 (given by official software != OpenCV checkerboard calibration experiment)

mnauf, 2022-02-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model | D415 |
| Firmware Version | 05.13.00.50 |
| Operating System & Version | Win10 |
| Kernel Version (Linux Only) | N/A |
| Platform | PC (Lenovo Legion 5i |
| SDK Version | 2.50.0 |
| Language | Python |
| Segment | not sure what it means |

### Issue Description
If I make a camera matrix using intrinsic parameters given by Intel Real Sense software, I get:
```
[[952.828 0. 646.699 ]
[0. 952.828 342.637 ]
[0.            0.        1.      ]] 
```
However, when I try to find camera matrix using the OpenCV calibration checkerboard pattern, I get the following after taking 13 pictures from different angles:
```
[[915.63599724   0.         618.21362605]
 [  0.         916.92786672 348.28458816]
 [  0.           0.           1.        ]]
```
Why is there a difference? Please note that my D415 is already well-calibrated and doing on-chip calibration against the textured wall didn't change its focal length and px and py shown in the RealSense software

MartyG-RealSense, 2022-02-28:
Hi @mnauf  I would recommend relying on the official intrinsics provided by SDK tools such as **rs-enumerate-devices** if possible.  If you do need to perform checkerboard calibration in OpenCV to obtain intrinsics from multiple image captures though, a discussion at the link below may provide useful insights.  

https://support.intelrealsense.com/hc/en-us/community/posts/360050772254-I-am-struggling-in-calibrating-realsense-rgb-camera-intrinsic-parameters

At the end of the case, a staff member of the **Calib.io** software offers advice about why incorrect results may be provided.

https://support.intelrealsense.com/hc/en-us/community/posts/360050772254/comments/360013452534

mnauf, 2022-02-28:
Hello, I am not sure what you mean by `rs-enumerate-devices`. By official software, I meant the attached screenshot
[image]
.

mnauf, 2022-02-28:
Hello, I am not sure what you mean by `rs-enumerate-devices`. By official software, I meant the attached screenshot 
[image]

MartyG-RealSense, 2022-02-28:
The RealSense SDK's 'rs-enumerate-devices' tool can provide a full listing of the intrinsics and extrinsics of a camera.  If you have installed the full RealSense SDK on Windows using the automated installer file **Intel.RealSense.SD-WIN10** then you can launch a pre-made version of the tool in calibration information mode using instructions at https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-615365409

mnauf, 2022-03-03:
I will use `rs-enumerate-devices`. Thanks. You may close this issue now

closed 2022-03-03 by MartyG-RealSense