realsenseai/librealsense #10274: Lens specifications for d435 color camera

snowface3, 2022-03-01:
Hi,
Can someone tell me the lens attached to the D435i color camera?
I understand it is an OmniVision OV2740 camera. I would like to find alternate lenses for this camera.

Thanks

MartyG-RealSense, 2022-03-01:
Hi @snowface3  The RGB sensor component of the RealSense D435i is the OmniVision OV2740.  Whilst there has been a case (in the link below) where the lenses of a D435 were removed and the camera recalibrated, it is absolutely not recommended as the likely outcome will be the breaking of the camera.

https://community.intel.com/t5/Items-with-no-label/Realsense-Camera-D435-with-different-imaging-lenses/td-p/592876

Intel make this point about modification here:

https://www.intel.com/content/www/us/en/support/articles/000035877/emerging-technologies.html

You can though change the properties of the existing sensor by purchasing an **optical filter** product and applying it over the top of the lenses on the outside of the camera.  Intel provide a white-paper document on this subject.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

You can also sync a RealSense camera with a separate non-RealSense external RGB high definition camera, such as a 4K resolution one.  This could be achieved using camera metadata - as described at https://github.com/IntelRealSense/librealsense/issues/2186 - or with **hardware sync**, as described in point 3 of the section of Intel's **External Synchronization** (Genlock) paper linked to below.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-4-genlock-usages

MartyG-RealSense, 2022-03-08:
Hi @snowface3  Do you require further assistance with this case, please?  Thanks!

snowface3, 2022-03-10:
Hi, thank you for the response. I do not need further assistance. Thank you for your support to the community.

closed 2022-03-10 by snowface3

MartyG-RealSense, 2022-03-10:
You are very welcome  :)  Thanks very much for the update!