realsenseai/librealsense #10275: Get distance from a static depth image

mnauf, 2022-03-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model | D415 |
| Firmware Version | 05.13.00.50 |
| Operating System & Version | Win10 |
| Kernel Version (Linux Only) | N/A |
| Platform | PC (Lenovo Legion 5i |
| SDK Version | 2.50.0 |
| Language | Python |
| Segment | not sure what it means |

### Issue Description
How to get distance, given that I have a depth image from D415 saved in my machine. Please note, that I can't do `Depth = depth_frame.get_distance(int(x),int(y))` here, since it is not a Livestream. I know the `depth scale = 0.0010000000474974513`.

mnauf, 2022-03-01:
Does `depth_image[int(cordinate[1])][int(cordinate[0])] * depth_scale` give us distance in meters?

mnauf, 2022-03-01:
But doing so gives me `distance = 0.216 meters`, even though the object was at least 0.5 meters away

mnauf, 2022-03-01:
This is what depth image looks like and this is how I read it: 
`cv2.imread(FILENAME, flags=cv2.IMREAD_ANYDEPTH)`
[image]

mnauf, 2022-03-01:
When I multiply this `depth_scale` with a live depth image in example `align-depth2color.py`, it gives the correct distance, because depth_image values there aren't in between `0 to 255`

mnauf, 2022-03-01:
I had saved the depth_image by 
`cv2.imwrite(FILENAME, depth_image)`

MartyG-RealSense, 2022-03-01:
Hi @mnauf  If you are not using get_distance then an alternative method of obtaining the real-world distance in meters is to multiply the raw pixel depth value **uint16_t** by the depth unit scale (which is **0.001** by default on the 400 Series camera models.  There is a discussion about this subject at https://github.com/IntelRealSense/librealsense/issues/7264

mnauf, 2022-03-03:
@MartyG-RealSense Thanks. The issue was, my saved depth image values were in uint8, so I dumped them again using uint16, and then everything worked fine. You may close this issue now.

MartyG-RealSense, 2022-03-03:
It's excellent to hear that you achieved a solution, @mnauf - thanks very much for the update!  As you suggested, I will close this case.  Thanks again!

closed 2022-03-03 by MartyG-RealSense