realsenseai/librealsense #10276: What will realsense viewer do when I export the 3D mesh?

CodeLHY, 2022-03-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        |D435i                                                       | 
| Firmware Version                    |05.13.00.50                                              | 
| Operating System & Version  |Win 10                                                     | 
| Kernel Version (Linux Only)    | xxx                                                           | 
| Platform                                  | PC                                                           |
| SDK Version                            | 2.50.0                                                      | 
| Language                                |  opencv/python                                      | 
| Segment			         | Robot                                                      | 

### Issue Description
I recorded a .bag file, and tried to use pyrealsense2 to parse the .bag file to get the RGB and also the depth map. However, when I tried to visualize the depth map in the point cloud format, I found that the objects are not vertical but oblique, I am not sure what will cause this kind of problem. At the same time, I found that, the mesh exported through the GUI of RealSenseViewer looks good. So I wonder what will RealSenseViewer do when I click the "export" button.
[image]

[image]

[image]

To help you find out the problem, I paste the main code here.
```python
"""
This file contains the tools to parse the .bag file, to get the original data.
Parser_Bag is a class to process a specific .bag file.
"""
from operator import index
from pdb import post_mortem
import pyrealsense2 as rs
import open3d as o3d
import numpy as np
import cv2
from copy import deepcopy
class Parser_Bag:
    def __init__(self, file_bag, post_proc = False, post_proc_dec=False, post_proc_spa=False, post_proc_tem=False, post_proc_hol=False) -> None:
        config   = rs.config()    # construct the config for realsense.
        pipeline = rs.pipeline()  # construct a pipeline
        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 15)   # Enable a device stream explicitly, with selected stream parameters.
        config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 15)
        rs.config.enable_device_from_file(config, file_bag, repeat_playback=False)  # specify the device as the recorded device from a file.  https://github.com/IntelRealSense/librealsense/issues/3147#issuecomment-458145775
        profile = pipeline.start(config)  # start the stream of color and depth
        # set as play back
        device   = profile.get_device()
        playback = device.as_playback()
        playback.set_real_time(False)
        # set align
        align_to = rs.stream.color
        align    = rs.align(align_to)
        
        depth_sensor      = device.first_depth_sensor()  # get the first depth sensor
        self.depth_scale  = depth_sensor.get_depth_scale()  # get the depth scale of the sensor

        index_frame = 0
        self.post_proc = post_proc
        self.frames_rgb = []
        self.frames_dpt = []
        self._get_filters()
        self.intrinsic    = rs.video_stream_profile(pipeline.get_active_profile().get_stream(rs.stream.depth)).get_intrinsics()
        self.cx = self.intrinsic.ppx
        self.cy = self.intrinsic.ppy
[... 76 lines cut ...]
        pt1 = (xmap.flatten()[:, np.newaxis] - self.cy) * pt2 / self.fy
        cloud = np.concatenate((pt0, pt1, pt2), axis=1)
        return cloud
    def save_cloud(self, cloud, path_save, color=None):
        '''
        Description: save the point cloud

        Args:
            cloud: the point cloud. np.array. [N, 3]
            color: the color of each point. np.array. [N, 3], None--> all white

        '''
        print("WARNING! save_cloud")
        pc = o3d.geometry.PointCloud()
        if color is None:
            N, _  = cloud.shape
            color = np.array([[1,1,1]]).repeat(N, 0)
        pc.points = o3d.utility.Vector3dVector(cloud)
        pc.colors = o3d.utility.Vector3dVector(color)
        o3d.io.write_point_cloud(path_save, pc)
if __name__ == "__main__":
    parser_bag = Parser_Bag("../realsense_data/20220301_174251.bag", post_proc=False, post_proc_dec=False, post_proc_spa=False, post_proc_tem=False, post_proc_hol=True)
    from ipdb import set_trace; set_trace()
    parser_bag.visualize_frame(525)
```

MartyG-RealSense, 2022-03-01:
Hi @CodeLHY  It looks as though you are applying depth to color alignment with **align_to** before the post-processing filters are applied.  Intel recommends performing alignment **after** post-processing to help to avoid distortion effects such as aliasing (jagged lines).  Could you try moving alignment to after the filter list to see whether the image improves, please?

CodeLHY, 2022-03-01:
> ```python
> post_proc
> ```

In fact, the post-processing flag is set as False. 
```python
parser_bag = Parser_Bag("../realsense_data/20220301_174251.bag", post_proc=False, .......
```

MartyG-RealSense, 2022-03-01:
Thank you for the information.  

Another difference between the RealSense Viewer and your own script may be that you are using the **align_to** method to perform depth to color alignment.  The Viewer is likely using the alternative **pc.calculate** alignment method to map RGB data onto depth points directly, as demonstrated by the SDK's C++ 'rs-pointcloud' example program.

https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

CodeLHY, 2022-03-01:
You are right, when I align the color to the depth map, the strange phenomenon disappears immediately.
```python
align_to = rs.stream.depth
align    = rs.align(align_to)
```
By the way, where can I find the script that describes what Realsense Viewer does when I click "export" button exactly? Maybe this will help me to learn how to process the raw data captured by the RealSense Camera.  Thank you for your prompt reply very much!

MartyG-RealSense, 2022-03-01:
The Viewer is making use of the **save_to_ply** instruction, which provides a range of options for customizing ply export.

https://github.com/IntelRealSense/librealsense/blob/master/common/viewer.cpp#L111-L114
https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1save__to__ply.html

Using **save_to_ply** in Python has a history of being a problem though in regard to saving RGB color to ply.  There is an alternative instruction called **export_to_ply** that has been shown to be able to save color to ply, though without normals.  https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293 provides a Python example of saving such a color ply.

closed 2022-03-14 by MartyG-RealSense