realsenseai/librealsense #10277: Unable to capture unrectified IR

shphilippe, 2022-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410  | 
| Firmware Version                   | 5.12.07.150 | 
| Operating System & Version |  Ubuntu 18.04.5  | 
| Kernel Version (Linux Only)    |  4.15.0-122-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0-0~realsense0.6128    | 
| Language                            | python                          | 
| Segment			|  robot                          | 

### Issue Description
The following python code: 
```
import pyrealsense2 as rs
import time

pipeline = rs.pipeline()
config = rs.config()

config.enable_stream(rs.stream.infrared, 1, 1920,1080, rs.format.y16, 15)
config.enable_stream(rs.stream.infrared, 2, 1920,1080, rs.format.y16, 15)

config.enable_record_to_file("out.bag")

pipeline.start(config)

time.sleep(10)
pipeline.stop()
```
Doesn't save anything.  The debug log of the library is the following:
```
 01/03 18:10:30,599 DEBUG [139838103299904] (context.cpp:109) Librealsense VERSION: 2.50.0
 01/03 18:10:30,600 INFO [139838103299904] (backend-v4l2.cpp:570) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:2/device/../../../../../../../../../../
 01/03 18:10:30,882 INFO [139838103299904] (backend-v4l2.cpp:570) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:2/device/../../../../../../../../../../
 01/03 18:10:30,883 DEBUG [139838103299904] (ds5-factory.cpp:1142) try_fetch_usb_device(...) failed.
 01/03 18:10:30,883 INFO [139838103299904] (context.cpp:382) Found 1 RealSense devices (mask 0xfe)
 01/03 18:10:30,894 INFO [139838103299904] (backend-v4l2.cpp:570) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:2/device/../../../../../../../../../../
 01/03 18:10:30,895 DEBUG [139838103299904] (ds5-factory.cpp:1142) try_fetch_usb_device(...) failed.
 01/03 18:10:30,895 INFO [139838103299904] (context.cpp:382) Found 1 RealSense devices (mask 0xfe)
 01/03 18:10:30,896 INFO [139838103299904] (backend-v4l2.cpp:570) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:2/device/../../../../../../../../../../
 01/03 18:10:30,898 INFO [139838103299904] (backend-v4l2.cpp:570) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:2/device/../../../../../../../../../../
 01/03 18:10:30,900 INFO [139838103299904] (ros_writer.cpp:24) Compression while record is set to ON
 01/03 18:10:30,900 DEBUG [139838103299904] (ros_writer.h:67) Recorded: "/file_version" . TS: 0
 01/03 18:10:30,900 DEBUG [139838103299904] (record_sensor.cpp:20) Created record_sensor
 01/03 18:10:30,900 DEBUG [139838103299904] (sensor.cpp:106) Registered token #0 to "on_before_streaming_changes"
 01/03 18:10:30,900 DEBUG [139838103299904] (record_sensor.cpp:56) Hooked to real sense
 01/03 18:10:30,900 DEBUG [139838103299904] (record_device.cpp:32) Created record_device
 01/03 18:10:30,900 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 16-bit Depth
 01/03 18:10:30,900 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 16-bit Depth
 01/03 18:10:30,900 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 8-bit Greyscale
 01/03 18:10:30,900 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format UYVY 4:2:2
 01/03 18:10:30,900 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 8-bit Greyscale
 01/03 18:10:30,900 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format Interleaved 8-bit Greyscale
 01/03 18:10:30,900 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format Interleaved 12-bit Greyscale
 01/03 18:10:30,909 INFO [139838103299904] (sensor.cpp:1474) Request: Y16 Infrared, Y16 Infrared, 
Resolved to: Y12I Infrared, 
 01/03 18:10:30,909 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 16-bit Depth
 01/03 18:10:30,909 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 16-bit Depth
 01/03 18:10:30,909 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 8-bit Greyscale
 01/03 18:10:30,909 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format UYVY 4:2:2
 01/03 18:10:30,909 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 8-bit Greyscale
 01/03 18:10:30,909 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format Interleaved 8-bit Greyscale
 01/03 18:10:30,909 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format Interleaved 12-bit Greyscale
 01/03 18:10:30,910 INFO [139838103299904] (backend-v4l2.cpp:1599) Video node was successfully configured to Y12I format, fd 11
 01/03 18:10:30,910 INFO [139838103299904] (backend-v4l2.cpp:1601) Trying to configure fourcc Y12I
 01/03 18:10:30,910 INFO [139838103299904] (backend-v4l2.cpp:1775) Metadata node was successfully configured to D4XX format, fd 14
 01/03 18:10:30,915 DEBUG [139838103299904] (global_timestamp_reader.cpp:176) time_diff_keeper::start: _users_count = 1
 01/03 18:10:30,915 DEBUG [139837935961856] (ros_writer.h:67) Recorded: "/device_0/sensor_0/Infrared_1/info" . TS: 0
 01/03 18:10:30,915 DEBUG [139837935961856] (types.cpp:74) recoverable_exception: No intrinsics are available for this stream profile!
 01/03 18:10:30,915 ERROR [139837935961856] (ros_writer.cpp:335) Error trying to get intrinsc data for stream 3, 1
 01/03 18:10:30,915 DEBUG [139837935961856] (ros_writer.h:67) Recorded: "/device_0/sensor_0/Infrared_1/info/camera_info" . TS: 0
 01/03 18:10:30,915 DEBUG [139837935961856] (ros_writer.h:67) Recorded: "/device_0/sensor_0/Infrared_2/info" . TS: 0
 01/03 18:10:30,915 DEBUG [139837935961856] (types.cpp:74) recoverable_exception: No intrinsics are available for this stream profile!
 01/03 18:10:30,915 ERROR [139837935961856] (ros_writer.cpp:335) Error trying to get intrinsc data for stream 3, 2
 01/03 18:10:30,915 DEBUG [139837935961856] (ros_writer.h:67) Recorded: "/device_0/sensor_0/Infrared_2/info/camera_info" . TS: 0
 01/03 18:10:35,922 WARNING [139837902681856] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds
 01/03 18:10:35,922 INFO [139837902681856] (types.h:898) Frames didn't arrived within 5 seconds
 01/03 18:10:35,922 DEBUG [139837935961856] (ros_writer.h:67) Recorded: "/device_0/sensor_0/notification/Frames Timeout" . TS: 0
 01/03 18:10:40,922 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 16-bit Depth
 01/03 18:10:40,922 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 16-bit Depth
 01/03 18:10:40,922 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 8-bit Greyscale
 01/03 18:10:40,922 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format UYVY 4:2:2
 01/03 18:10:40,922 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format 8-bit Greyscale
 01/03 18:10:40,922 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format Interleaved 8-bit Greyscale
 01/03 18:10:40,922 DEBUG [139838103299904] (backend-v4l2.cpp:1393) Recognized pixel-format Interleaved 12-bit Greyscale
 01/03 18:10:40,922 INFO [139837902681856] (backend-v4l2.cpp:959) V4L stream is closed
 01/03 18:10:40,926 DEBUG [139838103299904] (global_timestamp_reader.cpp:187) time_diff_keeper::stop: _users_count = 0
 01/03 18:10:40,926 DEBUG [139838103299904] (global_timestamp_reader.cpp:190) time_diff_keeper::stop: stop object.
 01/03 18:10:40,926 DEBUG [139838043727616] (global_timestamp_reader.cpp:254) Notification: time_diff_keeper polling loop is being shut-down
 01/03 18:10:40,997 DEBUG [139838103299904] (record_sensor.cpp:29) Destructed record_sensor
 01/03 18:10:40,998 DEBUG [139837944354560] (error-handling.cpp:87) Notification polling loop is being shut-down
```

What can I do to save the unrectified images ?

shphilippe, 2022-03-02:
I'm using a laptop PC, Lenovo Thinkpad T480. Here is lspci of it:
```
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 08)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.6 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #7 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1d.2 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #11 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point LPC Controller/eSPI Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (4) I219-V (rev 21)
03:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)
04:00.0 PCI bridge: Intel Corporation JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016] (rev 01)
05:00.0 PCI bridge: Intel Corporation JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016] (rev 01)
05:01.0 PCI bridge: Intel Corporation JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016] (rev 01)
05:02.0 PCI bridge: Intel Corporation JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016] (rev 01)
06:00.0 System peripheral: Intel Corporation JHL6240 Thunderbolt 3 NHI (Low Power) [Alpine Ridge LP 2016] (rev 01)
3c:00.0 USB controller: Intel Corporation JHL6240 Thunderbolt 3 USB 3.1 Controller (Low Power) [Alpine Ridge LP 2016] (rev 01)
3d:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983
```

Here is the lsusb of the camera confirming it's in USB3:
```
Bus 004 Device 070: ID 8086:0ad2 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
  idProduct          0x0ad2 
  bcdDevice           50.c7
  iManufacturer           1 Intel(R) RealSense(TM) Depth Camera 410 
  iProduct                2 Intel(R) RealSense(TM) Depth Camera 410 
  iSerial                 3 120623020606
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x09d8
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              440mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               5 Intel(R) RealSense(TM) Depth Camera 410  Depth
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
[... 1251 lines cut ...]
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  Platform Device Capability:
    bLength                28
    bDescriptorType        16
    bDevCapabilityType      5
    bReserved               0
    PlatformCapabilityUUID    {d8dd60df-4589-4cc7-9cd2-659d9e648a9f}
    CapabilityData[0]    0x00
    CapabilityData[1]    0x00
    CapabilityData[2]    0x03
    CapabilityData[3]    0x06
    CapabilityData[4]    0x44
    CapabilityData[5]    0x02
    CapabilityData[6]    0x20
    CapabilityData[7]    0x00
Device Status:     0x0001
  Self Powered
```
The camera has the Advanced Mode enabled in realsense-viewer and is listed as USB 3.2

When I use wait_for_frames() I'm able to capture raw unrectified images. And the API is not waiting 10s to deliver them it's much faster (less that 1 second). Why so ?

MartyG-RealSense, 2022-03-03:
Frames arrive very fast even when using wait_for_frames() to block delivery until a complete frame is available.  A sleep instruction though is telling the CPU not to do anything until the defined time period has elapsed.  

The link below leads to the official SDK documentation for the wait_for_frames() instruction.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html#ac77cdd3d2f4ce1e85ea30d475743cec8

shphilippe, 2022-03-03:
Thanks, closing as I have a working system now.

MartyG-RealSense, 2022-03-03:
That's great news, @shphilippe - thanks very much for the update!