realsenseai/librealsense #10281: How to trigger the camera by software manually, rather than trigger continuously. eg: trigger it by an API.

ping28198, 2022-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | () | 
| Operating System & Version |   Linux (Ubuntu 1804) | 
| Kernel Version (Linux Only)    |  (any)                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  { 2.50.0 }                          | 
| Language                            |  {c++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

MartyG-RealSense, 2022-03-02:
Hi @ping28198  It sounds as though you want to capture only a single frame rather than capturing every frame.  Is that correct, please?  If so then the RealSense SDK's **rs-save-to-disk** example program may be helpful.  When this program is run, it captures a single depth frame, exports it as a PNG image file and a CSV file and then closes.

https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk

An alternative approach would be to define a regular period of time to take a capture at (such as once every 5 minutes) and set the CPU to 'sleep' inbetween the moments of capture, as described in https://github.com/IntelRealSense/librealsense/issues/6281

You can also use the RealSense Viewer tool to pause live-streaming and save a frame with the **Snapshot** option, as described in https://github.com/IntelRealSense/librealsense/issues/9288#issuecomment-869015717

A more complex approach to controlling capture precisely is to use **hardware sync** in Genlock mode in conjunction with a RealSense camera so that the camera waits indefinitely until a trigger pulse is received before initiating capture.  This pulse could be transmitted by a **signal generator** device so that the trigger is only sent exactly when you want it to be.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

ping28198, 2022-03-03:
> Hi @ping28198 It sounds as though you want to capture only a single frame rather than capturing every frame. Is that correct, please? If so then the RealSense SDK's **rs-save-to-disk** example program may be helpful. When this program is run, it captures a single depth frame, exports it as a PNG image file and a CSV file and then closes.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk
> 
> An alternative approach would be to define a regular period of time to take a capture at (such as once every 5 minutes) and set the CPU to 'sleep' inbetween the moments of capture, as described in #6281
> 
> You can also use the RealSense Viewer tool to pause live-streaming and save a frame with the **Snapshot** option, as described in [#9288 (comment)](https://github.com/IntelRealSense/librealsense/issues/9288#issuecomment-869015717)
> 
> A more complex approach to controlling capture precisely is to use **hardware sync** in Genlock mode in conjunction with a RealSense camera so that the camera waits indefinitely until a trigger pulse is received before initiating capture. This pulse could be transmitted by a **signal generator** device so that the trigger is only sent exactly when you want it to be.
> 
> https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

Thank you very much, I want the **hardware sync** mode, the doc say: "Color: For the D435/D435i the color imager is rolling shutter and it is not hardware synced to the depth. This is an HW limitation, so the color sensor will NOT be Genlocked, but will instead be free-running at the frame rate specified for that imager." Whether RGB camera is hardware synced to the depth of D455? I know that the D455's RGB camera has global shutter.

MartyG-RealSense, 2022-03-03:
The External Synchronization paper was published before the D455 model with global shutters on the RGB sensor was released.  What the quoted statement is likely referring to is that on the D435 / D435i models the RGB sensor is not integrated onto the camera circuit board like it is on the D415 but is instead attached as a separate component via a cable.  Synchronization between depth and RGB streams **on the same camera** is therefore achieved by 'best match' of frame timestamps, as mentioned by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/1548#issuecomment-389070550

On the D450 camera board that is inside the D455 model, the RGB sensor is integrated on the PCB circuit board, as shown in the image below of the D450 that is extracted from the data sheet document for the 400 Series cameras.

[image]

In regard to whether the RGB and depth sensors on the D455 model support being synced, Intel studied it but no further action was deemed necessary.  My understanding is that this was because the depth-RGB sync was already found by the review process to be performing steadily on D455.

Point 3 of the section of the paper linked to below states that a RealSense camera's depth can be hardware synchronized with a separate RGB camera if the RGB camera is set as the **master** camera, which would make the RealSense camera the **slave** that synchronizes to it.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-4-genlock-usages

closed 2022-03-14 by ping28198