realsenseai/librealsense #10283: About load preset

GHCK57, 2022-03-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.12.14.50) | 
| Operating System & Version |   {Win (10)                                        | 
| Platform                                 | PC  |
| Language                            |  {python }                          | 

### Issue Description

Hi @MartyG-RealSense I hope you are fine, 

I bought a new one D415, as you said it is better for that job. I want to ask a few question to you.

Does realsense camera apply the "preset configuration" while streaming ? or Does it need set before start stream ?

MartyG-RealSense, 2022-03-03:
Hi @GHCK57  My recollection is that if you are setting a preset with scripting then applying it **after** the start is recommended, as described in the code in the Intel guide at the link below.

 https://www.intel.com/content/www/us/en/support/articles/000028416/emerging-technologies/intel-realsense-technology.html

GHCK57, 2022-03-03:
Thanks a lot @MartyG-RealSense 

What is the better way to set "preset conf" 

1-) 

```
preset_range = self.sensor.get_option_range(rs.option.visual_preset)
        #print('preset range:'+str(preset_range))

        for i in range(int(preset_range.max)):
            visulpreset = self.sensor.get_option_value_description(rs.option.visual_preset,i)            
            if visulpreset == "High Density":
                print('%02d: %s'%(i,visulpreset))
                self.sensor.set_option(rs.option.visual_preset, i)
```

or 

```
        Load the settings stored in the JSON file

        """
            
        with open (path_to_settings_file,'r') as file:
            json_text = file.read().strip()

        device = self.pipeline_profile.get_device()
        advanced_mode = rs.rs400_advanced_mode(device)
        advanced_mode.load_json(json_text)
```

MartyG-RealSense, 2022-03-03:
If you are happy to use one of the existing SDK presets then the first method is preferable because it is easier than using a string to load a custom preset that you have created.  

If you need to use a custom preset then loading it via a string with the second method is the appropriate method to use, as you would not be able to access the custom preset using the first method.

GHCK57, 2022-03-03:
thanks for your solutions @MartyG-RealSense

MartyG-RealSense, 2022-03-10:
Hi @GHCK57  Do you require further assistance with this case, please?  Thanks!

GHCK57, 2022-03-10:
Hi @MartyG-RealSense  

No, thanks for your support :)

MartyG-RealSense, 2022-03-10:
You are very welcome  :)  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-03-10 by MartyG-RealSense