realsenseai/librealsense #10284: Skeleton Tracking SDK License order

woodal, 2022-03-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello there!
Last January I bought Skeleton Tracking SDK License.
I want to purchase an additional Skeleton Tracking SDK License or trial version this time,
but the purchase site is not available.
What's the matter?

https://www.intelrealsense.com/skeleton-tracking/
[image]

MartyG-RealSense, 2022-03-03:
Hi @woodal  The **Cubemos Skeleton Tracking SDK** is a product of the Cubemos company, who are responsible for its support as it is not an Intel product.  The Cubemos SDK is no longer available, though downloads still work at the time of writing this (as you mentioned).   I am not aware of the reasons for its withdrawal.  

An alternative skeleton tracking product that is compatible with RealSense cameras is the Nuitrack SDK.

https://nuitrack.com/

woodal, 2022-03-03:
Thanks for your reply.
But just a month ago (late January 22) I bought the license from the Intel website. Has support stopped within a month?
[image]

closed 2022-03-16 by MartyG-RealSense