realsenseai/librealsense #10286: Focal length calibration

shphilippe, 2022-03-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410 | 
| Firmware Version                   | 5.12.07.150 | 
| Operating System & Version |   Ubuntu 18.04.5  | 
| Kernel Version (Linux Only)    |  4.15.0-122-generic   | 
| Platform                                 | PC Lenovo Thinkpad T480 |
| SDK Version                            |  2.50.0-0~realsense0.6128   | 
| Language                            |  Python       | 
| Segment			|  Robot                       | 

### Issue Description
In the last Realsense SDK release not, the following note seems very interesting for my usecase:
"As part of the Self-Calibration refactoring, Focal-Length calibration and Tare Target-based distance calculation were introduced, that require a printed target to be provided ([Target_for_print](https://librealsense.intel.com/Releases/UCAL/Calibration_Target_v1.pdf))."

When I read the self calibration white paper, nothing is describing the focal-length calibration and nothing is describing the calibration target linked in the release note.

Is any additional information available ? What should be the exact size of the calibration target when printed ? what is the optimal distance when performing the calibration ? is there any more info on what exactly this is calibrating:  fx,fy independently or both with the same ratio ? Should we/can we do this calibration in combinaison of other calibration such as the On chip extrinsic calibration or the dynamic calibration ? What is the difference between the tare calibration and the focal length calibration: I suspect one is just adding an offset to the distance, the other is changing the real focal length ? How much is the typical change in the focal length ? (0.1-0.2%, 1-2% etc ...)

Thanks !

MartyG-RealSense, 2022-03-03:
Hi @shphilippe  There is further information in the **API Changes** notes about the focal length calibration that has been introduced in SDK version 2.50.0.

https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2500

A recent question and answer from a Tare expert on the RealSense team at https://github.com/IntelRealSense/librealsense/issues/10213#issuecomment-1030589898 may also be helpful.

I will attempt to cover some of your questions that the Q&A above may not address.

The page dimensions of the print target PDF document for focial length calibration in the 2.50.0 release notes is 11 x 8.5", according to the PDF reader (note that it is displayed in side-on landscape orientation rather than 8.5x11" US Letter)

[image]

As you are using a D410 Depth Module Kit without an RGB sensor then you would not benefit from the RGB calibration that the Dynamic Calibration tool provides.  Using On-Chip, Tare and Focal Length calibration and not using Dynamic Calibration is therefore likely to be the most appropriate approach for you.

In regard to your questions "what exactly this is calibrating" and "what is the difference between the tare calibration and the focal length calibration", I do not currently have information available about those queries.

shphilippe, 2022-03-03:
Based on the source code (common/on-chip-calib.cpp:934), the default target size is: width: 175.0mm; height: 100.0mm. 
But according to realsense-viewer, it's 138mm by 82mm. Which one is better ? can I print a bigger calibration target in order to have a more precise calibration ? Why the region of interest is so small ?
What is the best distance to have the most bias-free calibration ?

MartyG-RealSense, 2022-03-04:
I have not got a D410 to test with.  I tested the focal length calibration option in the Viewer with a D415, D435i and D455 though and in all three cases the default target measurements were width 175 mm and height 100 mm.

[image]

My experience of cases where a custom calibration target image size has been used suggests that it can worsen accuracy and it is best to use the officially provided target images.  

My interpretation of the advice by Intel's Tare expert in https://github.com/IntelRealSense/librealsense/issues/10213#issuecomment-1030589898 is that a 20% sized ROI is the one that provides the most consistency and can be used for all camera distances from the target, but you can multiply the ROI size for every 0.5 meters that the camera is from the target - for example, use 20% ROI at 0.5 m but if the camera is 1.5 m away then you can multiply the 20% ROI size by 3 (because 0.5 m x 3 = 1.5 m). 

 In regard to the best distance: whilst I do not have a precise value to suggest, in general the noise (and thus the depth measuring error) on the image will increase over distance as the camera becomes further away from the observed surface.  The D415 model has around 2x less noise over distance than the D435 / D435i model, and so that D415 error level should apply to the D410 too.

shphilippe, 2022-03-05:
thank you for this detailed answer. The difference in realsense viewer is probably caused by me playing with the parameter and realsense viewer remembering the last value.

closed 2022-03-05 by shphilippe

MartyG-RealSense, 2022-03-05:
You are very welcome, @shphilippe - thanks for the update!