realsenseai/librealsense #10287: realsense L515 frame rate start getting slower and slower

Billccx, 2022-03-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Operating System & Version |  ubunru20.04 | 
| Platform                                 | PC |                 
| Language                            |  C++ | 

### Issue Description
Dear researchers:
I use two realsense L515 to get RGB and depth frames. However, I found that the frame rate start getting slower after running for a while, rather than keeping a constant fps. Why is was so? Here is my code:
```
#include <iostream>
#include <map>
#include <string>
#include <vector>
#include <librealsense2/rs.hpp>
#include <opencv2/opencv.hpp>
#include <chrono>
#include <time.h>

static cv::Mat frame_to_mat(const rs2::frame& f){
    using namespace cv;
    using namespace rs2;

    auto vf = f.as<video_frame>();
    const int w = vf.get_width();
    const int h = vf.get_height();

    if (f.get_profile().format() == RS2_FORMAT_BGR8){
        return Mat(Size(w, h), CV_8UC3, (void*)f.get_data(), Mat::AUTO_STEP);
    }
    else if (f.get_profile().format() == RS2_FORMAT_RGB8){
        auto r_rgb = Mat(Size(w, h), CV_8UC3, (void*)f.get_data(), Mat::AUTO_STEP);
        Mat r_bgr;
        cvtColor(r_rgb, r_bgr, COLOR_RGB2BGR);
        return r_bgr;
    }
    else if (f.get_profile().format() == RS2_FORMAT_Z16){
        return Mat(Size(w, h), CV_16UC1, (void*)f.get_data(), Mat::AUTO_STEP);
    }
    else if (f.get_profile().format() == RS2_FORMAT_Y8){
        return Mat(Size(w, h), CV_8UC1, (void*)f.get_data(), Mat::AUTO_STEP);
    }
    else if (f.get_profile().format() == RS2_FORMAT_DISPARITY32){
        return Mat(Size(w, h), CV_32FC1, (void*)f.get_data(), Mat::AUTO_STEP);
    }

    throw std::runtime_error("Frame format is not supported yet!");
}

void printIntrinsics(rs2_intrinsics intrin){
    std::cout<<"\ncolor intrinsics: ";
    std::cout<<intrin.width<<"  "<<intrin.height<<"  ";
    std::cout<<intrin.ppx<<"  "<<intrin.ppy<<"  ";
    std::cout<<intrin.fx<<"  "<<intrin.fy<<std::endl;
    std::cout<<"coeffs: ";
    for(auto value : intrin.coeffs)
    std::cout<<value<<"  ";
    std::cout<<std::endl;
    std::cout<<"distortion model: "<<(int)intrin.model<<std::endl;
}

int main(){
    rs2::align align_to_color(RS2_STREAM_COLOR);
    rs2::context ctx;
    std::vector<rs2::pipeline> pipelines;
    std::map<std::string, rs2::colorizer> colorizers;
    std::vector<std::string> serials; //相机id
[... 387 lines cut ...]
285--fps:11.1914
286--fps:10.4979
287--fps:8.86886
288--fps:7.33832
289--fps:9.2377
290--fps:10.7698
291--fps:10.9545
292--fps:10.259
293--fps:9.98652
294--fps:9.00471
295--fps:10.174
296--fps:11.1593
297--fps:9.96582
298--fps:10.6744
299--fps:7.03507
300--fps:9.48452
301--fps:11.2987
302--fps:11.5274
303--fps:10.2836
304--fps:10.0254
305--fps:9.945
306--fps:9.57176
```

As you can see, the fps slow down from 20 to 10. Besides that, I noticed that realsenseL515 get _‘fever’_ after running for a while. Is fps slowing down caused by overheating?

MartyG-RealSense, 2022-03-04:
Hi @Billccx  If performance is slowing down over time then there may be a **memory leak**, where the memory of the computer is progressively consumed over time and causes the program's performance to degrade and even freeze / crash.  In Ubuntu you can use a system monitoring tool such as **htop** to monitor memory usage whilst your program is running to see whether there is a leak.

If you are using the RealSense Viewer tool then memory leaks can occur for a small number of RealSense users if the **GLSL** options are enabled.  You can disable GLSL in the Viewer's settings interface using instructions at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023 to see whether it improves the reliability of performance.

In regard to the L515's temperature, if the casing is hot to the touch only a few minutes after streaming begins then this can indicate that there is an overheating problem related to USB (either a glitch on the USB port or a bad USB cable).  The camera's temperatures can be monitored in real-time in the RealSense Viewer under the 'Controls' category of the**L500 Depth Sensor** options.

[image]

Billccx, 2022-03-04:
@MartyG-RealSense 
Thanks a lot for such a quick reply! 
I tried htop to monitor the memory. However it seems that I failed to find the sign of memory leak?
Here is the htop output when I started to run, the fps (not precise) is around 30-40.

[image]

After running for a while, the htop output is as follow, the fps is around 10-15.

[image]

Could you please tell me why it was so? 
I'd be grateful if you would have a look at my source code.
Thanks!

MartyG-RealSense, 2022-03-04:
Thanks very much for testimg memory usage and for your images.  Next, please try going to the **RGB** section of the Viewer's options and disabling an option called **Auto-Exposure Priority** (whilst keeping Auto-Exposure enabled).  If Auto-Exposure is enabled and Auto-Exposure Priority is disabled then the RealSense SDK should try to enforce a constant FPS rate instead of permitting the FPS to vary.

[image]

Also, did you find that the camera casing was hot to the touch only a few minutes after streaming began, and did you try reversing the connector?

Billccx, 2022-03-04:
@MartyG-RealSense 
Thanks for your advice!
I haved disabled the "Auto-Exposure Priority" while keeping "Auto-Exposure" enabled. The fps in realsense-viewer is very stable. (Hardware fps 30/ viewer fps 17-22)
[image]

Unfortunately, when I try to run my source code, the problem still exist! Even though I have already set Auto-Exposure Priority off in my code:
```
for (auto&& serial : serials){
    rs2::pipeline pipe(ctx);
    rs2::config cfg;
    cfg.enable_device(serial);
    cfg.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 640, 480, rs2_format::RS2_FORMAT_Z16); 
    cfg.enable_stream(rs2_stream::RS2_STREAM_COLOR, 640, 480, rs2_format::RS2_FORMAT_RGB8); 
    auto prof=pipe.start(cfg);
    auto sensors = prof.get_device().query_sensors();
    for(auto item:sensors) std::cout<<item.get_info(RS2_CAMERA_INFO_NAME)<<std::endl;

    auto rgb=prof.get_device().first<rs2::color_sensor>();
    rgb.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
    rgb.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY,0);

    pipelines.emplace_back(pipe);
    // Map from each device's serial number to a different colorizer
    colorizers[serial] = rs2::colorizer();
}
```
the output of my code are as follows:
```
~/code/C++/librealsense-2.50.0/examples/rs_bazel$ ./bazel-bin/src4/fps

2
f1231209
f1230584
L500 Depth Sensor
RGB Camera
Motion Module
L500 Depth Sensor
RGB Camera
Motion Module
ready to fetch
for this first camera:

color intrinsics: 640  480  331.948  248.697  600.764  600.986
coeffs: 0.171335  -0.528704  -0.00208366  2.20354e-05  0.495262  
distortion model: 4
for this second camera:

color intrinsics: 640  480  333.593  244.954  601.325  601.453
coeffs: 0.157447  -0.49804  -0.00211366  -0.00051181  0.47009  
distortion model: 4
0--fps:8.02835e-05
1--fps:18.9283
2--fps:18.8002
3--fps:21.9819
4--fps:21.8269
5--fps:18.7547
6--fps:21.034
7--fps:23.6206
8--fps:18.766
9--fps:22.0264
10--fps:23.0489
11--fps:23.5322
12--fps:21.4321
13--fps:21.1721
14--fps:23.3046
15--fps:21.901
16--fps:23.1353
17--fps:23.1455
18--fps:22.9505
19--fps:22.2207
[... 204 lines cut ...]

        rs2::video_frame color0=aligned_frames0.get_color_frame();
        rs2::depth_frame depth0=aligned_frames0.get_depth_frame();

        rs2::video_frame color1=aligned_frames1.get_color_frame();
        rs2::depth_frame depth1=aligned_frames1.get_depth_frame();

        cv::Mat color0_mat=frame_to_mat(color0);
        cv::Mat color1_mat=frame_to_mat(color1);

        cv::Mat depth0_mat=frame_to_mat(depth0);
        cv::Mat depth1_mat=frame_to_mat(depth1);

        now=std::chrono::steady_clock::now();

        std::cout<<cnt<< "--fps:"<<
        1000000/(double)(std::chrono::duration_cast<std::chrono::microseconds>(now - start).count()) 
        << std::endl;
        cnt++;
    }

    return 0;
}

```

MartyG-RealSense, 2022-03-04:
I see that you are using multicam code in your project to access two L515 cameras.  It is recommended though to use **poll_for_frames()** instead of wait_for_frames() for an application that is accessing multiple cameras, like the SDK's **rs-multicam** example program does.

Billccx, 2022-03-05:
@MartyG-RealSense 
Thanks for your advice!
I think I have already found what leads to program slowing down. It is `rs2::align`.
It seems that after hundreds of iterations, the time cost on RGB and depth alignment grow larger and larger.
here is my code:
```
#include <iostream>
#include <map>
#include <string>
#include <vector>
#include <librealsense2/rs.hpp>
#include <opencv2/opencv.hpp>
#include <chrono>
#include <time.h>

int main(){
    rs2::align align_to_color(RS2_STREAM_COLOR);
    rs2::context ctx;
    std::vector<rs2::pipeline> pipelines;
    std::map<std::string, rs2::colorizer> colorizers;
    std::vector<std::string> serials;

    auto list = ctx.query_devices(); // Get a snapshot of currently connected devices
    if (list.size() == 0) {
        throw std::runtime_error("No device detected. Is it plugged in?");
    }
    else{
        std::cout<<list.size()<<std::endl;
    }

    for(auto&& item : list){
        std::string serial=item.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
        std::cout<<serial<<std::endl;
        serials.push_back(serial);
    } 

    for (auto&& serial : serials){
        rs2::pipeline pipe(ctx);
        rs2::config cfg;
        cfg.enable_device(serial);
        cfg.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 640, 480, rs2_format::RS2_FORMAT_Z16); 
        cfg.enable_stream(rs2_stream::RS2_STREAM_COLOR, 640, 480, rs2_format::RS2_FORMAT_RGB8); 
        auto prof=pipe.start(cfg);
        auto sensors = prof.get_device().query_sensors();
        for(auto item:sensors) std::cout<<item.get_info(RS2_CAMERA_INFO_NAME)<<std::endl;

        auto rgb=prof.get_device().first<rs2::color_sensor>();
        rgb.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
        rgb.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY,0);

        pipelines.emplace_back(pipe);
        // Map from each device's serial number to a different colorizer
        colorizers[serial] = rs2::colorizer();
    }

    int cnt=0;   
    std::cout<<"ready to fetch"<<std::endl;

    std::chrono::_V2::steady_clock::time_point start,now;
    std::chrono::_V2::steady_clock::time_point r,l;

    while (cnt<10000){
        start=now;
        rs2::frameset fs0,fs1;
        fs0=pipelines[0].wait_for_frames();
        fs1=pipelines[1].wait_for_frames();

        l=std::chrono::steady_clock::now();

        rs2::frameset aligned_frames0 = align_to_color.process(fs0);
[... 331 lines cut ...]
292-RGB depth alignment time cost: 0.073763 seconds       292--fps:13.5216
293-RGB depth alignment time cost: 0.0710003 seconds       293--fps:14.0463
294-RGB depth alignment time cost: 0.0762975 seconds       294--fps:13.0727
295-RGB depth alignment time cost: 0.0769347 seconds       295--fps:12.9685
296-RGB depth alignment time cost: 0.0745512 seconds       296--fps:13.3783
297-RGB depth alignment time cost: 0.0786076 seconds       297--fps:12.6904
298-RGB depth alignment time cost: 0.0734689 seconds       298--fps:13.5788
299-RGB depth alignment time cost: 0.0718114 seconds       299--fps:13.8906
300-RGB depth alignment time cost: 0.0722875 seconds       300--fps:13.7988
301-RGB depth alignment time cost: 0.0743864 seconds       301--fps:13.3774
302-RGB depth alignment time cost: 0.0746394 seconds       302--fps:13.3515
303-RGB depth alignment time cost: 0.0737597 seconds       303--fps:13.5241
304-RGB depth alignment time cost: 0.0699076 seconds       304--fps:14.2694
```
The time cost of the code below grow from 0.03 seconds to 0.07 seconds
```
rs2::frameset aligned_frames0 = align_to_color.process(fs0);
rs2::frameset aligned_frames1 = align_to_color.process(fs1);
``` 
Besides that, I have also noticed that my CPU usage is quite heavy after running for a while.
[image]

What should I do?  I do not understand why '**_RGB-depth alignment_**' operation takes more and more time.

I am looking forward to your reply. Thanks a lot！

MartyG-RealSense, 2022-03-05:
I have not seen your particular approach in your script before.  It looks as though it could be very costly in terms of processing burden.  

A RealSense team member provides a multicam script that uses **pipelines** at https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412350887 that you could compare to your own approach.

closed 2022-03-17 by MartyG-RealSense