realsenseai/librealsense #10288: Does pre-build packages support Ubuntu 20.04 with linux version 5.11.0.43？

hsyzhxy, 2022-03-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D415 }                                       | 
| Firmware Version                   | (05.12.10.00) | 
| Operating System & Version |    Linux (Ubuntu 20.04)                           | 
| Kernel Version (Linux Only)    |  (5.11.0-38-generic)                               | 
| Platform                                 | NUC                                                   ..  |
| SDK Version                           |  { librealsense2.so.2.50 }                        | 
| Language                              |  {C++ }                                                    | 
| Segment			      |  {others }                          | 

 Issue Description：
My system is Ubuntu 20.04,linux version 5.11.0.43   
Camera Model is D415, Firmware Version is 05.12.10.00.
Is it suitable for using pre-build packages？

hsyzhxy, 2022-03-04:
[image]

hsyzhxy, 2022-03-04:
I found the following information， it means linux version 5.11.0.43 is still not supported by Intel® RealSense™ SDK 2.0 (v2.50.0？

[image]

hsyzhxy, 2022-03-04:
It can run but is not stable during  our using.  

Sometimes "rs2_pipeline_wait_for_frames - pipe:0x2326020 :Frame didn't arrive within 5000"  occurred after running normally several hours.  then can't get any device by calling  ctx.query_devices（）even though I reboot the computer。

MartyG-RealSense, 2022-03-04:
Hi @hsyzhxy  Support for kernel 5.11 was added in SDK 2.50.0 via code submitted by a RealSense user, as described at https://github.com/IntelRealSense/librealsense/pull/9727 - the RealSense user who created the submission tested it with Ubuntu 20.04 and **5.11.0-34-generic**

Another RealSense user at https://github.com/IntelRealSense/librealsense/issues/10207#issuecomment-1028836935 who had Ubuntu OS freeze problems also then tried kernel 5.11.0-34-generic and found that it resolved their problem.

If you continue to experience kernel related conflicts then there is the option of building librealsense from source code with the **RSUSB backend** method, which is not dependent on Linux versions or kernel versions and does not require patching.

In regard to pre-built packages, https://github.com/IntelRealSense/librealsense/pull/8787 details that SDK 2.45.0 added support for kernel 5.8 on Ubuntu 20.04 (Focal) but it was not clear whether that only applied to patching a source-code build or DKMS packages too.  Currently at the time of writing this,  the package distribution installation instructions only list support up to kernel 5.4.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

[image]

hsyzhxy, 2022-03-06:
thanks a lot！how to build librealsense from source code with the RSUSB backend method？Is it https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md?

MartyG-RealSense, 2022-03-06:
The RSUSB process of building from source code is similar to the CMake build instructions in the link above but adds the build flag **-DFORCE_RSUSB_BACKEND=true**.  Instructions for building with the RSUSB backend on a PC can be found at https://github.com/IntelRealSense/librealsense/issues/6710#issuecomment-650611442

hsyzhxy, 2022-03-07:
thanks! I have completed Building from source with the RSUSB backend method:
 "cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true"

I'll observe in next two days to see if the problem happen again.

MartyG-RealSense, 2022-03-07:
Thanks very much, @hsyzhxy - I look forward to your next update.  Good luck!

hsyzhxy, 2022-03-09:
@MartyG-RealSense  the erification is still ongoing.
Now I‘m’ uncertain of some items in instructions https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md.
Is it necessary to execute step1-4 as below?  I  executed step1-2， but didn't run step2-3. Is there any effect on it?

[image]

hsyzhxy, 2022-03-10:
I was amazed to find my Linux kernel is now 5.13!   I run “”sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade“ follow 
[image]

now ./scripts/patch-realsense-ubuntu-lts.sh does't support 5.13 yet.

MartyG-RealSense, 2022-03-10:
Thanks very much @hsyzhxy for the update.  As mentioned earlier in this discussion, if a kernel is not yet supported then you can bypass the kernel requirements with an **RSUSB backend** source code build, and if you need a kernel-based build then **5.11.0-34-generic** is currently the most stable known 5.11 kernel to use with the SDK.

MartyG-RealSense, 2022-03-10:
That is correct, when using the RSUSB method you do not need to run the patch script.

hsyzhxy, 2022-03-11:
Unfortunately "rs2_pipeline_wait_for_frames - pipe:0x2326020 :Frame didn't arrive within 5000" occurred again.
I I'm going to downgrade the Linux kernel to 5.4

MartyG-RealSense, 2022-03-11:
With an RSUSB build of librealsense, changing the kernel version should not make a difference as RSUSB bypasses the kernel.

hsyzhxy, 2022-03-15:
I have degraded the kernel version to 5.4，and build with “”cmake ../ -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true" . But I met the following error, how to correct it？
[image]

hsyzhxy, 2022-03-15:
I have degraded the kernel version to 5.8, and build with "cmake ../  -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false", and then make install succeeded。

Does  SDK 2.0 v2.50.0 support  LINUX kernel version 5.8?

MartyG-RealSense, 2022-03-15:
Yes, the SDK supports kernel 5.8 when building from source code.

MartyG-RealSense, 2022-03-21:
Hi @hsyzhxy  Do you require further assistance with this case, please?  Thanks!

hsyzhxy, 2022-03-23:
no other question，thanks！

MartyG-RealSense, 2022-03-23:
Thanks very much @hsyzhxy for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-03-23 by MartyG-RealSense