realsenseai/librealsense #10289: "No device connected" in Blender 2.93.5 (python 3.9.2)

McGavish, 2022-03-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.50.0.3785 | 
| Language                            |  python 3.9.2  | 

### Issue Description

I would like to use Realsense D435i camera inside of Blender 2.93.5 https://www.blender.org/download/lts/2-93/
 I have installed Python wrappers in Blender's site-packages directory using PIP `pip install pyrealsense2` and `import pyrealsense2` command works properly when evoked from Blender so I assume modules are installed correctly.

However when I run the following code (from samples) I get an error - **"No device connected"** in line `pipeline_profile = config.resolve(pipeline_wrapper)`

>import pyrealsense2 as rs
>import numpy as np
>import cv2
>
>pipeline = rs.pipeline()
>config = rs.config()
>
>pipeline_wrapper = rs.pipeline_wrapper(pipeline)
>pipeline_profile = config.resolve(pipeline_wrapper)
>device = pipeline_profile.get_device()
>device_product_line = str(device.get_info(rs.camera_info.product_line))

The same code works totally fine when exectued in Powershell, camera also works fine in Realsense Viewer. 
Could you tell me what might cause this issue here?

MartyG-RealSense, 2022-03-04:
Hi @McGavish  It sounds as though you are trying to access pyrealsense2 using Blender's Python API.  Is that correct, please?

https://docs.blender.org/api/current/info_overview.html

I have not seen a previous case of the Python wrapper being used with Blender in this way.  There are a small number of cases though where **import pyrealsense2 as rs** does not work, but the following import command does:

 **import pyrealsense2.pyrealsense2 as rs**

McGavish, 2022-03-04:
Hi @MartyG-RealSense. Yes, I want to access pyrealsense2 through Blender's Python API.

I've checked `import pyrealsense2.pyrealsense2 as rs` and it imports without errors. It works the same as `import pyrealsense2 as rs` which also imports correctly. However it still fails on line `pipeline_profile = config.resolve(pipeline_wrapper)` with error "**No device connected**".  I haven't found any examples over the internet for using Realsense and Blender but since Blender uses Python and it's possible to install modules I hope it will work fine.

With code below I can access RS RGB camera so Blender has an access to the device. It seems that camera somehow doesn't work with realsense library.

>import cv2
>
>cam = cv2.VideoCapture(2)
>img_counter = 0
>
>while True:
>    ret, frame = cam.read()
>    if not cam.isOpened():
>        raise IOError("Cannot use webcam")
>    elif cam.isOpened():
>       cv2.imshow('frame', frame)
>    if not ret:
>       break
>    if cv2.waitKey(1) == 27: #ESC
>        break
>cam.release()
>cv2.destroyAllWindows()

MartyG-RealSense, 2022-03-04:
As you can access the camera's RGB from Blender's Python API, it does not seem to be a problem related to connecting to a USB device.  The RGB code that you used seems to be ordinary webcam access code though, and RealSense cameras can be used as non-RealSense basic webcams in video-using applications such as Skype, Discord etc.  So it just demonstrates that the camera is accessible over USB through Blender rather than whether it is RealSense-compatible.

I wonder whether the pyrealsense2 wrapper simply does not know how to find Python with a PYTHONPATH when used with Blender's interpreter, compared to when the wrapper is being used with Python 3.  You could try inputting the command **which python** into the Blender interpreter to see what it returns as the Python path.

McGavish, 2022-03-04:
I've checked that using (I hope you meant this one):
>import sys
>print(sys.path)

and it prints:
C:\blender-2.93.5-windows-x64\2.93\scripts\startup
C:\blender-2.93.5-windows-x64\2.93\scripts\modules
C:\blender-2.93.5-windows-x64\python39.zip
C:\blender-2.93.5-windows-x64\2.93\python\DLLs
C:\blender-2.93.5-windows-x64\2.93\python\lib
C:\blender-2.93.5-windows-x64\2.93\python\bin
C:\blender-2.93.5-windows-x64\2.93\python
C:\blender-2.93.5-windows-x64\2.93\python\lib\site-packages
C:\blender-2.93.5-windows-x64\2.93\python\lib\site-packages\win32
C:\blender-2.93.5-windows-x64\2.93\python\lib\site-packages\win32\lib
C:\blender-2.93.5-windows-x64\2.93\python\lib\site-packages\Pythonwin
C:\blender-2.93.5-windows-x64\2.93\scripts\freestyle\modules
C:\blender-2.93.5-windows-x64\2.93\scripts\addons\modules
C:\Users\mcgav\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\modules
C:\blender-2.93.5-windows-x64\2.93\scripts\addons
C:\Users\mcgav\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons
C:\blender-2.93.5-windows-x64\2.93\scripts\addons_contrib

All paths are pointing to Blender's directories. Maybe I should build this library with path to Blender's Python?

MartyG-RealSense, 2022-03-04:
As you are using Windows instead of Ubuntu, PYTHONPATH wouldn't apply.  Apologies about that.

The easiest way for pyrealsense2 to find the appropriate files may be to copy two files (realsense2.dll and pyrealsense2.pyd) to the same folder that your Python project script is located in, as described in the Python wrapper's Windows instructions in the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#windows

If you have installed the full RealSense SDK on Windows then you can find pre-built versions of these two files at the folder location below:

C: > Program Files (x86) > Intel RealSense SDK 2.0 > bin > x64

[image]

McGavish, 2022-03-06:
Hi @MartyG-RealSense. I've checked that and nothing helped. I've also tried using Realsense library which is built-in to Open3D and it's not working too but I could capture an error. When I used code below two times it crashes Blender the second time with that error. 

```
import open3d as o3d
o3d.t.io.RealSenseSensor.list_devices()
```
```
[Open3D WARNING] No RealSense devices detected. - first script launch
[Open3D WARNING] No RealSense devices detected. - second script launch and crash
Error   : UNKNOWN EXCEPTION
Address : 0x00007FFC099DDB25
Module  : SensorsApi.dll
Thread  : 0000a834
Writing: C:\Users\mcgav\AppData\Local\Temp\blender.crash.txt
```

I think all libraries are installed properly in Blender as I could read them when debugging but it somehow doesn't recognize Realsense cameras. Could you tell me if it's possible to reproduce this error in your case too? This is step-by-step how I installed this library.

Used Blender version: https://www.blender.org/download/release/Blender3.0/blender-3.0.1-windows-x64.zip/. 
It works the same in Blender 3.0.1 and 2.93.5 versions.

1. Open Blender and switch to "scripting" workspace and click "new" to add new script:
[image]

2. Paste code below to install packages and click "run" arrow:
```
import subprocess
import sys
import os

python_exe = os.path.join(sys.prefix, 'python.exe')
target = os.path.join(sys.prefix, 'Lib', 'site-packages')
 
subprocess.call([python_exe, '-m', 'ensurepip'])
subprocess.call([python_exe, '-m', 'pip', 'install', '--upgrade', 'pip'])
 
subprocess.call([python_exe, '-m', 'pip', 'install',' pyrealsense2', '-t', target])
subprocess.call([python_exe, '-m', 'pip', 'install',' opencv-python', '-t', target])
  
print('FINISHED')
```
[image]

3. Paste code this code to try use the camera: https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

I also found another way where Realsense library crashes Blender:
1. Run the code above
2. Unplug Realsense camera and plug it again
3. Instant crash

Could you please check that if it works the same?

McGavish, 2022-03-08:
@MartyG-RealSense Thanks for your research! There aren't any resources about using RealSense cameras with Blender. 

I've checked importing bag file using Open3D directly in Blender and it works fine. Camera connection is the only one problem here for both RealSense and Open3D libs.

I think this problem is connected with threading like in Blender's devtalk example and my "unknown exception" error posted above confirms that.
Sadly this problem blocks my project pretty well, hopefully I can find a way to stream the data to Blender without launching camera directly from Blender.

MartyG-RealSense, 2022-03-08:
One approach might be to set up a Python **flask** and then access the flask from Blender's Python API.

**Setting up a RealSense flask server**
https://support.intelrealsense.com/hc/en-us/community/posts/1500000429802/comments/360014096614

**Running Flask from Blender in single-threaded mode**
https://devtalk.blender.org/t/segmentation-fault-when-running-custom-built-bpy/11284/11

MartyG-RealSense, 2022-03-14:
Hi @McGavish  Do you require further assistance with this case, please?  Thanks!

McGavish, 2022-03-16:
Hi @MartyG-RealSense, thank you for your help and information about Flask, I will check it. 
I think these workarounds will work so we can close this issue.

MartyG-RealSense, 2022-03-16:
You are very welcome  :)   As you suggested, I will close this case.  Thanks again!

closed 2022-03-16 by MartyG-RealSense