realsenseai/librealsense #10292: SR305 Depth frame corners highly skewed/distorted

devshank3, 2022-03-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300 SR305 | 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |   Windows 11 / Ubuntu Jetson Nano  | 
| Kernel Version (Linux Only)    |  NA                           | 
| Platform                                 | PC/NVIDIA Jetson |
| SDK Version                            |  v2.50.0                        | 
| Language                            |  Realsense Viewer / General all programming languages                        | 
| Segment			|  NA                  | 

Dear Realsense team & fellow community,

I know SR305 camera series has reached EOL, But this is the only option I have for short range depth
in this era of hardware shortages. So please help me.

I have fairly a short knowledge on Structured light depth sensing. But would love to do a deep dive.

The issue is depth values at the corners of the depth frame seems to be distorted from the main portion of the frame.
And specifically the right corners bend forward & left corners go backward.

The distortion/bend offset error was around +- 3 to 5 cm from the center.
I asserted the offset numerically by doing a simple experiment. 
Exactly placing the SR305 from a plain wall 30 cm away.

I have attached the pictures/video for reference:

1. Depth:
[image]
3. IR:
[image]
3. Color:
[image]

3D View / model:
1. Front:
[image]
2. Side:
[image]
3. Top:
[image]
4. Side-Top:
[image]

1. A short video of the 3D model

https://user-images.githubusercontent.com/52159502/156920538-884ea585-40ad-4691-bcb0-9c26b7e2e6b8.mp4

My queries:

* Is this a normal behavior of Structured Light SR305 / TOF sensors ?
* Can this be an issue only in my sensor due to damage or mis-calibration ?
* Can I re-calibrate the SR305 by any means ?
* Is this a firmware issue (I'm using  3.27.3.0 ) ?
* Can this be resolved in the software side ?
* Is there any mathematical method to resolve this issue ?

Thanks a lot

MartyG-RealSense, 2022-03-06:
Hi @devshank3  It looks similar to a distortion that was reported in November 2018 for an SR300 at https://github.com/IntelRealSense/librealsense/issues/2752

[image]

There was not a solution in https://github.com/IntelRealSense/librealsense/issues/2752 and as the SR300 / SR305 models do not have a pre-made function to reset the camera to its factory-new default calibration settings, it may be difficult to correct this distortion.  Given that this problem has been reported independently by different SR300 users, it is unlikely to be related to a specific firmware version.

An SR300 calibration paper at the link below also reported the same distortion curvature.

https://core.ac.uk/download/pdf/295596101.pdf

[image]

The authors of this paper used a checkerboard / chessboard to calibrate the camera, though the technique  that is described is complex.

Intel offer an official SR300 checkerboard calibration system in the RealSense Store for $149.99.

https://store.intelrealsense.com/buy-intel-realsense-sr300-calibration-target.html

You could try resetting your SR305 camera in the RealSense Viewer if you have not done so already by going to the **More** option near the top of the options side-panel and selecting **Hardware Reset** from the drop-down menu to see whether this improves the image.

[image]

devshank3, 2022-03-06:
Thanks a lot for replying @MartyG-RealSense ,

So.... I can see this issue persists for long time.... from the issues you mentioned.
And I'm not sure if this calibration issue is in all the SR305 cameras or only a selected few. Please correct me here.

From [this](https://core.ac.uk/download/pdf/295596101.pdf) paper 

They specifically have mentioned 
[image]
[image]

Does this mean its an issue in all SR305 cameras ?

And yes I don't have a highly calibrated environment to carry out the processes mentioned in the paper.
---

Points 

1. And then I tried to reset the camera through the viewer multiple times & still the problem persists.

2. Secondly the mentioned official checker board, does it comes with a software ?

3. I tried moving away from the wall , 50 cm 70 cm so on .... the warping of the edges seems to increase proportionally error is also proportional (That's why i asked if there is a simple mathematical way to resolve it )

4. I came upon this paper [A Range-Independent Disparity-Based Calibration Model for Structured Light Pattern-Based RGBD Sensor](https://www.mdpi.com/1424-8220/20/3/639/htm), pretty similar to the above one. Not sure what I will be able to achieve.... But will try to understand it better.

MartyG-RealSense, 2022-03-06:
I do not have information about whether the SR300 version of the calibration board is accompanied by OEM calibration software like the 400 Series calibration board also sold in the Store has.

Given the low number of past reports that make reference to this particular SR300 distortion issue, it is probable that is not an issue that originates at the factory or it would likely have been reported more frequently.  My experience has been that SR300 has not been used often for 3D pointcloud generation though, with infrared or depth map images being a more common use.

devshank3, 2022-03-07:
Okay @MartyG-RealSense,

Seems like I will have to move forward.

For now temporary solution is to mask out the depth values of the corners.

**But if anyone from the future is reading this issue, if you have any solution please mention it in the thread.**

MartyG-RealSense, 2022-03-07:
Thanks very much @devshank3 for sharing your method with the RealSense community.  Good luck!

MartyG-RealSense, 2022-03-14:
Hi @devshank3  Do you require further assistance with this case, please?  Thanks!

devshank3, 2022-03-17:
Hi @MartyG-RealSense, Thanks a lot...

No further help... If I made any progress I will update the thread.

And I guess New #10319 D405 seems nice... But expensive compared to SR300 (I guess this is due to the global semiconductor industry shortage issue).... Hope it will be cheaper in near furture

MartyG-RealSense, 2022-03-17:
Thanks very much @devshank3 for your feedback.  As you do not require further assistance at the moment, I will close the case and you are welcome to re-open it at a future date if you have follow-up questions.  Good luck!

closed 2022-03-17 by MartyG-RealSense