realsenseai/librealsense #10294: Intermittent blurring of RGB image on D455 camera.

jimmek, 2022-03-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win 10|  
| Kernel Version (Linux Only)    |  NA                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0.3785                          | 
| Language                            |                            | 
| Segment			|                     | 

### Issue Description
Intermittently, the RGB camera autofocus loses track of the imaged grooved surface. It appears to be cyclic. Image1*.png is crisp at top and gets blurry toward the bottom. In the next image, Image2*.png has greater blurriness, and later images returning to a normal crisper image state later in the image sequence. Also, the problematic issue of the purple hue is also apparent. Can you suggest a cause for the intermittent blurriness of the image and settings that my reduce the blurriness?

[image]

[image]

MartyG-RealSense, 2022-03-07:
Hi @jimmek  The pink / purple RGB hue is a known characteristic of the D455 that can occur when the camera is used in bright sunlight and is not a bug.  Page 8 (the final page) of Intel's **Specification Clarification** PDF at the link below (the section titled **D455 RGB Camera in Bright Sunlight**) describes the issue and suggests remedies.

https://www.intelrealsense.com/download/13629/?_ga=2.89775899.1877976161.1602709969-1041004708.1589641371

In regard to the blurring, does it still occur if RGB is set at 848x480 and **60** FPS

[image]

MartyG-RealSense, 2022-03-13:
Hi @jimmek  Do you require further assistance with this case, please?  Thanks!

jimmek, 2022-03-13:
We use a frame rate of 15 fps at 1280x800 for RGB. I'll look at a few more of the data collections to see how persistent this problem may be and let you know. FYI the cameras are mounted on a trailer looking down at the pavement. We have an image overlap of about 50% between frames.

closed 2022-03-13 by jimmek

MartyG-RealSense, 2022-03-13:
Thanks very much for the update, @jimmek - good luck!