realsenseai/librealsense #10298: Repository issues on Linux

AlanLeeJu, 2022-03-09:
| Camera Model    | L515 | 
| Operating System & Version | Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  4.9      | 
| Platform    | NVIDIA Jetson Agx Xavier |

[image]

 I have no idea why be like this on that picture.

MartyG-RealSense, 2022-03-09:
Hi @AlanLeeJu  Could you confirm for me which instructions you are using for installation, please?  If you are installing librealsense on a Jetson from packages then the Jetson-specific procedure at the link below should be used.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

Alternatively, the website JetsonHacks offer a script called **installLibrealsense.sh** for building from packages on Jetson.

https://github.com/JetsonHacksNano/installLibrealsense#installlibrealsensesh

If you will be using ROS then install librealsense and the RealSense ROS wrapper separately.  The librealsense SDK should be installed first and then the ROS wrapper built from source code secondly after librealsense installation has been succesfully completed.

AlanLeeJu, 2022-03-14:
Hi~ Can I ask you some question?
(1)Is it normal to open `realsense-viewer` when I use the code down below?
`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

(2)I am useing the L515 which librealsense version must be above v2.36.0. But my version is v2.17.0.
How to update librealsense version from v2.17.0 to v2.36.0 or more ?
   
Thanks!

closed 2022-03-19 by AlanLeeJu