realsenseai/librealsense #10299: Windows 10 debug assertion failed error

mkelrajt, 2022-03-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435i | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0                         | 
| Language                            |  C/C++  | 

### Issue Description
While testing librealsense2 SDK release v.2.50.0 I've encountered the "Debug assertion failed error". 
[image]

The error only occurs in examples and tools that use openGL library (realsense-viewer, rs-gl) and that are built with Debug configuration and in which GLSL for processing is used.
When built with Release configuration no error is reported, however a memory leak seems to occur when GLSL for processing is used.

Debug assertion error has not occured when running realsense-viever or rs-gl example from librealsense2 SDK v2.49.0. The possible memory leak is still present on v2.49.0.
A similar error does not occur for debug configurations  on Ubuntu 18.04.

Ways to reproduce: 
	• Download librealsense2 v2.50.0 source from github 
	• Checkout to tag v2.50.0 
	• Use Cmake to configure project
	• Open project in Visual Studio 
	• Build project
	• Run realsense-viewer in debug mode
		○ Go to settings: check Use GLSL for rendering 
		○ Start depth stream
		○ No error 
		○ Go to settings. Check use GLSL for processing 
		○ Start depth stream 
		○ Debug assertion error occurs

I am aware that the GLSL memory leak has been reported before https://github.com/IntelRealSense/librealsense/issues/8110
The debug assertion error could be related to the memory leak since both occur when GLSL is used for processing. 
I've also tried the latest commit on the development branch and the assertion debug error occurs there also.

mkelrajt, 2022-03-10:
Hi @MartyG-RealSense thank you for your reply. 
I was aware of the memory leak, but the debug assertion error is new.

MartyG-RealSense, 2022-03-10:
As GLSL support has to be programmed into an application deliberately and is not enabled by default (GLSL is optional support for acceleration of certain SDK functions) you should be able to use Debug mode with 2.50.0 if you do not use rs-gl or you disable it in the Viewer.

MartyG-RealSense, 2022-03-17:
Hi @mkelrajt  Do you require further assistance with this case, please?  Thanks!

mkelrajt, 2022-03-21:
I have no further questions, if new issues arise I will open a new ticket. Thank you for your reply.

MartyG-RealSense, 2022-03-21:
Thanks very much @mkelrajt for the update.  As you do not require further assistance with this case, I will close it.  Thanks again!

closed 2022-03-21 by MartyG-RealSense