realsenseai/librealsense #10300: How to get rotation and translation between two RealsenseL515

Billccx, 2022-03-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Platform                                 | PC  |                     | 
| Language                            |  C++  | 

### Issue Description
Hello!
I have two RealSenseL515 and I wonder if there is a way to get the rotation matrix and translation vector from one RGB sensor on a device to another RGB sensor on the other device?
Thanks in advance!

MartyG-RealSense, 2022-03-09:
Hi @Billccx  Could you provide more information about what you are trying to achieve, please.  Thanks!

Billccx, 2022-03-09:
@MartyG-RealSense  Thank you for such a quick reply! Please forgive my confusing description.
I tried to do triangulation (structure only bundle adjustment) with two cameras, which means I have to get the pose information of each device in advance.
I looked up realsense [document](https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20) and found a function:
 `rs2_extrinsics get_extrinsics_to(const stream_profile& to) const`
It can fetch extrinsic transformation between two profiles on the **_same_** physical device.
Unfortunately, what I want to do is getting extrinsic transformation between two profiles on **_different_** physical devices.
Is there anyway to deal with it?

MartyG-RealSense, 2022-03-09:
Thanks very much for the additional information.

As you mention, extrinsics describe the transform between two different sensor coordinate systems on the same camera (depth to RGB, depth to infrared, etc).  

Extrinsics between cameras is a more difficult question.  I wonder if https://github.com/IntelRealSense/librealsense/issues/2664 may fit what you are trying to achieve.

MartyG-RealSense, 2022-03-15:
Hi @Billccx  Do you require further assistance with this case, please?  Thanks!

Billccx, 2022-03-15:
@MartyG-RealSense  Thanks a lot! I have no more questions.

MartyG-RealSense, 2022-03-15:
Thanks very much @Billccx for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-03-15 by MartyG-RealSense