realsenseai/librealsense #10301: How to increase D455 frame rate using  Raspberry Pi 4 4Gb

josenh14, 2022-03-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Raspbian  | 
| Kernel Version (Linux Only)    |  5.10.92-v8+                                       | 
| Platform                                 | Raspberry Pi 4 4Gb |
| SDK Version                            |  2.50.0                        | 
| Language                            |  opencv/python                          | 
| Segment			|  others                          | 

### Issue Description
Im working on a project that needs to develop a portable camera to record outdoors, so i was trying to record color and depth at 1280x720 30 FPS but seems that raspberry dont have enought power and i get like 7 FPS, lowering resolution to 640x480 helps but i get 18 FPS.

### Code
I cant use .bag because it uses a lot of space, so i save the output from the camera using .avi for color and saving fotograms as .tiff images so i dont lose the depth mesure. 

```
import pyrealsense2 as rs
import time
import threading
from threading import Thread
import numpy as np
import cv2
import multiprocess
from queue import Queue
import time

class CameraProcess(Thread):
    def __init__(self,frameQueue):
        Thread.__init__(self)
        self.frameQueue=frameQueue
        self.fourcc = cv2.VideoWriter_fourcc(*'XVID')
        self.out = cv2.VideoWriter("test.avi", self.fourcc, 30, (1280,720))

    def run(self):
        pass
    def getProcessedFrame(self):
        frameset=self.frameQueue.get()
        proc=self.processFrame(frameset)
        return proc
    def processFrame(self,frameset):
        global ind
        align_to = rs.stream.color
        align = rs.align(align_to)  
        depth_scale =  0.0010000000474974513
        print("Depth Scale is: " , depth_scale)
        aligned_frames = align.process(frameset)

        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
       
        distance_image = depth_image * depth_scale
       
        color_image = np.asanyarray(color_frame.get_data())

        self.out.write(color_image)
        cv2.imwrite(str(ind)+".tiff",depth_image)

class CameraStreamer(Thread):
    depthShape = [1280,720]
    rgbShape=[1280,720]
    def __init__(self,frameQueue):
        Thread.__init__(self)
        self.pipeline = rs.pipeline()
        self.config = rs.config()
        self.config.enable_stream(rs.stream.color, self.rgbShape[0], self.rgbShape[1], rs.format.bgr8, 30)
        self.config.enable_stream(rs.stream.depth, self.depthShape[0], self.depthShape[1], rs.format.z16, 30)
        self.frameQueue=frameQueue
        
    def run(self):
        self.profile = self.pipeline.start(self.config)
        self.loopFrames()
        
    def loopFrames(self):
        global ind
        t1=time.perf_counter()
        while True:
            ind += 1
            frameset = self.pipeline.wait_for_frames()
            self.frameQueue.put(frameset)
            print(f"Looper Thread | Acquired frame {frameset.frame_number} in {np.round((time.perf_counter()-t1),3)}")
            t1=time.perf_counter()

ind = 0
frameQueue=Queue()

#start threads
cameraStreamer=CameraStreamer(frameQueue)
cameraProcess = CameraProcess(frameQueue)

#start threads
cameraProcess.start()
cameraStreamer.start()

nframes=0
iniTime=time.perf_counter()
t1=iniTime
while True:
    fs = cameraProcess.getProcessedFrame()
    nframes+=1
    print(f"Time since last frame {np.round((time.perf_counter()-t1)*1000,3)}ms, qsize {cameraProcess.frameQueue.qsize()},fps: {np.round(nframes/(time.perf_counter()-iniTime),3)}")
    t1=time.perf_counter()
c.pipeline.stop()
```

Is there any way to reach 1280x720 30FPS with Rasberry? or  could you suggest and portable device like raspberry that could reach 1280x720 30FPS?

josenh14, 2022-03-14:
Hi @MartyG-RealSense thanks for your answer, finally i will use a tablet with Windows and the D435i camera but when i try tu use Z16H format dont work. In python i use config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30) but changing it to config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16h, 30) make an error. The example you provided was in C++ so they use RS2_FORMAT_Z16H insteed of rs.format.z16h, is there any place to see the list of formats ? so i can write it properly

MartyG-RealSense, 2022-03-14:
Z16H only works with recording compressed data to bag file with the **enable_record_to_file** instruction.  It cannot be used with live-streamed depth where recording to a bag file is not involved.

There is a list of formats in the official RealSense SDK documentation at the link below.

https://intelrealsense.github.io/librealsense/doxygen/rs__sensor_8h.html#ae04b7887ce35d16dbd9d2d295d23aac7

https://github.com/IntelRealSense/librealsense/issues/2172 has an example of using Python scripting with enable_record_to_file

josenh14, 2022-03-14:
I've notice that most of the weight of the .bag is on the rgb stream, 10 seconds of recording are 1gb rgb and 0,5gb depth. Can i reduce unnecessary data? I just want like a .mp4 i dont need more data from rgb stream.

josenh14, 2022-03-14:
Thanks, I've tried it but it doesn't improve the result. 

However, I have seen that if I use this code to save as .bag it starts to detect people and their age. Is there any way to deactivate all these features and only have the video frames in the color stream? so i can create a lighter .bag

```
def record():
	pipeline = rs.pipeline()
	config = rs.config()

	config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
	config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)
	config.enable_record_to_file('test.bag')
	pipeline.start(config)

	try:
		start=time.time()
		while time.time() - start < 20:
			pipeline.wait_for_frames()
	finally:
		pipeline.stop()
```

MartyG-RealSense, 2022-03-14:
I cannot think of any reason why person detection data would be displayed as a consequence of running the above script.

The RealSense Viewer's RGB stream has **OpenVINO** face detection enabled by default as a post-process that detects human faces and overlays a numeric value.  That face data is not recorded to a bag file though even if a bag were recorded in the Viewer instead of with a Python script.

closed 2022-03-27 by MartyG-RealSense