realsenseai/librealsense #10302: Depth image compression by colorization contains ambiguities

cansik, 2022-03-09:
This issue is about the Intel Whitepaper about [Depth image compression by colorization for Intel® RealSense™ Depth Cameras](https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras) #6289:

First of all we noticed that the formula for the uniform depth colorization lacks information about the datatype.

<img src="https://render.githubusercontent.com/render/math?math=d_{normal} = \frac{d - d_{min}} {d_{max} - d_{min}} * 1529">

Looking at the formula, it seems the calculation is done in `float`, but it lacks a specification if the value is **rounded** or **floored**. We noticed **artefacts** at very specific points in the encoded depth image if we stored the `float` calculation directly into an `int` (flooring). With rounding to an int, the artefacts disappeared.

Second, the inverse hue encoding does not seem to work as described with the formula:

<img src="https://render.githubusercontent.com/render/math?math=disp = \frac{1}{d}, disp_{max} = \frac{1}{d_{min}}, disp_{min} = \frac{1}{d_{max}}">
<img src="https://render.githubusercontent.com/render/math?math=d_{normal} = \frac{disp - disp_{min}}{disp_{max} - disp_{min}}">

Let's calculate a real-world example:

```python
# input
d = 2200 # depth in mm
d_min = 300
d_max = 6000

# calculate disparity values
# if any value = 0 => set to 0
disp = 1 / d # 0.0004545454545
disp_max = 1 / d_min # 0.003333333333
disp_min = 1 / d_max # 0.0001666666667

# norm
d_norm = (disp - disp_min) / (disp_max - disp_min) # 0.09090909089
```

And now? We guess this is just the value to multiply with the independent values (`1529`) as in the original normalisation formula. So the inverse normal calculation would be (again **rounded**):

<img src="https://render.githubusercontent.com/render/math?math=d_{normal} = \frac{disp - disp_{min}}{disp_{max} - disp_{min}} * 1529">

Are our assumptions right?

MartyG-RealSense, 2022-03-09:
Hi @cansik  In general, it is not recommendable to use the depth compression by colorization paper.  Because the depth recovery part of the process has to be performed outside of the RealSense SDK and because there are only a few programming references for the method outside of the paper, it can be a difficult technique to implement.  It is also best suited to C++ users because of the difficulty of coverting the code in the paper for other languages such as Python.

One also has to make the assumption that the result of the division in the **dnormal** equation is multiplied by 1529 because a maths sign is not confirmed in the paper's version of the formula.

[image]

If you do wish to continue working with the technique in this paper: dmin and dmax (minimum depth and maximum depth) are indeed defined as floats in the paper's C++ code.

[image]

In regard to the meaning of **1529**, my interpretation of the paper is that the color values are in the range 0-255, and the hue value is arrived at with the following equation:

**(hue value x 6) - 1**

The hue value will be between 0 and 255.

So the maximum hue value of '1529' would be achieved by multiplying 255 by 6 to get 1530 and then subtracting 1 from 1530 to arrive at the final value of 1529.

**(255 x 6) - 1 = 1529**

My further understanding is that '6' is derived from RGB hue being able to have 6 gradations of intensity in both directions.  

[image]

I do not have enough information about the compression technique in this paper to confirm whether your final assumptions about dnormal are correct, unfortunately.

cansik, 2022-03-10:
@MartyG-RealSense Thank you very much for this detailed answer. I have written an implementation of this paper in [Python (Encode / Decode)](https://github.com/cansik/space-stream/blob/main/spacestream/codec/UniformHueColorization.py#L32-L89), as well as decoders for HLSL and GLSL. And I had to make my assumptions for the code to run properly in the first place. Most important was the rounding of the `float` values after normalisation on the encoder side, so that the if-conditions cover all cases.

I just wanted to ask if these are known problems. Perhaps it would also be possible to expand the white paper.
Do you accept suggestions for improvement via readme.com?

MartyG-RealSense, 2022-03-10:
I have not previously seen your particular problem with artifacts in the image when not rounding up floats to ints.  The most recent query before your one - which can be found at https://github.com/IntelRealSense/librealsense/issues/10145 - was regarding why the RGB values were negative.

There are no plans to expand the depth compression white-paper or encourage its use.  It should be considered as experimental, though RealSense users are welcome to try it and adapt it if they so wish.

MartyG-RealSense, 2022-03-16:
Hi @cansik  Do you require further assistance with this case, please?  Thanks!

cansik, 2022-03-16:
@MartyG-RealSense Thank you for your help, will close this issue.

closed 2022-03-16 by cansik

MartyG-RealSense, 2022-03-16:
Thanks very much @cansik for the update!

cheind, 2024-10-17:
Although this comes late in the game, for future reference, here is a vectorized Python implementation with the equations corrected (to the best of my knowledge)

https://github.com/cheind/hue-depth-encoding