realsenseai/librealsense #10304: Unable to read IMU data on the D435i

zbreit, 2022-03-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.07.150 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX |
| SDK Version                            |  2.50.0                           | 
| Language                            |  python                          | 
| Segment			|  Robot                         | 

### Issue Description
Hello :wave:! I'm trying to get a D435i working on my NVIDIA Jetson Xavier NX, but I am unable to stream data from my IMU.

Inside the `realsense-viewer`, I see all three camera modules (RGB, Stereo, and Motion). When I activate the Motion Module, nothing appears in my Window other than a set of axes.

[image]

Similarly, when I try to read IMU data inside ros melodic using `ros-melodic-realsense2-camera@2.3.2`, I get the following error:

```bash
$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_gyro:=true enable_accel:=true
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
[... 146 lines cut ...]
/tf
/tf_static

$ rostopic echo /camera/accel/sample # Blank

$ rostopic echo /camera/gyro/sample # Blank

$ rostopic echo /camera/accel/imu_info
frame_id: "camera_accel_optical_frame"
data: [1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0]
noise_variances: [0.0, 0.0, 0.0]
bias_variances: [0.0, 0.0, 0.0]
---

$ rostopic echo /camera/gyro/imu_info
frame_id: "camera_gyro_optical_frame"
data: [1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0]
noise_variances: [0.0, 0.0, 0.0]
bias_variances: [0.0, 0.0, 0.0]
---
``` 

I'm not sure exactly what's wrong with my setup, but I have a suspicion that it's an issue with my realsense SDK installation. I installed the realsense SDK using the "Install with Debian Packages" instructions in the [Jetson installation instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md). I saw that some realsense features may require applying the kernel patches described in the "Building from Source Using the Native backend" section, but I wasn't sure if that applied to the D435i IMU.

Any guidance will be greatly appreciated!

MartyG-RealSense, 2022-03-10:
Hi @zbreit  Your image of the RealSense Viewer shows that you are in the 3D point cloud mode, which does not display IMU information.  To view the Motion Module data (gyro and accelerometer), left-click on the **2D** option in the top corner of the Viewer window.

If you installed the RealSense SDK from packages then you do not need to patch the kernel as the patch is included in the packages. Your log shows that the gyro and accel are detected and published.

In regard to the **control_transfer returned** warnings in the log, these can be ignored if they are only appearing at the start of launch or appearing about once a minute instead of generating continuously.  If the control_transfer returned warnings are generated continuously though then this may indicate a serious communication problem with the device.  

Are the warnings being generated in the log constantly after launch has completed successfully with the message ,'RealSense Node is Up!'

zbreit, 2022-03-11:
Hi @MartyG-RealSense. Thanks for the info--I can't believe I missed 2D mode in the `realsense-viewer`. I'll have to try it out tomorrow or Monday when I have access to the camera again.

As far as I remember, the `control_transfer returned` warnings only appeared at the start of launch. I don't think the motion module was enabled in the `realsense-viewer` when I performed my `roslaunch` test.

MartyG-RealSense, 2022-03-11:
A couple of other RealSense ROS users had the same problem this year with IMU data and experienced the same control_transfer returned error.  One of those users advised at https://github.com/IntelRealSense/realsense-ros/issues/2212#issuecomment-1007003514 that using librealsense **2.48.0** and wrapper **2.3.1** solved the problem in their particular case.

MartyG-RealSense, 2022-03-14:
There was another case at https://github.com/IntelRealSense/realsense-ros/issues/2213 with a Jetson Xavier NX where the IMU data was not being received in the Viewer, and also not in the RealSense ROS wrapper.  They too experienced the **Motion module failed** error.  The camera worked correctly on s PC, indicating that there was not a hardware failure in the IMU.  They reported at https://github.com/IntelRealSense/realsense-ros/issues/2213#issuecomment-1008126279 that the IMU problem was resolved after using JetPack 4.5.1.

zbreit, 2022-03-14:
Update 1: I successfully got IMU data in the `realsense-viewer` by installing `librealsense2=2.48.0`. On my Jetson NX running Ubuntu 18.04, I had to run the following commands:
```bash
# Uninstall all realsense packages
sudo apt remove librealsense2 ros-melodic-realsense2-camera librealsense2-udev-rules ros-melodic-librealsense2

# Reinstall librealsense2 at version 2.48.0 and place those packages on hold (to prevent
automatic upgrades)
sudo apt install \
	librealsense2=2.48.0-5ubuntu5~1.gbp156c1b \
	librealsense2-udev-rules=2.48.0-5ubuntu5~1.gbp156c1b \
	librealsense2-utils=2.48.0-5ubuntu5~1.gbp156c1b \
	librealsense2-gl=2.48.0-5ubuntu5~1.gbp156c1b \
	librealsense2-net=2.48.0-5ubuntu5~1.gbp156c1b
sudo apt-mark hold \
	librealsense2 \
	librealsense2-udev-rules \
	librealsense2-utils \
	librealsense2-gl \
	librealsense2-net
	
# To remove the hold on the package versions, just run `sudo apt-mark unhold`
```

I will setting up `ros-melodic-realsense2-camera=2.3.1` momentarily.

MartyG-RealSense, 2022-03-14:
Great to hear about your success with the Viewer!  Thanks so much for sharing the details of your method with the RealSense community. :)

zbreit, 2022-03-14:
Update 2: I was able to install the old version of `realsense-ros=2.3.1`. Unfortunately, installing an old version of ros packages using `apt` isn't very well supported, so I had to build the package from source

```bash
# Setup catkin workspace for realsense-ros=2.3.1
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src/
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout 2.3.1
cd ..
catkin_init_workspace
cd ..

# Fix bug w/ catkin_make clean for Jetsons (see 
# https://github.com/ros-perception/vision_opencv/issues/345#issuecomment-663416626).
# If you're not comfortable with the sed command, feel free to manually change `opencv`
# to `opencv4` in the file, as mentioned in the issue above.
sudo sed -i 's/set(_include_dirs "include;\/usr\/include;\/usr\/include\/opencv")/set(_include_dirs "include;\/usr\/include;\/usr\/include\/opencv4")/g' /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake
rosdep install --from-paths src --ignore-src -r -y
sudo apt purge ros-melodic-librealsense2
catkin_make clean
```

Before running the next commands, I had to fix a bug with an undefined `find_if()` symbol in `~/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp`. To fix this issue, I replaced all instances of `find_if` that were missing namespaces to `std::find_if` (as per https://github.com/IntelRealSense/realsense-ros/issues/910#issuecomment-1062554386).

Finally, I could run

```bash
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc
```

In the end, I was able to run
```bash
roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_gyro:=true enable_accel:=true
```

However, I still have a problem: IMU samples don't seem to be posted to the appropriate ros topics. Specifically, `/camera/gyro/sample` and `/camera/accel/sample` are both empty when I run `rostopic echo _`. Any advice?

MartyG-RealSense, 2022-03-14:
There is a version mismatch between librealsense and the ROS wrapper.  SDK 2.50.0 should be matched with wrapper **2.3.2** as the 2.3.1 wrapper is designed for use with SDK 2.48.0.

zbreit, 2022-03-16:
Aaah, you are right, but I don't know how to correct this issue. To summarize my steps so far:
1. I installed `librealsense2@2.48.0` over `apt`.
2. I downloaded the `ros-realsense2-camera` git repo, checked out the tag `2.3.1`, and tried building with `catkin`.

I thought this would all work out, but it appears that during during step 2, I installed `ros-melodic-librealsense2` accidentally with the following command:

```bash
~/catkin_ws $ rosdep install --from-paths src --ignore-src -r -y
executing command [sudo -H apt-get install -y ros-melodic-librealsense2]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  at ros-melodic-ddynamic-reconfigure
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  ros-melodic-librealsense2
0 upgraded, 1 newly installed, 0 to remove and 14 not upgraded.
Need to get 0 B/12.5 MB of archives.
After this operation, 36.1 MB of additional disk space will be used.
Selecting previously unselected package ros-melodic-librealsense2.
(Reading database ... 216832 files and directories currently installed.)
Preparing to unpack .../ros-melodic-librealsense2_2.50.0-1bionic.20211115.142954_arm64.deb ...
Unpacking ros-melodic-librealsense2 (2.50.0-1bionic.20211115.142954) ...
Setting up ros-melodic-librealsense2 (2.50.0-1bionic.20211115.142954) ...
Processing triggers for libc-bin (2.27-3ubuntu1.5) ...
#All required rosdeps installed successfully
```
So it appears that `ros-melodic-librealsense2@2.50.0` is incorrectly installed by `rosdep` even when I was checked out to `ros-realsense2-camera@2.3.1` in the git repo. If I want to get `ros-melodic-librealsense@2.48.0`, what do I have to do?

MartyG-RealSense, 2022-03-16:
Can you test whether the command below uninstalls your package installation of the wrapper, please.

`sudo apt-get remove ros-melodic-realsense2-camera`

You could then try building ROS wrapper 2.3.1 from source code using the wrapper's **Method 2** installation procedure.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.1

If you installed librealsense from packages and feel that you need to uninstall it then all librealsense related packages can be uninstalled with the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

zbreit, 2022-03-16:
Here is the output of the command:
```bash
sudo apt-get remove ros-melodic-realsense2-camera
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'ros-melodic-realsense2-camera' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  at ros-melodic-ddynamic-reconfigure
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
```

I have been trying to build the ROS wrapper from source, but it doesn't seem to like that I've installed `librealsense2@2.48.0` using `apt`.

MartyG-RealSense, 2022-03-16:
When installing the wrapper from source, it should not be affected by how librealsense was installed (by apt or by source code).  

When installing librealsense from packages on Jetson, you should ideally be doing so either by using the package instructions on Intel's Jetson installation page or with the JetsonHacks **installLibrealsense.sh** script.

**Intel Jetson installation instructions**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

**JetsonHacks**
https://github.com/JetsonHacksNano/installLibrealsense

Note about the above link that if you build librealsense from packages with **installLibrealsense.sh** then you should not then build from source code with the **buildLibrealsense.sh** script.

If you want to build from source code with the JetsonHacks method, use buildLibrealsense.sh instead of installLibrealsense.sh

zbreit, 2022-03-16:
I'll give these a shot, thanks.

zbreit, 2022-03-17:
I found the issue! I was missing `librealsense2-dev`, which prevented `rosdep` from detecting that I already had `librealsense2` installed. I've modified my commands above to fix the issue.

MartyG-RealSense, 2022-03-17:
That's excellent news, @zbreit - thanks very much for the update about your success  :)

MartyG-RealSense, 2022-03-23:
Thanks very much @zbreit for the update!  If you are satisifed with the outcome of this case then please do feel free to close it.  Thanks again!

closed 2022-03-23 by zbreit