realsenseai/librealsense #10305: Realsense 515 loses ability to transfer by USB 3.0

MaChYD, 2022-03-10:
Hi, 

I have a Realsense 515 that is setup in a USB 3.0 system. After a couple of months it stopped reading as a usb 3 and only reads as a usb 2 device. Fortunately, i have an additional unit with me and i plugged it in to check the system, i.e usb cable and pertaining system components and it reads as a usb 3 device. i used the realsense viewer for these tests. what should i do now?

Regards,
Matt

MartyG-RealSense, 2022-03-10:
Hi @MaChYD  Is there any difference if you try turning and inserting the micro-sized end of the USB cable in both directions (USB-C cables are two-way insertion at the micro-sized end).

MaChYD, 2022-03-10:
Hi Marty, yeah tried that

MartyG-RealSense, 2022-03-10:
Has the camera been stored inactive in a cool dry place such as a drawer or cupboard for more than a couple of weeks before the problem manifested?

MaChYD, 2022-03-10:
yes i think stored for 2 weeks clean airconditioned environment

MartyG-RealSense, 2022-03-10:
Storing cameras inactive in this way for a period of 2 weeks or more has been known to cause camera hardware failure in a small number of cases.  Usually the result is an inability for the camera to be detected or for the depth stream to not be able to start.  So your particular issue of USB 3 not being available is milder compared to the failures in those other cases.

In a couple of the cases, the camera restored its functionality on its own after a few weeks.  For example, one that was shipped from India to Intel in California was working normally by the time that it arrived, so environment may be a factor in the self-recovery.   So the best course of action may be to keep testing the camera for a few weeks and if it has not recovered by then then it is probable that it is a non-correctable hardware failure causing the USB issue.

MaChYD, 2022-03-10:
ouh no, thank you for the reply. For my current case reliability is important, how then should i handle this?

MartyG-RealSense, 2022-03-10:
You could attempt to return the camera to the retailer that you purchased it from using the **Return Material Authorization (RMA)** procedure of that particular retailer.  

If the camera was purchased from Intel's official RealSense Store then you can contact the store using the form at the link below, selecting **Return/Warranty** s the topic.

https://www.intel.com/content/www/us/en/forms/support/intel-realsense-store-inquiry.html

If the camera was purchased from a non-Intel retailer then they may not accept a return though as the camera was purchased more than 30 days ago.  It will depend upon the policies of the particular retailer.

MaChYD, 2022-03-10:
return would mean a replacement?

MartyG-RealSense, 2022-03-10:
The L515 model is now End of Life (retired) and has been removed from the RealSense Store.  You would have to talk to the retailer, but it is probable that there would not be replacement units available and a refund may be issued instead if the return if accepted.

MartyG-RealSense, 2022-03-16:
Hi @MaChYD  Do you have an update about this case that you can provide, please?  Thanks!

MaChYD, 2022-03-16:
Hi Marty,

We are talking to the retailer, will update you when we get an update

Regards,
Matthew

MartyG-RealSense, 2022-03-16:
Thanks very much, Matthew.  Please do update me when you have news.  Good luck!

MartyG-RealSense, 2022-03-29:
Hi Matthew, do you have an update that you can provide about your return discussions with the retailer, please?  Thanks!

MartyG-RealSense, 2022-04-04:
Hi Matthew, do you require further assistance with this case please?  Thanks!

closed 2022-04-16 by MartyG-RealSense