realsenseai/librealsense #10307: motion streams in labview?

johnras, 2022-03-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version | Win10 | 
| Kernel Version (Linux Only)    | NA | 
| Platform                                 | PC |
| SDK Version                            | LabVIEW with more recent dll file| 
| Language                            | labview | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I would like to capture the data from the accelerometer and the gyroscope in LabVIEW.  It looks like some of the vi's might have the added those options as inputs, but I am not seeing anything that shows how to actually pull the motion stream off the device.  Is there an example of this available in the LabVIEW SDK that I've missed?

Thanks.

MartyG-RealSense, 2022-03-11:
Hi @johnras  The LabVIEW wrapper is a layer on top of the RealSense DLLs and so you should be able to access the full range of the RealSense SDK's functions.  The RealSense D435i and D455 camera models that are equipped with an IMU are also compatible with the wrapper.  The stream guidance in the wrapper documentation at the link below also indicates that the IMU should be accessible.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/labview#understanding-the-programming

You are correct though that there are no LabVIEW-specific examples or programming references for accessing the IMU through the wrapper.

closed 2022-03-17 by johnras

MartyG-RealSense, 2022-03-17:
Thanks very much @johnras for the update.  Good luck!