realsenseai/librealsense #10309: D435 Poor image quality caused mainly by under/overexposure

LadaOndris, 2022-03-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.07.150 | 
| Operating System & Version |   Win 10 / Linux (Arch)  | 
| Kernel Version (Linux Only)    |                                 | 
| Platform                                 | PC (for now)  |
| SDK Version                            |                          | 
| Language                            |  python                           | 
| Segment			|                     | 

### Issue Description

Hello, I am writing regarding an issue we have with a D435 camera. The problem is that the camera is supposed to be used in various conditions with different lighting. Moreover, the lighting constantly changes during the day, depending on the sunlight intensity. I seem to fail to find a solution to maintaining solid depth image quality. 
I studied a few RealSense articles with explanations and recommendations, so I am familiar with the concept of exposure, which is the problem we are having. The image is either underexposed or overexposed. I read the following article for instance: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

When an object is exposed to direct sunlight, I found the best exposure to be set at 3 μs. However, in the evening when only a room light is shining while using the camera's laser at 100 mW, the best exposure seems to be around 4 000 μs, which is a huge difference from the previous example. Both cases consider an object positioned very close to the camera. If the object moves away, it may become underexposed. 

Our goal is to use the camera for gesture recognition and determining the fingers' positions of a moving hand in the direction of the camera⁠, approaching it from 1 meter to 10-15 cm⁠, which requires different exposure times to avoid underexposure or overexposure. Thus, it is unrealistic to use manual exposure. To our dismay, the autoexposure is too slow and incorrect. I found that it requires the hand to stop at a certain position and wait a little while for the camera to adjust. In some cases, it even fails to adjust sufficiently, and, therefore, the image quality is much lower in comparison to manual exposure.

From the article I mentioned above, I understand that options such as ROI and Mean Intensity Set Point can be used with autoexposure. I do not understand how to use the Mean Intensity Set Point. I also do not see the ROI button in Intel RealSense Viewer.

I need to find settings that will not require adjustments in the final product depending on the time of the day and other conditions. The solution must give reliable images in good quality. Are there solutions to this?

The following images demonstrate what I described above:

Week sunlight, autoexposure. Really poor image quality. 
[image]

Much stronger sunglight.  I experimented with both autoexposure and manual exposure, but none provided me with good results.
[image]

Sometimes, I am able to get pretty decent images.
[image]
[image]

MartyG-RealSense, 2022-03-11:
Hi @LadaOndris  Could you test whether you see a noticable improvement if you enable the **Emitter Always On** option in the Viewer, please?

[image]

The D435's projector pulses in line with exposure, whilst the projector of the D415 model is always on.  Enabling Emitter Always On replicates the behaviour of the D415's always-on projector.

LadaOndris, 2022-03-11:
@MartyG-RealSense Thank you, Marty, for such a quick response. Unfortunately, I cannot find this option in the menu. See the image below.

[image]

MartyG-RealSense, 2022-03-11:
It should be located just below the camera name.

[image]

LadaOndris, 2022-03-11:
Do you mean in the More selection menu? I don't see it there.

[image]

MartyG-RealSense, 2022-03-11:
My apologies, I was thinking of 'Hardware Reset' when I meant 'Emitter Always On', which is elsewere in the side-panel.

It is not clear why this option is not being listed for your D435.  Do you have both of the Infrared options listed, or are one or both of the Infrared streams missing?

[image]

LadaOndris, 2022-03-11:
Yes, I can see both of them and they both work.
Shouldn't I also be able to see the Change ROI button? I'm not sure, however, whether this model supports it.

MartyG-RealSense, 2022-03-11:
The **Set ROI** button should appear next to 'Enable Auto Exposure' once the depth stream is enabled.

[image]

LadaOndris, 2022-03-11:
Exactly, that's what I mean. As you can see in the images I posted above, there is no such button.

MartyG-RealSense, 2022-03-20:
Hi @LadaOndris  Do you require further assistance with this case, please?  Thanks!

LadaOndris, 2022-03-20:
I appreciate your help. I still need to try the setting in direct sunlight, hopefully in a couple of days. I'll let you know then.

MartyG-RealSense, 2022-03-20:
Okay, thanks very much for the update.  Good luck!

LadaOndris, 2022-03-23:
I tried to increase the **laser power** as you mentioned. It does not help when there is already enough natural light, which is the case that I discuss here. For demonstration, both images below had laser power at maximum.

You said in your comment above that the black areas correspond to black or dark gray places due to underexposure. According to my observations, it is true only partly. Notice the bottom right part of the image in my previous post---the forearm is almost black in infrared but the camera still produced valid depth. 

The black areas **increase with the decreasing distance** from the sensor. See the below images. The first one depicts the hand at a distance of 30 cm, while the second one at 16 cm. It is clearly visible that with shorter distances, the number of black areas increases. 

[image]
[image]

You suggested changing the **Mean Intensity Setpoint**, which really comes useful when adjusting the intensity. It does not, however, help me get rid of the black areas in this case. Changing in whichever direction, it causes either underexposure or overexposure of certain areas. There is no value that solves this problem.

I should mention that the above images were captured using the **Hand preset**. If I set the **preset to Default** or High density, then the number of black areas significantly decreases, which leads me to believe that the black areas are created by thresholding and other settings that cause the camera to discard those depth values. See the image below, which was taken using the **Default preset**:
[image]

LadaOndris, 2022-04-04:
Sorry for the late answer. I have been occupied with other work. I will try changing the depth units and I will let you know.
Thank you for your answers so far!

MartyG-RealSense, 2022-04-04:
Okay, no problem.  Thanks very much for the update and I look forward to your next report.  Good luck!

closed 2022-04-28 by MartyG-RealSense