realsenseai/librealsense #10311: RGB on matlab

jo4omath, 2022-03-12:
Hello, I am using an d435 and I cant get the rgb matrix.
I used the code below and the variable color will be a Nx3 matrix.I supposed the values on this matrix are in rgb, I putted a red objet in front of the camera and the greater value of column one was 16 which doesnt correspond to red color.

Any suggestion to help me hoe to get the rgb matrix?

Thank you all.

pipe=realsense.pipeline();
pointcloud=realsense.pointcloud();
profile=pipe.start();

fs=pipe.wait_for_frames();
color_frame=fs.get_color_frame();

pointcloud.map_to(color_frame);
color = color_frame.get_data();
color = [color(1:3:end)', color(2:3:end)', color(3:3:end)'];
pipe.stop();

MartyG-RealSense, 2022-03-13:
Hi @jo4omath  A RealSense user shared their MATLAB script for obtaining the RGB matrix values 1:3, 2:3 and 3:3 at the link below.

https://community.intel.com/t5/Items-with-no-label/MATLAB-RGB-point-cloud/m-p/584724#M10956

In case this page becomes unavailable in future, I have pasted the script below.

**************

% Make Pipeline object to manage streaming

pipe = realsense.pipeline();

% Make Colorizer object to prettify depth output

colorizer = realsense.colorizer();

% define point cloud object

pcl_obj = realsense.pointcloud();

% Start streaming with default settings

profile = pipe.start();

align_to = realsense.stream.color;

alignedFs = realsense.align(align_to);

% Get streaming device's name

dev = profile.get_device();

name = dev.get_info(realsense.camera_info.name); 

% Get frames. We discard the first couple to allow

% the camera time to settle

for i = 1:5

fs = pipe.wait_for_frames();

end 

%create a point cloud player

player1 = pcplayer([-1 1],[-1 1],[-1 1]);

frameCount =0;

while isOpen(player1) && frameCount < 2000

frameCount = frameCount+1;

fs = pipe.wait_for_frames();

%align the depth frames to the color stream

aligned_frames = alignedFs.process(fs);

depth = aligned_frames.get_depth_frame();   

color = fs.get_color_frame();

%get the points cloud based on the aligned depth stream

pnts = pcl_obj.calculate(depth);

pcl_obj.map_to(color);

colordata = color.get_data();

colordatavector = [colordata(1:3:end)',colordata(2:3:end)',colordata(3:3:end)'];

vertices = pnts.get_vertices(); 
  
view(player1,vertices,colordatavector) 

end

% Stop streaming

pipe.stop();

*************

The RealSense user who created the script advises that if an error related to align occurs when running the script then they suggest commenting out validateattributes on line 17 of the MATLAB wrapper's align.m file to get it to work

jo4omath, 2022-03-13:
Thank you !!! @MartyG-RealSense

closed 2022-04-01 by MartyG-RealSense