realsenseai/librealsense #10315: Installation issues with pyrealsense2

RaynardZhu, 2022-03-15:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| Language                            |  python | 

### Issue Description
pyrealsense2 already installed but pycharm still reported an error

[image]

the compiler is python 3.7

MartyG-RealSense, 2022-03-15:
Hi @RaynardZhu  Have you tried installing pyrealsense2 on your Windows PC with pip packages using the instruction **pip install pyrealsense2** as described at https://github.com/IntelRealSense/librealsense/issues/3019

https://github.com/IntelRealSense/librealsense/issues/6402#issuecomment-629763906 has a past guide that I wrote for installing pyrealsense2 on Windows with pip install for Python 3.7.

RaynardZhu, 2022-03-16:
My problem has been solved. Thanks!
I just changed my compiler 
[image]
To choose the second and it will be ok

MartyG-RealSense, 2022-03-16:
It's great to hear that you achieved a solution, @RaynardZhu - thanks very much for the update!

closed 2022-03-22 by MartyG-RealSense