realsenseai/librealsense #10316: how to set threshold filter for d455 realsense camera on matlab

XonarTech, 2022-03-15:
I have a MATLAB project where I'm trying to limit the range of the depth camera of a d400 camera. on the realsense-viewer I found a feature called "threshold filter" which I like. I'm looking for the MATLAB command that achieves the same result. the lib 2.49 wrapper doesn't seem to have it. can someone point me to the right path?

MartyG-RealSense, 2022-03-15:
Hi @XonarTech  Whilst there is not a specific example of a MATLAB wrapper script for the threshold filter available, I hope that https://github.com/IntelRealSense/librealsense/issues/8411 will help you to make progress towards implementing the filter in your project.

Alternatively, you could implement a maximum depth by instead loading a **JSON** camera configuration file, as described at https://github.com/IntelRealSense/librealsense/issues/8970#issuecomment-832781086

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.

closed 2026-10-23 by MartyG-RealSense