realsenseai/librealsense #10317: Number of frames in a .bag file, and timestamp of each frame

hedeya1980, 2022-03-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 2.48.0.3381 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                   | 
| Language                            |  C++                          | 
| Segment			|  others                           | 

### Issue Description
I'm processing the .bag files saved from the D455 depth camera. I need a method to get the number of frames in the .bag file. I searched a lot, but couldn't find how to get the number of frames.
Also, I need to get the timestamp of each frame. I'm using the '.get_timestamp()', but I don't know whether it's the correct method or not. Also, sometimes the '.get_frame_timestamp_domain()' results in 'Global time', and sometimes it's 'System time', what is the difference?

Thanks in advance.

MartyG-RealSense, 2022-03-15:
Hi @hedeya1980   A C++ script at the link below for counting the number of frames in a bag file may be a helpful reference.

https://support.intelrealsense.com/hc/en-us/community/posts/360031847974/comments/360007849114

You could also check the frame count with the RealSense SDK's C++ tool called **rosbag-inspector**.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rosbag-inspector

In a past case, one of the RealSense support team members suggested the following method for checking frame count with rosbag-inspector:

"Open the target bag file, and look at the **topics /device_0/sensor_0/depth_0/image/data** or **/device_0/sensor_1/color_0/image/data** and look at the count".

[image]

***********

In regard to retrieving timestamps, yes, **frames.get_timestamp()** is the appropriate method to use.  

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1

https://github.com/IntelRealSense/librealsense/issues/2188 is an excellent reference about different types of timestamp and how they work.

System Time is the camera's software timestamp, and is used when hardware timestamps are unavailable (such as when support for hardware metadata has not been enabled in the RealSense SDK).  . If hardware timestamps are available, librealsense will take advantage of them.

On Windows, metadata support can be added using information in the link below, or more easily by instead using a pop-up box in the RealSense Viewer that may appear when launching it to request if you want to enable metadata support.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows

When global time is enabled, device time and system time are synchronized. 

https://github.com/IntelRealSense/librealsense/pull/3909 

https://github.com/IntelRealSense/librealsense/issues/4505#issuecomment-553590316

hedeya1980, 2022-03-16:
Thanks a lot @MartyG-RealSense for the detailed and valuable answer. I managed to get the number of frames from using the code snippet you referred to. I'm reading the wonderful resources you referenced, and will get back to you if I still need any clarifications. Thanks a lot.

MartyG-RealSense, 2022-03-16:
You are very welcome, @hedeya1980 - I'm pleased that I could help. Please do feel free to post follow-up questions if you have them. Good luck!

MartyG-RealSense, 2022-03-17:
An example of C++ scripting for printing FPS is provided by a RealSense user at https://github.com/IntelRealSense/librealsense/issues/7749

In that same case, information about the difference between Hardware FPS and Viewer FPS can be found at https://github.com/IntelRealSense/librealsense/issues/7749#issuecomment-725979247

MartyG-RealSense, 2022-03-23:
Hi @hedeya1980  Do you require further assistance with this case, please?  Thanks!

hedeya1980, 2022-03-24:
Hi @MartyG-RealSense ,
It seems that there is no direct method to get the FPS in a recorded bag file. I think I have to keep calculating the time between frames, and estimate a moving average FPS.
Also, I noticed that when I use the get_position method, and calculate the difference between the current frame and last frame, the difference is sometimes zero. Is that OK?

MartyG-RealSense, 2022-03-24:
At https://github.com/IntelRealSense/librealsense/issues/7488#issuecomment-704124850 a RealSense team member provides a link to the official C++ code in the RealSense SDK that is responsible for calculating FPS.  It may provide some useful insights for your own approach.

At https://github.com/IntelRealSense/librealsense/issues/8417 a RealSense user shares their own C++ method for calculating FPS.

If you are calculating the difference between the current frame and last frame, conceivably '0' could occur each time that the current frame **is** the last frame.  If bag playback does not have repeat disabled then playback could loop back to the beginning and print '0' as the time difference each time that the last frame is reached.

hedeya1980, 2022-04-03:
Hi @MartyG-RealSense ,
I'm sorry for being late to get back to you.
When I use the following code:
`
while(k< stoi(argv[8]))
    {
        //pipe.wait_for_frames();
        bFrame = pipe.try_wait_for_frames(&data);
        curPos = playback.get_position();
        if (curPos < lastPos)
            break;
        else
        {
            std::cout << "time between frames: " << curPos - lastPos << std::endl;
            counter++;
            lastPos = curPos;
            k++;
        }
    }
`
I get the following result:
[image]

which means that sometimes the time between successive frames can be zero.
Also, I understand that the

MartyG-RealSense, 2022-04-03:
When you earlier said that you are calculating the difference between the "current frame and last frame", did you mean the **previous** frame, and not the final frame of the bag file, please?

If you were referring to the previous frame, then '0' might be an event where there was a bad frame.  In that event, the SDK would recover by returning to the last known good frame and then continue the streaming onwards from there.  So perhaps the time is 0 because the same frame number is repeating as the SDK recovers from the 'hiccup' in the streaming.   For example, frame 1, 2, 3, 3, 4, 5, 6, 6, 7

MartyG-RealSense, 2022-04-16:
Case closed due to no further comments received.

closed 2022-04-16 by MartyG-RealSense

hedeya1980, 2022-04-16:
Hi @MartyG-RealSense,
Sorry for being late to reply. Your answers were very helpful. 
Thanks a lot. Really appreciating all your help.

MartyG-RealSense, 2022-04-16:
No problem at all, @hedeya1980 - I'm pleased that I could help.  Thanks for the update!