realsenseai/librealsense #10318: Segfault when playing back prerecorded file

xerxesb, 2022-03-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Prerecorded video from D435 | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  5.10.25-linuxkit                                         | 
| Platform                                 | x64 VM, x64 Docker container (nvidia/cuda:11.4.2-cudnn8-runtime-ubuntu18.04)  |
| SDK Version                            |  2.50                          | 
| Language                            |  Python 3.6.9                          | 
| Segment           |  Other                          | 

### Issue Description
* Our application starts by loading a small "blank" BAG file to display to the user prior to a camera being connected. 
  * This file is only a couple of second long, and was recorded on a D435 while covering the sensors so that everything was blackened.
* After approx 3 minutes of pre-recorded video looping (sometimes sooner), our application crashes with a Segmentation Fault.
* This fault does not seem to occur at a point during the execution of our main loop (i.e. on a specific function call), which leads me to believe the segfault is triggered by a background thread within the realsense sdk 
* I've attached a GDB session to the runtime and I get the following backtrace when the segfault occurs
```
Thread 12 "python3" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe5f1e700 (LWP 45183)]
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:435
435     ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: No such file or directory.
(gdb) bt
#0  0x00007ffff7b70bb4 in __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:435
#1  0x00007fff575b9fe5 in XXH32_update () at /usr/lib/x86_64-linux-gnu/liblz4.so.1
#2  0x00007fff3ad0e14e in decompressBlock () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#3  0x00007fff3ad0e269 in roslz4_decompress () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#4  0x00007fff3ad0e3fd in roslz4_buffToBuffDecompress () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#5  0x00007fff3ad0c861 in rosbag::LZ4Stream::decompress(unsigned char*, unsigned int, unsigned char*, unsigned int) () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#6  0x00007fff3acfbddc in rosbag::ChunkedFile::decompress(rosbag::compression::CompressionType, unsigned char*, unsigned int, unsigned char*, unsigned int) () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#7  0x00007fff3ad02a6b in rosbag::Bag::decompressChunk(unsigned long) const () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#8  0x00007fff3ab1e142 in std::shared_ptr<diagnostic_msgs::KeyValue_<std::allocator<void> > > rosbag::Bag::instantiateBuffer<diagnostic_msgs::KeyValue_<std::allocator<void> > >(rosbag::IndexEntry const&) const ()
    at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#9  0x00007fff3ab1ef60 in diagnostic_msgs::KeyValue_<std::allocator<void> >::ConstPtr librealsense::ros_reader::instantiate_msg<diagnostic_msgs::KeyValue_<std::allocator<void> > >(rosbag::MessageInstance const&) ()
    at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#10 0x00007fff3aaef3d1 in librealsense::ros_reader::get_frame_metadata(rosbag::Bag const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, librealsense::device_serializer::stream_identifier const&, rosbag::MessageInstance const&,
librealsense::frame_additional_data&) () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#11 0x00007fff3aaf930f in librealsense::ros_reader::create_image_from_message(rosbag::MessageInstance const&) const () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#12 0x00007fff3aafca65 in librealsense::ros_reader::create_frame(rosbag::MessageInstance const&) () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#13 0x00007fff3aafd0c2 in librealsense::ros_reader::read_next_data() () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#14 0x00007fff3a98a01c in  () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#15 0x00007fff3a98ba49 in  () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#16 0x00007fff3aa61060 in  () at /usr/local/lib/python3.6/dist-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so
#17 0x00007fffeb1e1a50 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#18 0x00007ffff77ca6db in start_thread (arg=0x7fffe5f1e700) at pthread_create.c:463
#19 0x00007ffff7b0361f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
* Here is a (modified/redacted) snippet of how we set up our pipeline based on our code:
```python
def init():
  pipeline = rs.pipeline()
  config = rs.config()
  rs.config.enable_device_from_file(config, file_path, True) # Repeat playback = True
  config.enable_all_streams()
  
  # This block retrieves the sensor intrinsics by starting the pipeline. Once we have the intrinsics we stop it
  profile = pipeline.start(config)
  depth_profile = rs.video_stream_profile(profile.get_stream(rs.stream.depth))
  color_profile = rs.video_stream_profile(profile.get_stream(rs.stream.color))
  self.depth_intrinsics = Intrinsics(depth_profile.get_intrinsics())
  self.color_intrinsics = Intrinsics(color_profile.get_intrinsics())
  self.pipeline.stop()

def start():
  playback = profile.get_device().as_playback()
  playback.set_real_time(True)

  # Wait for the first frame with a longer timeout because large recorded files can take a while for the first frame to arrive
  self.pipeline.wait_for_frames(FRAME_WAIT_TIMEOUT_STARTUP)

```

Other notes:
* This failure only happens in some environments. For environments where the failure occurs, it is reliably reproducible.
  * We have observed this failure in a Docker container based on nvidia/cuda:11.4.2-cudnn8-runtime-ubuntu18.04
  * We have observed this failure sometimes occurs if run in a VM (but not all VMs)

MartyG-RealSense, 2022-03-16:
Hi @xerxesb   When playing back pre-recorded bag files, having **set_real_time** set to True can increase the likelihood of instability.  Does stability improve if you set it to False instead, please?

It is also worth mentioning that although the RealSense SDK can work with Windows 11 and Intel did not experience problems when informally testing the SDK tools and examples with it, the Windows version of the SDK is officially validated for use with Windows 10 and so there may be unpredictable consequences that arise from use with Windows 11, especially in areas such as VM use that have not been tested for on Win11.

xerxesb, 2022-03-16:
Thanks for responding, @MartyG-RealSense. I've tried re-running with `set_real_time(False)` and it still fails in the same way.

Thanks for the clarification re Win11 host running a VM. The key sticking point I think is that it reliably fails in our docker container, irrespective of the host operating system (Windows, Mac, Ubuntu). 

Do you have other suggestions for debugging the issue?

MartyG-RealSense, 2022-03-16:
Intel published a librealsense Docker tutorial when SDK 2.50.0 was released, so you could check the details in it against your own Docker configuration procedure

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

MartyG-RealSense, 2022-03-23:
Hi @xerxesb  Do you require further assistance with this case, please?  Thanks!

xerxesb, 2022-03-23:
Thanks @MartyG-RealSense - I read the docker instructions, but the content mostly applies to passing through the hardware. Our problem exists with the pre-recorded video playback. Also given that this problem occurs in a virtual environment leads me to think its not Docker specific.

According to the stacktrace, the segfault occurs deep in pyrealsense, so I'm not sure how else to debug this issue.

MartyG-RealSense, 2022-03-24:
vGiven that problems have previously been reported in bag reading applications that loop playback like your application is doing, https://github.com/IntelRealSense/librealsense/issues/6887 may provide helpful insights.

You could test whether the loop is the problem in your particular application by recording a longer bag and setting playback looping to false.

`rs.config.enable_device_from_file(self.config, file_path, repeat_playback=False)`

MartyG-RealSense, 2022-03-31:
Yes, that is fine to keep this case open for a further time period.  Good luck!

MartyG-RealSense, 2022-04-13:
Hi @xerxesb  Do you have an update about this case that you can provide, please?  Thanks!

xerxesb, 2022-04-18:
@MartyG-RealSense between public holidays and leave I haven't had a chance to look further.

Is it possible to postpone for a little longer? Thank you

MartyG-RealSense, 2022-04-18:
That is no problem at all to keep this case open for a further time period.  Thanks for the update!

xerxesb, 2022-04-27:
Hi @MartyG-RealSense - Apologies for taking so long to run this test.

I've just re-rerun the system with a longer bag file with `repeat_playback=False` and confirmed that it still crashes. The time taken to crash is slightly different, but it segfaults all the same

xerxesb, 2022-04-30:
Hi @MartyG-RealSense

I tried as you suggested and copied the following lines into `init`, but it did not change the result
```python
playback = self.profile.get_device().as_playback()
playback.set_real_time(True)
```

> I would also note that commonly in Python bag reading scripts, wait_for_frames() is placed within a while True: condition, like in the RealSense SDK's read_bag_example.py Python example program.
Yes. Our system has a separate async coroutine that gets the frame from the camera device in a loop. I omitted it for brevity, but we essentially do the same thing.

I have made a little more progress investigating though:
* If I set `repeat_playback=False`, I observe the call to `get_frame` will intermittently hang, and time out with a `RuntimeError: Frame didn't arrive within X` exception. For example, the system will start up, the .BAG file will start playing for 20 seconds (time is variable), and then suddenly things will hang and I get the frame timeout.

If I set `repeat_playback=True`, I observe the video will start playing, but it will suddenly restart from the beginning. It does this several times before finally SEGFAULT'ing. For example, the video plays for 20 seconds (again timing varies), restarts from the beginning, plays for 15 seconds, restarts, plays for 10 seconds, restarts, plays for 5 seconds,  SEGFAULT.

My suspicion is that there is some deadlock causing the call to `get_frame` to time out. If `repeat_playback=False` this bubbles to the client as a `RuntimeError`. If `repeat_playback=True` my guess is that the framework is restarting playback on its own. If it restarts too many times it results in the segfault.

If you can recommend a way to seta a breakpoint debug inside the library (specifically on when `RuntimeError` is raised) that might help narrow it down.

I have separately tried to reproduce using a test script but I haven't had any success. Clearly the complexity of our system adds confounding factors so I'm not sure which elements to try replicate in the test script (e.g. use of coroutines, timing/latency adjustments, using the frame_data once returned etc).

MartyG-RealSense, 2022-04-30:
Have you tried inserting an exception catch on the line immediately after pipe.stop to detect an exception and print the details of it?  For example in Python:

```
self.pipeline.stop()
print("Exited without errors.")
except Exception as ex:
print('Exception occured", ex)
```

xerxesb, 2022-05-05:
Sorry @MartyG-RealSense! I didn't see that you replied to me straight away!

I'll have a look at your suggestion and let you know

xerxesb, 2022-05-05:
@MartyG-RealSense Unfortunately that didn't change anything. I wrapped all calls to `stop` in exception blocks and none of them were triggered.

MartyG-RealSense, 2022-05-06:
If your goal is to provide the user with a black screen whilst waiting for camera connection, an alternative to looping the 2 second blank bag may be to play the bag and then immediately **pause** playback so that the image remains on the same blank black frame indefinitely.  A method for implementing this in Python is provided in a discussion at the link below.

https://stackoverflow.com/questions/58482414/frame-didnt-arrived-within-5000-while-reading-bag-file-pyrealsense2

xerxesb, 2022-05-09:
Thanks for the suggestion @MartyG-RealSense, but that's not how it's used.

We have a system that performs computer vision which is loosely decoupled from the input source. The user can switch between live camera source and pre-recorded sources depending on what vision processing they're testing or working on. Oftentimes, pre-recorded files give us repeatability when testing changes to vision processing algorithms.

When launched, our system starts in a "default" state using a simple blank bag file as a pre-recorded input source, but that's only until the user selects their desired input capture source (either live camera or a specific pre-recorded video)

MartyG-RealSense, 2022-05-09:
I went back to the start of this case to re-examine it from the beginning.  I note that you are using kernel **5.10.25-linuxkit**.  The librealsense SDK does not officially support kernel 5.10 and although the SDK can work with unsupported kernels it can have unforseen consequences in regards to stability.  The most recent supported kernels are 5.4 when building the SDK from Debian packages, and 5.8 and 5.11 when building it from source code.

The kernel can though be bypassed if building from source code with the **RSUSB backend** installation method, which is not dependent on Linux versions of kernel versions and does not require patching.

xerxesb, 2022-05-09:
Hmmm....That's quite relevant, actually.

My working VM image is Ubuntu 18.04 LTS (kernel v5.8), so if the kernel issue is the root cause, it checks out that the software works in that environment.

I'll check with my colleague about what kernel version he's running when he gets the same issue and get back to you

xerxesb, 2022-05-12:
Hi @MartyG-RealSense. I have an update for you.

My colleague is using Ubuntu 18.04 in a virtual machine (Windows host). Inside the VM, he is running a docker container with the application. We confirmed that the same failure happens in his setup.

The container reports system info as `Linux cdf5878596cc 5.4.0-109-generic #123~18.04.1-Ubuntu SMP Fri Apr 8 09:48:52 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux` which checks out as the container runtime is likely just passing straight through to the host's kernel.

MartyG-RealSense, 2022-05-12:
Thanks very much for the update!

Have you checked for the possibility of a **memory leak**, where the computer's available memory capacity is progressively consumed over time until it becomes unstable or crashes?  You can check for this in Ubuntu by launching your program and using a system monitoring tool such as **htop** to look for the free memory reducing.  This could account for the default occurring without an obvious specific cause by a line of code and at similar but not always the same time periods after the program has started running.

xerxesb, 2022-05-12:
I decided to run the application's container image on bare metal (rather than in a VM as in my comment above) and can confirm that the same problem occurs on an Ubuntu 18.04 installation.

The container system info is `Linux 31ac91f6079c 5.4.0-91-generic #102~18.04.1-Ubuntu SMP Thu Nov 11 14:46:36 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

MartyG-RealSense, 2022-05-12:
I just posted a new comment above at the exact moment that you did. :)

xerxesb, 2022-05-12:
Confirming - there is no memory leak. Just monitoring through htop and the consumption stays fixed throughout the life of the process until it receives the segfault

MartyG-RealSense, 2022-05-12:
Has the code been tested outside of a Docker container to see whether the segfault still occurs when Docker is not involved?

xerxesb, 2022-05-12:
So far we've only experienced the problem in the docker environment, yes

MartyG-RealSense, 2022-05-13:
There was a case at https://github.com/IntelRealSense/librealsense/issues/8601 where set_real_time=true and bag looping was used and they were experiencing segfaults.  They thought that it might be segfaultjng when pipeline stop was called.

xerxesb, 2022-05-18:
Thanks @MartyG-RealSense. I had a look through those but we don't call `stop` after its initialised (so we go `start -> get intrinsics -> stop -> start`). Once the bag file starts playing after the second call to `start` it never stops.

I have observed though that the bag file will periodically restart playing back from the beginning before it even gets to the end of the video. This will happen several times prior to the segfault

MartyG-RealSense, 2022-05-18:
My understanding is that a bag returning to the start before it reaches the final frame can be a consequence of using **set_real_time = true**.

xerxesb, 2022-05-19:
I'll have a look into whether changing real-time to `False` will change the behaviour.

xerxesb, 2022-05-30:
Hi @MartyG-RealSense. I'm sorry but I've had to move onto other issues for now, and this one is again not at the top of the pile.

I don't feel like we've sufficiently worked our way to the bottom of the issue - is there a way to freeze this issue temporarily for a few weeks until I can get a chance to work on it?

Thanks

MartyG-RealSense, 2022-05-30:
It's no problem at all if you would like to keep this case open for a few weeks.  Alternatively, you could temporarily close it with the 'Close Issue' button under the comment writing box and re-open it when you are ready to resume.  Thanks!

xerxesb, 2022-06-20:
Hi @MartyG-RealSense. I appreciate you being so patient. I've been swamped with other priorities, so I'm thinking the best option at this stage is for me to close it until this bug becomes a blocking issue for us.

Thanks for your help in trying to get to the bottom, I'll close the issue temporarily for now and re-open it when we have the headspace. 

For any future readers who scroll this far with the same problem: **TL;DR: Didn't get to the bottom of the problem**

closed 2022-06-20 by xerxesb

MartyG-RealSense, 2022-06-21:
Thanks very much @xerxesb for the update.  Please do feel free to re-open the issue at a future date when you need to.  Good luck!