realsenseai/librealsense #10319: Announcement: Introducing the RealSense D405 close-range stereo depth camera

MartyG-RealSense, 2022-03-16:
Hi everyone,

Intel have introduced a new RealSense 400 Series stereo depth camera model, the **RealSense D405**.  It is designed for very close range scanning, with an ideal depth sensing range between 7 cm and 50 cm.  It is available now from the official online RealSense Store.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d405.html

A new March 2022 edition of the data sheet document for the 400 Series cameras that includes the D405 is also now available.  Please visit the link below and click on the **PDF format** link to download the new edition as a PDF file.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

mcanyucel, 2022-03-18:
Thanks for the news!

In the readme of this repo, there is a section titled "Special notice from Intel® RealSense™ regarding the recent press announcement", which states that Intel is winding down the RealSense business. So does this new device release means it is not being *winded down* as fast?

I also have a question; the same section above states that 

> The products identified in this notification will be discontinued and unavailable for additional orders after Feb 28, 2022.

Which products does this line refer to? We are planning to use D455 in one of our near future projects, and wondering if it will be possible to buy D455 from Intel.

MartyG-RealSense, 2022-03-18:
Hi @mcanyucel  The RealSense T265 Tracking Camera, the L515 lidar depth camera and the F450 / F455 facial authentication products have been declared End of Life (retired).  The 'Touchless Controller' and 'Dimensional Weight Software (DWS) tools are also discontinued.  

The 400 Series stereo depth camera range, including the D455, is continuing though.

mcanyucel, 2022-03-18:
> The 400 Series stereo depth camera range, including the D455, is continuing though.

Ah, that's good news. Thanks!

devshank3, 2022-03-18:
Hello @MartyG-RealSense & Intel Realsense team / support

The play video button link seems to be broken.

[image]

And any sample datasets from the device would be great, Because the price is steep & for making the decision to buy,....... a sample dataset is good to have.

MartyG-RealSense, 2022-03-18:
I checked the link and it is working for my Windows computer and the Chrome browser.  The page below should appear when the Play Video link is clicked on.

[image]

There is not an official sample bag file for the D405.  I have recorded a 5 second long 176 mb test bag with the D405 that contains RGB, Depth and Infrared streams.  It can be obtained by downloading the bag file **D405-test.bag** from the Google Drive link below.

https://drive.google.com/file/d/1c3AJAhwTCj3OqUF3fNB2yi2OP4gvT2X1/view?usp=sharing

devshank3, 2022-03-18:
Thanks a lot @MartyG-RealSense for sharing the bag file. 

I created a brightclove account and logged in, but the video was not found.
[image]
I'm not sure if this video was shared only to internal people.

MartyG-RealSense, 2022-03-18:
I experienced sign-in problems.  I will report the issue to Intel.  Thanks very much for highlighting it!

AndreV84, 2022-04-07:
@MartyG-RealSense which realsense sdk version/ ros realsense wrapper version can be used with d405? Thanks

MartyG-RealSense, 2022-04-07:
SDK 2.50.0 can be used with D405, but only the new 4.x series of ROS2 wrappers support D405.  There are no plans to add D405 support to the ROS1 wrapper.  So at the time of writing this, SDK 2.50.04 and ROS2 wrapper 4.0.4 would be a compatible match for using D405.

hakunaMatataHub, 2022-05-03:
@MartyG-RealSense 
I read in the data sheet that this does not has IR projectors and moreover has IR cut filter, so will this be suited for scenes which have sparse features?
I have been using intel D435i which works well for medium range but not very close. But it has IR projectors hence does not make a problem in scenes which has sparse features inherently. IIRC D435i does feature matching beween stereo image pairs for depth triangulation, hence I expect the same working principle for D405 as well. 

I would also appreciate any input about performance from the community if anyone has tested D405 with sparse scenes.

MartyG-RealSense, 2022-05-03:
Hi @hakunaMatataHub  If the objects that a D405 are observing have low texture detail or no texture then the absence of a projector and the inability to see external IR patterns from a different 400 Series model or an IR pattern projector may result in sparse depth detail.  Using an external **visible** pattern projector instead of an IR one to add texture to surfaces that the D405 can perceive may help.

In regard to the experiences of other RealSense users with low-detailed objects used with D405, https://github.com/IntelRealSense/librealsense/issues/10351 and the link below may be helpful references.

https://support.intelrealsense.com/hc/en-us/community/posts/5745342403603-green-Chilli-and-D405

hakunaMatataHub, 2022-05-04:
@MartyG-RealSense 
Thank you for the reply. 
I want through the discussion in #10351 and the links provided. Does D435i has a rotating wedge mask in front of the IR projector?

MartyG-RealSense, 2022-05-04:
@hakunaMatataHub  There is not a physical mask device for covering the projector on camera models that have one such as the D435i but you can disable it in software.  In the RealSense Viewer, setting the **Enable IR Emitter** option's drop-down menu to 'Off' instead of its default setting of 'Laser' will turn the projector off.  Setting the value of the **Laser Power** option to '0' will do the same.

You can also control the IR Emitter and the Laser Power settings in scripting.  A C++ example of this is at the link below under the 'librealsense2' heading.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser