realsenseai/librealsense #10320: Disable resampling/super resolution in Realsense Viewer

wl2776, 2022-03-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win 10) | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0                          | 

### Issue Description
I am trying to estimate camera depth of field.
To do this, I am targeting the camera at a bar code and analyze how it shows the lines.

This is the screenshot of the Realsense Viewer window. I've used mouse wheel to zoom.
[image]

And this is the screenshot of the IrfanView window, showing color frame, captured with the Realsense Viewer.
I've zoomed into approximately the same region as in the Realsense Viewer.

[image]

IrfanView allows to turn off all resampling, and I can clearly see borders of pixels.

I cannot see pixel borders in the window of the Realsense viewer. This suggests that it applies some resampling/super-resolution algorithms. 

I was unable to find any option to turn them off.

MartyG-RealSense, 2022-03-17:
Hi @wl2776  The new ** RealSense D405** close-range camera model released last week is best suited to very close range observation.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d405.html

If you need to use the D435 model though then I would recommend moving the camera further away from the barcode and then maximizing the Viewer's RGB option **Sharpness** to '100' to enhance the quality of the RGB image.

[image]

In https://github.com/IntelRealSense/librealsense/issues/7187#issuecomment-684879758 I tested RGB 'Sharpness' increasing with a D415 camera on the barcode of a drinks multipack.

wl2776, 2022-03-17:
1. Setting sharpness to 100 doesn't show borders of pixels, they are still blurred
2. I want to turn off any quality enhancement instead of tweaking it.

MartyG-RealSense, 2022-03-17:
It is possible to capture raw RGB **RAW16** frames from the camera hardware using OpenCV color conversion, as demonstrated with Python scripting at https://github.com/IntelRealSense/librealsense/issues/7275

The RAW16 RGB format cannot be directly streamed in the RealSense Viewer though.  The Viewer will also apply the default values of a particular camera model's RGB settings to the image.  

You could try disabling the RGB **Decimation Filter** in the Post-Processing sub-section of the RGB options to see whether that improves your results if it is not already disabled (indicated by a red colored icon beside it).

[image]

Once you position a D435 closer than 0.1 m / 10 cm to an object then below about 7 cm blurring will set into the image, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/7631#issuecomment-714504852

wl2776, 2022-03-17:
The post-processing is already disabled.
For now it seems impossible to ask the Realsense Viewer to get off all its enhancement attempts.

Here is the result I've got from the simple python script
[image]

It seems to me that my issue is not completely understood. 
Please, take a closer look at the image. 
It is possible to view sharp vertical and horizontal borders between pixels, if the image is displayed *not* in the Realsense Viewer.

You cannot determine such borders in the Realsense viewer, because it applies some image enhancement algorithm

[image]

MartyG-RealSense, 2022-03-17:
The RealSense Viewer tool is not designed for the depth-of-field analysis purpose that you have in mind.  It provides easy access to the camera's capabilities through a graphical interface without having to create program scripting.

Because the optics of RealSense camera sensors are fixed, once the camera is below a certain distance from the observed surface then the image will start to blur.  So you will likely not be able to achieve your goal with the Viewer tool.

wl2776, 2022-03-17:
I see. But this could be a good feature that is rather easy to implement.

closed 2022-03-23 by wl2776