realsenseai/librealsense #10321: Motion Module failure!

Jsc-Nuaa, 2022-03-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version | Linux (Ubuntu 18.04) |     
| Platform                                 | PC |
| SDK Version                            |v2.50.0 }                          | 

Hi, I had completed the  imu-calibartion of my camera based on th rs-imu-calibration.py.  I didn't save the raw data of my camera and I wrote the new data to my camera.Then I opened realsense-viewer and found that the Motion Module failure. What should I do to fix this issue? Thanks.
[image]
[image]

Jsc-Nuaa, 2022-03-18:
And I thought I didn't save the data to device successfully, what should I do ? Thanks!

MartyG-RealSense, 2022-03-18:
Hi @Jsc-Nuaa  In the RealSense Viewer, the IMU typically continues to work normally after a **Motion module failure** error occurs, making it more of a small annoyance than a serious problem.  Is your Viewer able to display the gyro and accel information correctly after the error occurs, please?

If you are attempting to record data to a bag file then any streams that are active at the time when the record button is clicked will be written into the bag file.  A stream will not be recorded if it is enabled **after** the record button is clicked.

Jsc-Nuaa, 2022-03-18:
@MartyG-RealSense  Thanks! But, when I retry the IMU-Calibration, it reports error "No IMU DATA". And I check my device through realsense-viewer, it reports "IMU extrinnsic table not found ;using CAD values". I guess it may be a serious problem.

Jsc-Nuaa, 2022-03-18:
I am so sorry that it suddenly works, even if I don't why it works. Thanks for your help!

MartyG-RealSense, 2022-03-18:
It's no trouble at all.  I'm pleased to hear that it is working for you now  :)

Jsc-Nuaa, 2022-03-18:
[image]

closed 2022-03-18 by Jsc-Nuaa