realsenseai/librealsense #10322: Node.js wrapper on npm register is out of date

sparkyb, 2022-03-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.8                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                          | 
| Language                            |  nodejs  | 
| Segment			|  Entertainment                          | 

### Issue Description
I'm trying to use my RealSense camera from Node.js. If I follow the documented directions and try to install it from npm with `npm install node-librealsense` it fails to install. It builds the native librealsense just fine, but the Node.js wrapper fails to compile. It looks like the version published to the npm register is 4 years old and doesn't support modern Node.js. The [nodejs wrapper](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs) in this repo has had a lot of updates in the last couple years. When I clone this repo and build from there, it works. However, that doesn't help me declare a working dependency for my app. If the README is going to continue to recommend installing from npm, can a new release be published? If not, is there a recommended way to use the nodejs wrapper from my app? So far my only idea is to fork this repo to publish my own version of the nodejs wrapper on the npm register with a different package name and version, or at least to clone this repo and build an npm package tarball which I'd have to ship with my app and install from the tarball. I'd prefer an updated official package than either of those options.

MartyG-RealSense, 2022-03-19:
Hi @sparkyb  The most recent example of a successful Node.js wrapper build with **npm install** on this support forum was in November 2021 at https://github.com/IntelRealSense/librealsense/issues/9983

I would recommend using npm install if possible and not installing the Nodejs wrapper from source code, as source-code builds typically have problems.

In https://github.com/IntelRealSense/librealsense/issues/9983 the RealSense users in that case were building with the instruction below:

`npm install --save node-librealsense`

They found though that although the wrapper built successfully, the examples would not work.  Their solution in their particular case was to upgrade their Node.js version to **14.18.1** and to use the command below before performing the npm install.

`sudo apt-get install xorg-dev libxcb1-dev libglfw3 libglfw3-dev`

Support for Node.js 14 was added to the RealSense SDK from version 2.44.0 onwards.  Support for Node.js versions newer than 14 is not currently available in the RealSense SDK.

sparkyb, 2022-03-23:
I'm not sure what you mean by "not installing the Nodejs wrapper from source code". That's exactly what I'm trying to avoid. I'm trying to install the Node.js wrapper *without* having to clone the librealsense git repo at all and just using `node install --save node-librealsense`. I'm not sure what their setup was in that other issue you referenced, but it isn't working for me. I was using node 14.16.1, but I just upgraded to 14.19.1 and it had no effect.

The building I was referring to is what happened when you run `node install --save node-librealsense`. The Node.js wrapper is a native module and it doesn't download a prebuilt binary, it downloads [a tarball](https://registry.npmjs.org/node-librealsense/-/node-librealsense-0.2130.0.tgz) with a stripped down copy of the librealsense source and the wrapper source. preinstall it runs the [build-librealsense.js](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/nodejs/scripts/npm_dist/build-librealsense.js) script which builds librealsense and then it uses `node-gyp` to compile the wrapper.

The first part is completing successfully (building librealsense). But the wrapper is not compiling because it's calls to C++ Node functions don't match the changed function signatures (log below). [This commit](https://github.com/IntelRealSense/librealsense/commit/a03f2afd7e47faf9e7dde5d08c3356387eefc568) from 2020 has the fixes. It says "support nodejs 12, but it drop compatibility with nodejs 6". However, [the package currently uploaded to the npm registry](https://www.npmjs.com/package/node-librealsense) (the one you get when you run `npm install --save node-librealsense`) is from 2018 and does not have this change.

Someone needs to bump the version number in [package.json](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/nodejs/package.json) and then run `node ./scripts/generate-dist-package.js`. That will create a new tarball that can be uploaded to the npm registry. Then someone with ownership of that package on npm need to run `npm publish <tarball>`. When I git clone this repo, I am able to run `generate-dist-package.js` and create the Node.js wrapper tarball, even without building librealsense (I can't upload this package to npm though because I don't have permission). I was able to install the Node.js wrapper from that tarball with `npm install --save <tarball>`, even on another machine that never cloned this repo. That module built from the latest version works fine. My current plan is to package that tarball with my app, but I'd prefer to have the package on npm updated so I can install from there.

Here's the log from my `npm install --save node-librealsense`:
```
0 verbose cli [
0 verbose cli   '/usr/bin/node',
0 verbose cli   '/usr/bin/npm',
0 verbose cli   'install',
0 verbose cli   '--save',
0 verbose cli   'node-librealsense'
0 verbose cli ]
1 info using npm@8.5.5
2 info using node@v14.19.1
3 timing npm:load:whichnode Completed in 0ms
4 timing config:load:defaults Completed in 1ms
5 timing config:load:file:/usr/lib/node_modules/npm/npmrc Completed in 2ms
6 timing config:load:builtin Completed in 2ms
7 timing config:load:cli Completed in 1ms
8 timing config:load:env Completed in 0ms
9 timing config:load:file:/home/level99/temp/.npmrc Completed in 0ms
10 timing config:load:project Completed in 1ms
11 timing config:load:file:/home/level99/.npmrc Completed in 0ms
12 timing config:load:user Completed in 1ms
13 timing config:load:file:/usr/etc/npmrc Completed in 0ms
14 timing config:load:global Completed in 0ms
15 timing config:load:validate Completed in 0ms
16 timing config:load:credentials Completed in 0ms
17 timing config:load:setEnvs Completed in 0ms
18 timing config:load Completed in 7ms
19 timing npm:load:configload Completed in 7ms
20 timing npm:load:setTitle Completed in 0ms
21 timing config:load:flatten Completed in 1ms
22 timing npm:load:display Completed in 2ms
23 verbose logfile /home/level99/.npm/_logs/2022-03-23T19_48_45_115Z-debug-0.log
24 timing npm:load:logFile Completed in 2ms
25 timing npm:load:timers Completed in 0ms
26 timing npm:load:configScope Completed in 0ms
27 timing npm:load Completed in 14ms
28 timing arborist:ctor Completed in 0ms
29 silly logfile start cleaning logs, removing 1 files
30 timing arborist:ctor Completed in 0ms
31 timing idealTree:init Completed in 8ms
32 timing idealTree:userRequests Completed in 2ms
33 silly idealTree buildDeps
34 silly fetch manifest node-librealsense@*
35 http fetch GET 200 https://registry.npmjs.org/node-librealsense 459ms (cache revalidated)
36 silly placeDep ROOT node-librealsense@0.2130.0 OK for: temp@1.0.0 want: *
37 silly fetch manifest bindings@^1.2.1
38 silly fetch manifest fs-compare@0.0.4
39 silly fetch manifest jsonfile@^3.0.1
40 silly fetch manifest nan@^2.8.0
41 silly fetch manifest pngjs@^3.3.0
42 http fetch GET 200 https://registry.npmjs.org/bindings 52ms (cache revalidated)
43 http fetch GET 200 https://registry.npmjs.org/pngjs 60ms (cache revalidated)
44 http fetch GET 200 https://registry.npmjs.org/jsonfile 80ms (cache revalidated)
45 http fetch GET 200 https://registry.npmjs.org/nan 92ms (cache revalidated)
46 http fetch GET 200 https://registry.npmjs.org/fs-compare 394ms (cache revalidated)
47 timing idealTree:#root Completed in 863ms
48 silly placeDep ROOT bindings@1.5.0 OK for: node-librealsense@0.2130.0 want: ^1.2.1
49 silly placeDep ROOT fs-compare@0.0.4 OK for: node-librealsense@0.2130.0 want: 0.0.4
[... cut ...]
117 verbose unfinished npm timer reify 1648064925276
118 verbose unfinished npm timer reify:build 1648064926718
119 verbose unfinished npm timer build 1648064926719
120 verbose unfinished npm timer build:deps 1648064926719
121 verbose unfinished npm timer build:run:install 1648064987771
122 verbose unfinished npm timer build:run:install:node_modules/node-librealsense 1648064987771
123 verbose code 1
124 error A complete log of this run can be found in:
124 error     /home/level99/.npm/_logs/2022-03-23T19_48_45_115Z-debug-0.log
```

MartyG-RealSense, 2022-03-24:
When I said about "not installing the Nodejs wrapper from source code", I am just highlighting that an npm install of the Node.js wrapper is usually the best installation method to use, and that if npm install is used then the source code installation instructions on the wrapper documentation page should be skipped.

I note that the specification information at the top of this case states that you are using firmware driver version **05.11.01.100**.  This firmware is very old, dating back to February 2019.  I would highly recommend updating the firmware of your camera to a modern version.  

It looks as though the commit alters the **package.json** SDK wrapper file at wrappers/nodejs/package.json and not the Node.js API at https://www.npmjs.com/package/node-librealsense and so it is consistent that the Node.js version support has been updated but the API page is still 4 years old.

[image]
 
Similar to the commit that added Node.js 12 support, RealSense users are welcome to create a Pull Request (PR) that adds support for newer Node.js versions.  The RealSense team can then consider that PR for merging into the SDK as a commit.

https://github.com/IntelRealSense/librealsense/pulls

sparkyb, 2022-03-25:
> It looks as though the commit alters the **package.json** SDK wrapper file at wrappers/nodejs/package.json and not the Node.js API at https://www.npmjs.com/package/node-librealsense and so it is consistent that the Node.js version support has been updated but the API page is still 4 years old.

I don't understand this. I'm not sure what distinction you're making between the "SDK wrapper" and "Node.js API". In addition to package.json, that commit I referenced updates wrappers/nodejs/src/addon.cpp which is the source code for the wrapper, the bit that ties librealsense to Node.js. It was updated to support Node.js 12+, but it has never been published to NPM. So if I install the Node.js wrapper from git it works, but if I install from NPM it is still the old version that doesn't support Node.js 12+ yet.

MartyG-RealSense, 2022-03-25:
I am not involved in RealSense development as I am a support engineer, so I cannot offer insight on development-related aspects of the Node.js wrapper such as this source vs npm question, unfortunately.

MartyG-RealSense, 2022-03-31:
Hi @sparkyb  Bearing in mind the response provided in the comment above, do you require further assistance with this case, please?  Thanks!

poofytoo, 2022-03-31:
Hi,

Could you please connect us to someone who might have permissions to upload to the [npm library](https://www.npmjs.com/package/node-librealsense), and then follow @sparkyb's instructions of:

`Someone needs to bump the version number in [package.json](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/nodejs/package.json) and then run node ./scripts/generate-dist-package.js. That will create a new tarball that can be uploaded to the npm registry. Then someone with ownership of that package on npm need to run npm publish <tarball>`

MartyG-RealSense, 2022-03-31:
Hi @poofytoo  This is not a task that the Intel RealSense team can carry out, unfortunately.  

As with other types of RealSense wrapper, RealSense users are welcome to create a Pull Request (PR) that provides updates for the Node.js wrapper. The RealSense team can then consider that PR for merging into the SDK as a commit.

https://github.com/IntelRealSense/librealsense/pulls

Intel commit to maintaining the working status of wrappers that are contributed to the RealSense SDK, though updates beyond essential maintenance to keep them functioning are not guaranteed.

poofytoo, 2022-03-31:
Thank you so much for the quick response! It looks like the updating is/was carried out by a intel realsense team member as specified in the package.json and on the [npm library](https://www.npmjs.com/package/node-librealsense): "Kenny Yuan <kaining.yuan@intel.com>" While I understand this individual may no longer work on the Intel RealSense, is there another team member who can do this?

MartyG-RealSense, 2022-03-31:
@poofytoo  I will forward your request to Intel.

poofytoo, 2022-03-31:
> @poofytoo I will forward your request to Intel.

Thank you!

MartyG-RealSense, 2022-04-03:
Hi @poofytoo  I received feedback from Intel regarding your request to update the npm packages.  They confirmed my earlier response in this discussion at https://github.com/IntelRealSense/librealsense/issues/10322#issuecomment-1084647903 that RealSense users create a Pull Request to submit changes to the Node.js wrapper.

sparkyb, 2022-04-04:
I would be happy to submit a pull request, however that would not help. A pull request only allows me to fix an issue with the code in GitHub. However, the code in GitHub works great. The problem is that it is not published and I cannot install it directly from GitHub without it being published to NPM. If it is the decision of Intel that you no longer wish to publish the Node.js wrapper to NPM, then I will submit a PR to update the README to remove the confusing recommendation to users that they install a severely outdated version of the Node.js wrapper from NPM.

MartyG-RealSense, 2022-04-04:
Hi @sparkyb  You are welcome to submit such a PR, though ultimately it will be the decision of the RealSense development team whether that PR is accepted and activated.  Rather than removing the NPM, adding a cautionary message to the NPM section of the README may work better.

MartyG-RealSense, 2022-04-16:
Hi @sparkyb  Do you still plan to submit a PR to update the README of the Node.js wrapper, please?  Thanks!

MartyG-RealSense, 2022-04-28:
Case closed due to no further comments received.

closed 2022-04-28 by MartyG-RealSense

callumhay, 2022-08-03:
I just want to add to this discussion that after continuing to experience issues with trying to install the npm package as described in the previous comments and in the README, I was able to finally get the nodejs wrapper working by following the comments of @sparkyb 

i.e., 
```
node ./scripts/generate-dist-package.js
cd <some_project_dir>
npm install <some_relative_path>/librealsense/wrappers/nodejs/dist/node-librealsense-0.2130.0.tgz
```

In other words: can we please get the npm package up to date asap so it can be installed seamlessly using the node package manager?

MartyG-RealSense, 2022-08-03:
It's great news that you were successful, @callumhay  :)

As discussed in this case, the most achievable outcome is likely to be for a RealSense community member to submit a pull request to update the installation instructions readme.

Komsomol, 2023-08-22:
None of the node-js exmaple links work. All 404. Even coming from the official website.

MartyG-RealSense, 2023-08-22:
Hi @Komsomol  The NodeJS wrapper has been removed from the RealSense SDK from version 2.54.1 onwards and is no longer supported, unfortunately.  It was no longer possible to update the wrapper to support newer NodeJS versions than 14 and so could not meet the needs of users who were working with modern NodeJS versions.

An option for you to use it though may be to visit the release page of the previous 2.51.1 SDK version which precedes the removal and then download the SDK's source code and build it from that code.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

Fortunately the Internet Archive's **Wayback Machine** website that archives the internet managed to save the example pages to their own archives before they were removed from the official RealSense site.

https://web.archive.org/web/20210906064205/https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs/examples