realsenseai/librealsense #10324: How to remove color from an image?

jo4omath, 2022-03-20:
Hello, I am working with an D435i and I want to remove a color from my image. 
The language that I am using is MATLAB.
I have a RGB matrix nx3 and  my goal is remove certain type of color. 
The problem is how can I get an 3D image from that matrix that I changed? 
Anyone knows where can I get documentation in order to understand the formation from a matriz to an image?
Thank you.

MartyG-RealSense, 2022-03-20:
Hi @jo4omath  It sounds as though you may have already achieved the removal of a certain type of color, but in case you have not I will provide some references below for ways to approach doing so.

A common approach for segmenting a particular color is to use an OpenCV function called **inRange**.  A tutorial from the official OpenCV documentation at the link below provides example C++ code for doing so.  The page also has Java and Python versions of the code for people who are using those languages.

https://docs.opencv.org/3.4/da/d97/tutorial_threshold_inRange.html

As you were using PCL instructions in your other recent case at https://github.com/IntelRealSense/librealsense/issues/10311#issuecomment-1066076702 then you may prefer a PCL-based approach though.  The PCL instruction **conditional_removal** may help, as it can be used with a color component in RGB color space or HSI color space.

https://pointclouds.org/documentation/classpcl_1_1_conditional_removal.html

The link below has a scripting example of conditional_removal being used to segment color from an PointXYZRGB point cloud.

https://stackoverflow.com/questions/59974437/pcl-color-filter

*********

Could you provide more information please about the '3D image' that you are attempting to generate, please.

jo4omath, 2022-03-21:
@MartyG-RealSense Thank you for your useful comment. 
I would like to ask you if this functions are available  in MATLAB? 

What I am trying to do is remove the background. I have some rocks in the image and I want to determine its coordinates and dimensions. I got an image and then I calculated its RGB matrix, what I did was change the points with high red value to [0 0 0](I am supposing that the background is red), I was thinking about transform now the matrix to an image  but for what I read in your first link thats not how it works.

MartyG-RealSense, 2022-03-27:
Hi @jo4omath  Do you require further assistance with this case, please?  Thanks!

jo4omath, 2022-03-27:
Hello @MartyG-RealSense  sorry for didnt answer before. The documentation that you provided was helpful, I am working on it. 
If any question comes up I will get in touch. 
Thank you so much.

MartyG-RealSense, 2022-03-27:
You are very welcome, @jo4omath - please do let me know if there is something that I can help you with if you have further questions that arise.  Good luck!

jo4omath, 2022-04-02:
Hi @MartyG-RealSense, now I can remove a color from an image, thanks for your help! As you know I am trying to get an 3D representation of an object. I have search for documentation but I didnt find one that could help me, could you provide some documentation to do that in matlab? I visited the intel realsense online page that have examples but it did not help me. 
Thanks!

MartyG-RealSense, 2022-04-02:
Are you attempting to create a 3D pointcloud or generate a solid 3D mesh object from a pointcloud, please?

jo4omath, 2022-04-02:
Given an object I am trying to determine its 3D pointcloud. The main goal is calculate its volume.

MartyG-RealSense, 2022-04-02:
Previous research I did about the subject of estimating volume from a pointcloud indicated that the popular opinion is that the point cloud should be converted to a 3D mesh before doing the volume estimation.  The RealSense SDK does not have built-in functions for doing this pointcloud to mesh conversion, as it is best suited to specialized 3D processing libraries such as Open3D and PCL.  The link below discusses doing so with PCL.

https://stackoverflow.com/questions/55629892/2019-point-cloud-volume-estimation-with-c/55681149#55681149

If you wanted to try to perform the volume estimation with a built-in function inside MATLAB itself then the **convhull** instruction can apparently help to achieve that.

https://www.mathworks.com/matlabcentral/answers/145417-calculate-the-volume-of-shape-descibed-by-3d-points
https://www.mathworks.com/help/matlab/ref/convhull.html

closed 2022-04-16 by MartyG-RealSense