realsenseai/librealsense #10325: Purple image while capturing single frame with fixed exposure

bhomaidan1990, 2022-03-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version | Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra )                                         | 
| Platform                                 | NVIDIA Jetson Nano 4G  |
| SDK Version                            |  2.50.0  | 
| Language                            | python | 
| Segment			| others | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I get a purple image while capturing single frame with manual exposure:

(note: resized to half-scale image example)
[image]

```py
#! /usr/bin/python3
# -*- coding: utf-8 -*-

import numpy as np
import cv2
import pyrealsense2 as rs

def grab_frame(resolution=(1920, 1080), verbose=True):
    """
    Function: grab_frame, to grab an image.
    ---
    Parameters:
    @param: resolution, tuple, image frame resolution. L515 RGB resolution [(1920, 1080), (1280, 720), (960, 540)]
    @param: verbose, boolean, to show the output of the function. 
    ---
    @return: color_array, nd-array, color image frame.            
        
    """
    # Start reading frames
    pipeline = rs.pipeline()
    config = rs.config()
    # RGB Stream
    config.enable_stream(rs.stream.color, resolution[0], resolution[1], rs.format.bgr8, 30)

    # Start streaming
    pipeline.start(config)

    # Get the sensor once at the beginning. (Sensor index: 1)
    sensor = pipeline.get_active_profile().get_device().query_sensors()[1]

    # Set the exposure anytime during the operation
    sensor.set_option(rs.option.exposure, 800.000)
    try:
        frames = pipeline.wait_for_frames()
        color_frame = frames.get_color_frame()
        # Convert images to numpy arrays
        color_array = np.asanyarray(color_frame.get_data())
        cv2.namedWindow("res", cv2.WINDOW_NORMAL)
        cv2.imshow("res", color_array)
        cv2.waitKey(2000)
        cv2.imwrite("img2.png", color_array)
    except Exception as e:
        print(e)
    finally:
        cv2.destroyAllWindows()
        pipeline.stop()
        # return color_array

if __name__ == "__main__":
    grab_frame()
```

MartyG-RealSense, 2022-03-21:
Hi @bhomaidan1990  If you are using a manual RGB exposure of '800' when its default is '156', I wonder whether you were doing so because the image becomes dark when auto-exposure is disabled (like in the image below) and increasing RGB exposure brightens it?

[image]

If so, does leaving the RGB exposure at the default value and instead increasing **gain** to '1500' enable you to capture a single RGB frame with correct coloring?

`sensor.set_option(rs.option.gain, 1500)`

[image]

bhomaidan1990, 2022-03-21:
@MartyG-RealSense thanks for your rapid reply, 
1. "*manual RGB exposure of '800' when its default is '156'*", the point is to increase capture speed and reduce exposure convergence time.
2. "*does leaving the RGB exposure at the default value and instead increasing RGB gain to '1500' enable you to capture a single RGB frame with correct coloring?*" No, changing the exposure back to auto and changing the gain gave the same purple image!

MartyG-RealSense, 2022-03-21:
It looks as though the camerais looking down from the ceiling towards the floor.  Do you have a strong light source such as a ceiling light near the camera?

bhomaidan1990, 2022-03-21:
> It looks as though the camera is looking down from the ceiling towards the floor. Do you have a strong light source such as a ceiling light near the camera?

- "*the camera is looking down from the ceiling towards the floor*" **True**.
- "*Do you have a strong light source such as a ceiling light near the camera?*" I have multi indirect lights to make industrial fixed lighting and not sunlight, there is 1-metre distance at least from the camera, and all lights are on top of it.

MartyG-RealSense, 2022-03-21:
Next, please try commenting out the `cv2.imshow("res", color_array)` line to eliminate the possibility that there is a conflict when performing imshow before imwrite.

MartyG-RealSense, 2022-03-21:
Does the same issue occur if you take a single-frame PNG snapshot in the RealSense Viewer using the camera-shaped snapshot option on the top of the RGB panel?

[image]

bhomaidan1990, 2022-03-21:
> Does it make a difference if you skip the first several frames by using a 'for' instruction directly before wait_for_frames()
> 
> ```
> for i in range(5):
> frames = pipeline.wait_for_frames()
> ```

I confirm that this delay has solved the problem, but is there a way to get the minimum delay possible? thanks in advance.

closed 2022-03-21 by bhomaidan1990

MartyG-RealSense, 2022-03-21:
Great to hear that a 'for' statement resolved your problem!

You can try a lower value than 5 in the range bracket so that the number of frames skipped is reduced.