realsenseai/librealsense #10327: 2 cameras used : Same parameters but pointcloud different

PierreLouisT, 2022-03-22:
Hi ! I am using both D455 cameras and appling a calibration with XML document. When I use the viewer from Intel, I got those 2 pointclouds : 
[image]
[image]

I have exactly the same parameters for both and when I use the calibration factory, pointclouds are identic, so my problem could come from my calibration but why there is a difference whereas the same doc xml is used ? 

Thank you for the help you could bring me, 

Pierre Louis

MartyG-RealSense, 2022-03-22:
Hi @PierreLouisT  When using the camera with the missing pointcloud detail in the Viewer, could you check whether a post-processing filter called **Threshold Filter** is enabled (indicated by a blue icon beside it).  If it is enabled then please left-click on the blue icon to turn its color to red (Off) and see whether the pointcloud improves.

[image]

PierreLouisT, 2022-03-22:
Thank you Marty for your quick and precise answer, the threshold doesnt change anything and all the parameters are identic so I do not think the problem comes from here. In both case I apply the same parameters and post processing filters ;)

MartyG-RealSense, 2022-03-22:
Thank you.  If you are using the same XML calibration document for both cameras then that may be causing your problem.  Each individual camera has unique calibration values that are determined by the manufacturing process at the factory, so I would recommend trying to create two separate XML calibration documents for each camera to see whether it resolves the problem.

PierreLouisT, 2022-03-22:
Ok, I got it, thank you for your advice, I come back to you to tell you if it worked or not

PierreLouisT, 2022-03-23:
Hi ! Unfortunately it didn't change anything, For one camera my calibration is as well as I wanted but the pointcloud of the second one remains noisy and shattered
As you said each cameras has individual parameters and I adapted my XML in consequences to be coherent but didn't worked. 
Do you have any other suggestions ? 
Thank you

MartyG-RealSense, 2022-03-23:
After you have loaded in an XML calibration file, have you then written the calibration to the camera hardware's internal storage with the **-w** option?

[image]

PierreLouisT, 2022-03-23:
I found the solution, the problem was that camera has not the same parameters as translation and rotation matrix when they are calibrated with factory parameters.  I had not pay attention very for those and for now, when I modified also those with a calibration doc I got what I needeed. Thank you Marty for helped me a lot, your first suggestion was the good one. 

Hope this issue will help somebody else, 

If you are using multi-camera for a certain calibration, pay attention for factory parameters for each cameras.

closed 2022-03-23 by PierreLouisT

MartyG-RealSense, 2022-03-23:
It's great to hear that you achieved a solution, @PierreLouisT - thanks so much for sharing your findings with the RealSense community!