realsenseai/librealsense #10330: Motion module cannot be activated - Permission denied

nedo99, 2022-03-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.05.50 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.10                                         | 
| Platform                                 | N/A  |
| SDK Version                            |  2.50                          | 
| Language                            |  C++ and Realsense viewer                          | 
| Segment			|  Robot                          | 

### Issue Description
Hi,

Trying to use gyro and accel, but it is not working due to Permission denied! I start `realsense-viewer` and when I try to toggle the 'Motion module' I get this message:
```
Backend in rs2_open_multiple(sensor:0x7ffa50018e00,
profiles:0x7ffa50f9a908, count:2):
Failed to open scan_element
/sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.5/0003:8086:0B3A.001F/HID-SENSOR-200076.4.auto/iio:device1/scan_elements/in_anglvel_x_en
Last Error: Permission denied
```

While I was clicking that one, on the command line, got this message:
```
[140714690418496] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 22/03 15:02:25,490 WARNING [140713025062656] (backend-hid.h:48) write_fs_attribute with gyro_3d-dev1 failed. The specified path /sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.5/0003:8086:0B3A.001F/HID-SENSOR-200076.4.auto/iio:device1/trigger/current_trigger is not valid
 22/03 15:02:25,490 ERROR [140714690418496] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.5/0003:8086:0B3A.001F/HID-SENSOR-200076.4.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
 22/03 15:02:25,490 ERROR [140714690418496] (backend-hid.cpp:1003) Hid device is busy!
```

`udev` and `dev` librealsense packages are installed on the system. Infra and RGBD work fine.

Looking forward to your replies!

Regards,
Nedim

MartyG-RealSense, 2022-03-22:
Hi @nedo99  The error  **Failed to open scan_element** can indicate that the udev rules may be not installed or improperly installed, as described at https://github.com/IntelRealSense/librealsense/issues/3242#issuecomment-464775017 and https://github.com/IntelRealSense/librealsense/issues/5049#issuecomment-542175378

nedo99, 2022-03-23:
I've tested with librealsense 2.48 on another machine and it is the same issue.

Kernel version is `5.13.0-30-generic`. And it was built using `sudo apt get`

MartyG-RealSense, 2022-03-23:
Problems with using 5.13 have been previously reported, and also problems with some 5.11 versions (bearing in mind that 5.11 support was provided via a user-contributed patch rather than an official feature).  A couple of RealSense users have reported that **5.11.0-34-generic** worked well for them when other 5.11 versions were unstable.

nedo99, 2022-03-23:
Good to know. Thanks for the updates!

MartyG-RealSense, 2022-03-29:
Hi @nedo99  Do you require further assistance with this case, please?  Thanks!

nedo99, 2022-03-29:
> Hi @nedo99 Do you require further assistance with this case, please? Thanks!

Waiting for the higher kernel support. I guess that is it. If you think that is enough to close this ticket I am fine with it.

adamanov, 2024-07-23:
> Thanks very much for the update. As there is no information available about future kernel support updates at the time of writing this, I will close this case. Thanks again!

Hello I do have a similar issue, and unfortunately i could not figure out how to solve it, and this issue is consistent with other computer and other realsense camera, Any other ideas how to solve it? We are not able to use basically the IMU module ...

**Here is our setup**

- Platfrom Ubuntu 22.04 & ROS2 Humble
- RealSense ROS v4.55.1
- Built with LibRealSense v2.55.1
Device Name                   |    Serial Number      |  Firmware Version
Intel RealSense D435I     |    233522070877     |  5.13.0.55

`Backend in rs2_open_multiple(): to open scan_element Last Error: Permission denied`

MartyG-RealSense, 2024-07-23:
Hi @adamanov  The camera firmware driver version that should be used with SDK 2.55.1 and wrapper 4.55.1 is **5.16.0.1**.

The IMU topics are disabled in the ROS wrapper by default and can be enabled by adding the commands below to the launch instruction.

**enable_accel:=true enable_gyro:=true unite_imu_method:=2**

These commands enable the IMU accel and gyro streams and publish them both in a single **/imu** topic.

adamanov, 2024-07-23:
> Hi @adamanov The camera firmware driver version that should be used with SDK 2.55.1 and wrapper 4.55.1 is **5.16.0.1**.
>
> The IMU topics are disabled in the ROS wrapper by default and can be enabled [...]

yes i realiesed that there is need to upgrade, however whenever i do it, the device will automatically disconnected, and the process is basically interrupted  

[image]

MartyG-RealSense, 2024-07-24:
@adamanov  When a firmware update is initiated, once the process gets to around 30% then the camera disconnects and reconnects.  Sometimes the reconnection fails though, causing the firmware update to fail and the camera to go into 'recovery mode'.

The **rs-fw-update** firmware update tool has a function for recovering a camera that is in recovery mode.

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

If the .bin firmware file for firmware 5.16.0.1 has been downloaded to your computer then the **-r** flag can be used when launching rs-fw-update.  For example:

`rs-fw-update -r -f Signed_Image_UVC_5_16_0_1.bin`

adamanov, 2024-12-03:
> @adamanov When a firmware update is initiated, once the process gets to around 30% then the camera disconnects and reconnects. Sometimes the reconnection fails though, causing the firmware update to [...]

Hello, 

sorry for a late response, I just come back to this problem with IMU, 
I did perform what you asked, and still the results are same 

```
lenova@lenova:~/realsense/Signed_Image_UVC_5_16_0_1$ rs-fw-update -r -f Signed_Image_UVC_5_16_0_1.bin 

Update to FW: Signed_Image_UVC_5_16_0_1.bin

Recovering device: 
[0ADB] Intel RealSense D4XX Recovery, update serial number: 852323053435, firmware version: 5.16.0.1

Firmware update started. Please don't disconnect device!

Firmware update progress: 100[%]

Firmware update done
Waiting for new device...
 03/12 15:56:37,834 ERROR [130383070889536] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video36 Last Error: Permission denied
 03/12 15:56:37,834 ERROR [130383070889536] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video37 Last Error: Permission denied
 03/12 15:56:37,835 ERROR [130383070889536] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video38 Last Error: Permission denied
 03/12 15:56:37,835 ERROR [130383070889536] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video39 Last Error: Permission denied
 03/12 15:56:37,835 ERROR [130383070889536] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video40 Last Error: Permission denied
 03/12 15:56:37,835 ERROR [130383070889536] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video41 Last Error: Permission denied

Recovery done
```

```
lenova@lenova:~/realsense/Signed_Image_UVC_5_16_0_1$ rs-fw-update -r -f Signed_Image_UVC_5_16_0_1.bin

Update to FW: Signed_Image_UVC_5_16_0_1.bin

No recovery devices were found!

```

here is also device info

```
lenova@lenova:~$ rs-enumerate-devices  
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	843112070679
    Firmware Version              : 	5.16.0.1
    Recommended Firmware Version  : 	5.16.0.1
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.0/video4linux/video48
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	852323053435
    Firmware Update Id            : 	852323053435
    Dfu Device Path               : 	
```

any other ideas, how could we get the IMU dates?

MartyG-RealSense, 2024-12-03:
Hi @adamanov  It looks as though your firmware update process has completed successfully and the camera has recovered and come out of the recovery state back to normal operation.

```
Firmware update progress: 100[%]
Firmware update done
Recovery done
No recovery devices were found!
```

Are you able to successfully use the camera with RealSense applications such as realsense-viewer to stream depth and RGB, even if you have problems with IMU data?

I note that your computer is using kernel 6.8.  Support for that kernel was not introduced until beta version 2.56.2 of the librealsense SDK.  Currently you are using the older librealsense version 2.55.1 whch does not have kernel 6.8 support.  

I would therefore recommend trying to build librealsense from source code using the **libuvc backend** procedure if you have not tried it already.  This installation bypasses the kernel and so can resolve problems where the camera is not working correctly due to a kernel related issue.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc backend procedure provides a pre-made build script called 'libuvc_installation.sh' that will install librealsense in libuvc kernel bypass mode when run.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

adamanov, 2024-12-05:
>Are you able to successfully use the camera with RealSense applications such as realsense-viewer to stream depth and RGB, even if you have problems with IMU data?

Yes, I see everything (RGB RGBD IR images) besides IMU data in `ros2` and via `realsense-viewer`  

> https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

with installation from source using libuvc backend procedure. it does work right now :). Here is a bellow configuration. Thank you for your help 

```
header:
  stamp:
    sec: 1733425104
    nanosec: 269305856
  frame_id: camera_imu_optical_frame
orientation:
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance:
- -1.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: -0.0010646508308127522
  y: -0.0005323254154063761
  z: 0.003728023264557123
angular_velocity_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
linear_acceleration:
  x: -0.08825984597206116
  y: 0.16671304404735565
  z: -9.81645679473877
linear_acceleration_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
```

MartyG-RealSense, 2024-12-06:
Its great to hear that the IMU is working for you!

I note that your linear acceleration's Y value is **-9.8164**.  This shows that your IMU's calibration is excellent.

adamanov, 2024-12-10:
> Its great to hear that the IMU is working for you!
> 
> I note that your linear acceleration's Y value is **-9.8164**. This shows that your IMU's calibration is excellent.

oh really? I did a calibration also using the script which you guys provided :)  then it is kinda a confirmation that it works perfectly ))