realsenseai/librealsense #10333: distance problem

zzcqinag, 2022-03-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515  | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    | 4.14.13  | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.45.0  | 
| Language                            | C | 
| Segment			| Robot | 

### Issue Description
l put L515 in the middle of some boxes, and l find The upper part is less distant than the lower part in realsense-viewer,but l think 
the distance in the upper part and the lower part are the same, Does anyone have this problem？
[image]

MartyG-RealSense, 2022-03-24:
Hi @zzcqinag  You are correct that if the front of the boxes is at approximately the same distance from the camera then the depth shading should be around the same color range throughout.

Would it be possible please to provide an RGB image of the scene that the camera is observing in order to identify factors in the scene that may be negatively affecting the results?

zzcqinag, 2022-03-24:
Hi @MartyG-RealSense, thanks for your answer.
the RGB image as follow.
[image]

MartyG-RealSense, 2022-03-24:
Thanks very much for the RGB image, it's very helpful.  If the camera observes a regular horizontal and / or vertical pattern (such as a horizontal row of fence posts or a vertical stack of window blind slats) then it can treat it as a **repetitive pattern** and become confused, resulting in an inaccurate depth image or 'ghost' depth areas on the image that are not present in the real-world scene.  At https://github.com/IntelRealSense/librealsense/issues/6713#issuecomment-651114720 a RealSense team member provides advice about trying to negate the effects of a repetitive pattern.

closed 2022-04-08 by MartyG-RealSense