realsenseai/librealsense #10335: D405  How to measure depth

patrickpoirier51, 2022-03-24:
Hello,

I just received the D405 for evaluation
Uploaded Realsense Viewer 2.50.0 and updated formware to 5.13.00

Here is my test setup, please note that I am using the Blue Usb cable of the T265 as the supplied cable for D405 is too big for such a small device:
[image]

Issues:
Using the mouse scrool wheel to zoom is intermittent
We cannot measure distance when zoom is to a certain level 
That is the closest level we can measure:
[image]

It is too small for any valuable work...  The next picture show the correct zoom level for precise measurement:
[image]

Questions:
How do we measure depth with the SDK, are there any examples ?
How to set reference point of top of surface and detect depth and measure ?

Info:
There is no reference to D405 on the latest release.
https://github.com/IntelRealSense/librealsense/releases

MartyG-RealSense, 2022-03-24:
Hi @patrickpoirier51  Version 2.50.0 of the RealSense SDK is already compatible with D405 even though it is not yet listed on the supported models on the Releases page.  Support for a new model is typically added before its public release so that the camera can be tested.

The current RealSense ROS2 wrapper (the 4.x wrapper series) is also compatible with D405, though it is not planned that D405 support will be added to the ROS1 wrapper 

I performed extensive tests with the Viewer.  I found that although the coordinate measurement in 3D point cloud mode also disappeared on other camera models such as D455 when zooming in very close with the scroll wheel, the coordinates disappeared sooner on D405, making it difficult to maintain the coordinate display at extreme zoom-in.

A workaround that I found to preserve the coordinates at extreme zoom-in on D405 was to go to **Advanced Controls > Depth Table** and reduce the value of the **Depth Units** option from its D405 default of '10' to a lower value such as '5' or '1'.

[image]

patrickpoirier51, 2022-03-24:
Thanks Marty for your fast response - as usual ;-)
Yes, I played a little with this feature but I was expecting a better integration on the viewer as it is a precision device

So you confirm: there is no example provided exclusively for the D405 ?
We must build from the general SDK  and Tools to fit our need ?

MartyG-RealSense, 2022-03-24:
There is currently no D405-specific example program.  Usually though, the existing example programs for RealSense depth cameras work automatically with different camera models without having to have their code edited.  The images below are D405 used with rs-capture, rs-multicam and rs-pointcloud in that order.

[image]

[image]

[image]

patrickpoirier51, 2022-03-24:
Ok thanks
So I will pave the way.... just like with th T265 ;-)

MartyG-RealSense, 2022-04-01:
Hi @patrickpoirier51  Do you require further assistance with this case, please?  Thanks!

patrickpoirier51, 2022-04-01:
No
Thanks

closed 2022-04-01 by patrickpoirier51