realsenseai/librealsense #10336: BUG in x86 DLL

POWER-KI, 2022-03-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D405| 
| Operating System & Version |   Win 10 | 
| SDK Version                            | [Intel.RealSense.SDK-WIN10-2.50.0.3785.exe]

rs-hello- realsense.cpp if compiled as:
x86 -> CRASH 
x64 -> works.

Regards

MartyG-RealSense, 2022-03-24:
Hi @POWER-KI  Does the  **rs-hello-realsense.exe** pre-built example program launch correctly on your x86 computer, please?  As you have installed the full SDK for Windows, you should be able to find the program and the other pre-built .exe examples at the folder location below:

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > tools**

If you wish to build the example programs in Virtual Studio, Intel provide guidance for doing so at the link below.

https://www.intel.com/content/www/us/en/support/articles/000030463/emerging-technologies/intel-realsense-technology.html

If you were attempting to build rs-hello-realsense.cpp in Visual Studio for an x86 system, you could check whether the build target is configured for x86.

https://docs.microsoft.com/en-us/visualstudio/ide/how-to-configure-projects-to-target-platforms?view=vs-2022

POWER-KI, 2022-03-24:
The pre-built works (but I think that is an X64 version).

To check the problems:
1) Open with VS(2019) rs-examples;
2) compile hello-realsense and :  
- in x86 configuration it crash;
- in x64 configuration it works.

We have tested this because other programs tha we have with the new DLL crash.

Regards,
CAP

POWER-KI, 2022-03-24:
I think that the problems is  in
`rs2_create_device(_list.get(), index, &e);`

MartyG-RealSense, 2022-03-24:
Could you try building the starter C++ project in the tutorial linked to below please to see whether that project experiences the same problem.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

POWER-KI, 2022-03-25:
The x86 DLL with: 
L515 works;
D435 works;
D405 **CRASH**.

henryjliu, 2022-04-27:
Any update to this?  We also need the x86 compiled version for D405

MartyG-RealSense, 2022-04-27:
Hi @henryjliu  There is no further information to provide at this time.

henryjliu, 2022-04-27:
I tested a very simple program and it crashes on last line.  I am somewhat motivated to tackle this as would like to use D405 in our application and it needs to be x86.  Is there a cmake that creates the x86 compiled libraries?

            var ctx = new Context();
            var list = ctx.QueryDevices(); // Get a snapshot of currently connected devices
            if (list.Count == 0)
                throw new Exception("No device detected. Is it plugged in?");

            var dev = list[0];//throws system.accessviolationexception

MartyG-RealSense, 2022-04-27:
Building the RealSense SDK from source code should generate the RealSense.dll files, though I cannot offer guarantees regarding whether it will make a difference to whether the generated .dll works with D405.

**Windows source code build**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

There is no need to carry out the 'Enabling metadata on Windows' section of the instructions as metadata can be enabled through the RealSense Viewer program.

**Linux source code build**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

henryjliu, 2022-04-27:
I’m hoping to compile it in debug mode on x86 then it will tell me what is
wrong. However it appears x86 is not an out of box visual studio compile
option like x64. Do you have the compile scripts for x86?

MartyG-RealSense, 2022-04-27:
The RealSense.dll files have the same filename whether generated for x86 or x64.  

I do not have details of a specific librealsense compilation procedure for Visual Studio.  Microsoft do though provide guidance at the link below for setting the **Configuration Manager** of Visual Studio for the **Win32** platform (x86)

https://docs.microsoft.com/en-us/visualstudio/ide/how-to-configure-projects-to-target-platforms?view=vs-2022

*************

The Win32 platform name is used for C++ projects, and it means x86.  Visual Studio considers both project-level platforms and solution-level platforms, and the project platforms come from the language-specific project systems.  C++ projects use Win32 and x64, but the solution platforms use x86 and x64.  When you choose x86 as the solution configuration, Visual Studio selects the Win32 platform for C++ projects. 

To see both project-level platform and solution-level platform settings, open Configuration Manager and note the two platform settings.  The solution-level platform is shown in the Active solution platform dropdown, and the table shows the project-level platform for each project.

************

For building with CMake, my understanding is that you can set the target platform with the build term **CMAKE_GENERATOR_PLATFORM** - for example, `CMAKE_GENERATOR_PLATFORM=x86`

henryjliu, 2022-04-27:
Ok I got it to work with cmake -A win32.  I ran the rs-hello-realsense.cpp in debug and it crashes here:

henryjliu, 2022-04-27:
template <class _Keyty>
    _Tree_find_result<_Nodeptr> _Find_lower_bound(const _Keyty& _Keyval) const {
        const auto _Scary = _Get_scary();
        _Tree_find_result<_Nodeptr> _Result{{_Scary->_Myhead->_Parent, _Tree_child::_Right}, _Scary->_Myhead};
        _Nodeptr _Trynode = _Result._Location._Parent;
        while (!_Trynode->_Isnil) {
            _Result._Location._Parent = _Trynode;
            if (_DEBUG_LT_PRED(_Getcomp(), _Traits::_Kfn(_Trynode->_Myval), _Keyval)) {
                _Result._Location._Child = _Tree_child::_Right;
                _Trynode                 = _Trynode->_Right;
            } else {
                _Result._Location._Child = _Tree_child::_Left;
                _Result._Bound           = _Trynode;
                _Trynode                 = _Trynode->_Left;
            }
        }

        return _Result;
    }

henryjliu, 2022-04-27:
Works fine with D435 but crashes with D405

henryjliu, 2022-04-27:
Not sure what this _Find_lower_bound code does.  Is there a way to escalate to Realsense Engineer?

bostjanv, 2022-04-28:
_Find_lower_bound is part of VS C++ standard library implementation, but the bug must be somewhere higher in the call stack.

MartyG-RealSense, 2022-04-28:
Thanks very much for your ongoing debug process.  Please continue logging it here and I will pass the results on to Intel when you have completed your tests.  Good luck!

henryjliu, 2022-04-29:
I verified the same thing with compiling libraries in debug mode using cmake -a win32 vs cmake to generate the x64 debug libraries:
x64 D405 -> ok
x86 D405 -> crash
x64 D435 -> ok
x86 D435 -> ok
x64 L515 -> ok
x86 L515 -> ok

Seems with source code and compiling of debug mode should be able to figure out what's the difference between x64 and x86 code for D405.  Seems it crashes when it creates the device.  I'm not really a cpp expert so considering outsourcing it.

gwen2018, 2022-05-04:
@henryjliu thanks for reporting the issue to RealSense team. we reproduced the problem and will investigate. In the meantime, please try the solution below.
 
Remove/comment out the line highlighted below in src\ds5\advanced_mode\advanced_mode.cpp. The single line of code caused the library to crash with d405 but it's redundant. Removing it should fix the issue.

https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/advanced_mode/advanced_mode.cpp#L25
    // "Remove IR Pattern" visual preset is available only for D400, D410, D415, D460, D465
    std::string pid = _depth_sensor.get_info(RS2_CAMERA_INFO_PRODUCT_ID);

henryjliu, 2022-05-05:
Thanks for the workaround - confirmed it works now but will warrant further testing.  Not sure why it works fine in x64 and not x86 so prob should be deeper issue to be resolved.

gwen2018, 2022-05-06:
@henryjliu glad to know the workaround worked. thanks for the confirmation. on x86, a memory alignment issue on x86 along call stack in that particular line of code caused the crash. that's why the issue only appeared on x86. the rest of sdk use a different call stack, so the issue should not impact other areas. we have created an internal ticket to investigate it further to be sure.

RealSenseCustomerSupport, 2022-09-19:
----------------------------------------------

This issue has been resolved in RealSense SDK 2.51.1 (https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2511).

MartyG-RealSense, 2022-10-07:
Hi everyone, can anyone confirm if the x86 dll crash issue is resolved for them when using SDK 2.51.1 please?  Thanks!

MartyG-RealSense, 2026-10-25:
Case closed due to the discussed issue having been resolved in SDK version 2.51.1.

closed 2026-10-25 by MartyG-RealSense